evcc-io/evcc #5034: Reconnect MQTT on Connection lost

christoph-ng, 2022-11-03:
### Describe the bug

After a downtime of the mqtt broker evcc doesn't reconnect to the mqtt broker

### Steps to reproduce

manual stop mqtt broker 

### Configuration details

mqtt:
  broker: mqtt.home.lan:1883
  topic: evcc # root topic for publishing, set empty to disable
  # user:
  # password:

### Log details

[mqtt  ] ERROR 2022/11/03 09:07:27 tcp://mqtt.home.lan:1883 connection lost: EOF
[mqtt  ] ERROR 2022/11/03 09:09:45 error: write tcp 192.168.0.71:35554->192.168.0.251:1883: write: connection reset by peer
[mqtt  ] ERROR 2022/11/03 09:09:45 tcp://mqtt.home.lan:1883 connection lost: write tcp 192.168.0.71:35554->192.168.0.251:1883: write: connection reset by peer

### What type of operating system are you running?

Docker container

### Version

evcc version 0.105.2 (ad0591e2)

andig, 2022-11-03:
> After a downtime of the mqtt broker evcc doesn't reconnect to the mqtt broker

Is there any indication of that? We *are* already reconnecting.

christoph-ng, 2022-11-03:
Sorry was my mistake, I'm just querrying some values which are gone after the restart of the mqtt broker.
The values are not retained, so they are shown again after they are published again.

closed 2022-11-03 by christoph-ng