evcc-io/evcc #5035: OpenHab Binding fails after update to 0.106

pauxus, 2022-11-03:
### Describe the bug

States are not correctly read in Openhab after update to 0.106 (from 0.105.2).

State json is correctly read via api, but resolving fails. Did the output of api/state change between 0.105.2 and 0.106?

I will create a parallel issue in openhab

### Steps to reproduce

- Use openhab binding for evcc
- Upgrade from easier version to 0.106
- Thing in openhab goes offline (failed to get state)
- Log in openhab online shows failure message and the state json result

### Configuration details

irrelevant

### Log details

``` 
2022-11-03 09:18:37.663 [DEBUG] [ab.binding.evcc.internal.EvccHandler] - Running refresh job ...
2022-11-03 09:18:37.666 [TRACE] [ab.binding.evcc.internal.api.EvccAPI] - GET http://localhost:7070/api/state - {"result":{"auth":{"vehicles":null},"batteryConfigured":true,"batteryPower":-4569,"batterySoC":31,"bufferSoC":100,"currency":"EUR","gridConfigured":true,"gridPower":-87,"homePower":1474,"loadpoints":[{"chargeCurrent":10.365217391304347,"chargeCurrents":[11,0,0],"chargeDuration":2663000000000,"chargePower":2400,"chargeRemainingDuration":26449460678210,"chargeRemainingEnergy":7886.888888888886,"chargeTotalImport":14.724,"chargedEnergy":1242.000000000001,"charging":true,"climater":"off","connected":true,"connectedDuration":0,"enabled":true,"maxCurrent":16,"minCurrent":6,"minSoC":0,"mode":"minpv","phaseAction":"inactive","phaseRemaining":0,"phasesActive":1,"phasesConfigured":null,"phasesEnabled":3,"pvAction":"inactive","pvRemaining":0,"targetEnergy":0,"targetSoC":80,"targetTime":null,"targetTimeProjectedStart":null,"title":"Garage","vehicleCapacity":10.4,"vehicleDetectionActive":false,"vehicleFeatureOffline":false,"vehicleOdometer":860,"vehiclePresent":true,"vehicleRange":0,"vehicleSoC":11,"vehicleTargetSoC":0,"vehicleTitle":"Multivan"},{"chargeCurrent":0,"chargeCurrents":[0,0,0],"chargeDuration":78000000000,"chargePower":0,"chargeRemainingDuration":-1,"chargeTotalImport":5.648,"chargedEnergy":0,"charging":false,"climater":"off","connected":true,"connectedDuration":0,"enabled":false,"maxCurrent":16,"minCurrent":6,"minSoC":0,"mode":"pv","phaseAction":"inactive","phaseRemaining":0,"phasesActive":3,"phasesConfigured":null,"phasesEnabled":3,"pvAction":"inactive","pvRemaining":0,"targetEnergy":0,"targetSoC":80,"targetTime":null,"targetTimeProjectedStart":null,"title":"Haustür","vehicleCapacity":46,"vehicleDetectionActive":false,"vehicleFeatureOffline":false,"vehicleOdometer":1873.2,"vehiclePresent":true,"vehicleRange":278,"vehicleSoC":86,"vehicleTargetSoC":0,"vehicleTitle":"e-Corsa"}],"prioritySoC":100,"pvConfigured":true,"pvPower":8530,"residualPower":100,"savingsAmount":0.27292077269657483,"savingsEffectivePrice":0.08038528297380132,"savingsGridCharged":0.0021763677842174885,"savingsSelfConsumptionCharged":1.2405489668026124,"savingsSelfConsumptionPercent":99.82487137554486,"savingsSince":1667394989,"savingsTotalCharged":1.2427253345868299,"siteTitle":"BiBo","tariffFeedIn":0.08,"tariffGrid":0.3,"vehicles":["e-Corsa","Multivan"]}}

2022-11-03 09:18:37.667 [DEBUG] [ab.binding.evcc.internal.EvccHandler] - Failed to get state

```

### What type of operating system are you running?

Linux

### Version

0.106

pauxus, 2022-11-03:
The problem seems to be the change: #4875 

Workaround: Don't use floats for vehicle capacity. 

The rest should probably be done in the binding.

emcepe, 2022-11-03:
New is chargeTotalImport

pauxus, 2022-11-03:
Yes, but gson ignores unknown values, so that is not the problem here. Thus new fields are backward compatible. Problem is the incompatible change from int to float.

My problem here was that I had already a float in my configuration, which evcc silently accepted, but rounded down. 

The Openhab binding on the other hand didn't like "10.4" for a `long`

andig, 2022-11-03:
/cc @florian-h05 

Kein evcc Thema ;)

closed 2022-11-03 by andig

florian-h05, 2022-11-03:
@andig Danke für den Ping!

@pauxus I found the issue on the addons repo, will continue the discussion there.

andig, 2022-11-03:
@pauxus auf keinen Fall. Wir können keine Downstreamabhängigkeiten berücksichtigen. Wir stellen die weder zur Verfügung, noch können wir sie supporten. Wenn das jemand machen möchte bitte auf Slack melden.

pauxus, 2022-11-03:
@andig gutes Argument. Ist halt die Frage, wo die beste Stelle ist, um so kleine Infos unterzubringen, FAQ oder so.