evcc-io/evcc #5042: Loading statistics are not saved

JanDragon, 2022-11-03:
### Describe the bug

The statistics that can be retrieved via the button at the bottom right are not saved. Although the date is the one where I have made the last update and what should be correct, however, the data have all been set to 0.

[image]
[image]
[image]

### Steps to reproduce

Load a vehicle via photovoltaic and then update your Evcc version. The data of the statistics are not displayed.

### Configuration details

https://hastebin.com/kilewopuqa.yaml

### Log details

[main  ] INFO 2022/11/03 23:01:08 evcc 0.106.1
[main  ] INFO 2022/11/03 23:01:08 using config file: /etc/evcc.yaml
[main  ] WARN 2022/11/03 23:01:08 `uri` is deprecated and will be ignored. Use `network` instead.
[main  ] INFO 2022/11/03 23:01:08 listening at :7070
[site  ] INFO 2022/11/03 23:01:10 site config:
[site  ] INFO 2022/11/03 23:01:10   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2022/11/03 23:01:10     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/11/03 23:01:10     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/11/03 23:01:10     battery 1: power ✓ energy ✗ currents ✗ soc ✓
[lp-1  ] INFO 2022/11/03 23:01:10 loadpoint 1:
[lp-1  ] INFO 2022/11/03 23:01:10   mode:        pv
[lp-1  ] INFO 2022/11/03 23:01:10   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2022/11/03 23:01:10   meters:      charge ✓
[lp-1  ] INFO 2022/11/03 23:01:10     charge:    power ✓ energy ✗ currents ✓
[lp-2  ] INFO 2022/11/03 23:01:10 loadpoint 2:
[lp-2  ] INFO 2022/11/03 23:01:10   mode:        pv
[lp-2  ] INFO 2022/11/03 23:01:10   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-2  ] INFO 2022/11/03 23:01:10   meters:      charge ✓
[lp-2  ] INFO 2022/11/03 23:01:10     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] DEBUG 2022/11/03 23:01:10 phase timer inactive
[lp-1  ] DEBUG 2022/11/03 23:01:10 pv timer inactive
[lp-1  ] ERROR 2022/11/03 23:01:13 charger: Get "http://192.168.0.57/status": dial tcp 192.168.0.57:80: connect: no route to host
[lp-2  ] DEBUG 2022/11/03 23:01:13 phase timer inactive
[lp-2  ] DEBUG 2022/11/03 23:01:13 pv timer inactive
[server] FATAL 2022/11/03 23:01:13 unlinkat /tmp/evcc: operation not permitted
[main  ] FATAL 2022/11/03 23:01:13 listen tcp :7070: bind: address already in use

### What type of operating system are you running?

Linux

### Version

evcc version 0.106.1

naltatis, 2022-11-04:
Which version did you update from?

andig, 2022-11-04:
> [main ] FATAL 2022/11/03 23:01:13 listen tcp :7070: bind: address already in use

Dein evcc läuft 2x, das geht nicht...

StevieC121176, 2022-11-04:
Das gleiche passiert bei mir auch, allerdings sind die Daten sobald ein Fahrzeug geladen wird wieder da.

chkatz, 2022-11-04:
Selbiges Problem bei mir. 
Nach einem Update (zuletzt heute auf 0.106.1) ist die Statistik erstmal leer. Erst nachdem ein Ladevorgang gestartet wird, werden die Daten wieder geladen.

naltatis, 2022-11-04:
Ok, Problem gefunden. In der UI/API werden die Daten nur aktualisiert wenn eine Ladevorgang aktiv ist. Ich bau einen Fix, dass wir das zukünftig auch einmal beim hochfahren tun. Ist also kein Problem der Datenspeicherung (da geht nichts verloren) sondern nur ein Anzeigethema.

emcepe, 2022-11-04:
FYI: via MQTT sind die Werte in dem Fall auch alle 0.

andig, 2022-11-04:
Welche Werte? Statistik ist egtl. nicht für MQTT designed.

emcepe, 2022-11-04:
savingsAmount
savingsEffectivePrice
savingsGridCharged
savingsSelfConsumptionCharged
savingsSelfConsumptionPercent
savingsSince
savingsTotalCharged

closed 2022-11-04 by andig

andig, 2022-11-04:
Ah, sorry- ich war gedanklich bei den Ladesessions.

PR #5048 (2022-11-04): Publish savings data on startup
Previously savings data were only updated/published when charging is active. With the new persistence feature we also publish these data once on startup.

fixes #5042