evcc-io/evcc #5055: OCPP: Improper handling of EVDisconnected

tarelda, 2022-11-05:
### Describe the bug

EVCC ignores fact that EV disconnected and tries to disable disabled charger. Also spams charger with RemoteStartTransaction requests. 

### Steps to reproduce

1. Start charging your EV
2. Stop charging on your EV **without touching EVCC**.

### Configuration details

chargers:
  - name: carport-wallbox
    type: template
    template: ocpp
    timeout: 8m

loadpoints:
  - title: Wiata
    charger: carport-wallbox
    mode: "off"
    phases: 0
    minCurrent: 6
    maxCurrent: 16

### Log details

[site  ] DEBUG 2022/11/05 12:05:28 ----
[lp-1  ] DEBUG 2022/11/05 12:05:28 charge power: 0W
[site  ] DEBUG 2022/11/05 12:05:29 pv power: 0W
[site  ] DEBUG 2022/11/05 12:05:30 grid power: 709W
[ocpp  ] TRACE 2022/11/05 12:05:32 *core.StatusNotificationRequest: &{ConnectorId:1 ErrorCode:NoError Info: Status:Preparing Timestamp:<nil> VendorId: VendorErrorCode:}
[site  ] DEBUG 2022/11/05 12:05:33 grid currents: [2.6 0.68 0.66]A
[site  ] DEBUG 2022/11/05 12:05:34 site power: -471W
[lp-1  ] DEBUG 2022/11/05 12:05:34 charger status: B
[lp-1  ] INFO 2022/11/05 12:05:34 car connected
[lp-1  ] DEBUG 2022/11/05 12:05:34 vehicle api refresh
[lp-1  ] DEBUG 2022/11/05 12:05:34 pv timer elapse
[lp-1  ] DEBUG 2022/11/05 12:05:34 pv timer inactive
[main  ] ERROR 2022/11/05 12:05:34 vehicle status: 408 Request Timeout
[lp-1  ] DEBUG 2022/11/05 12:05:34 available power 471W < 4140W min 3p threshold
[lp-1  ] DEBUG 2022/11/05 12:05:34 start phase scale1p timer
[lp-1  ] DEBUG 2022/11/05 12:05:34 phase scale1p in 3m0s
[lp-1  ] DEBUG 2022/11/05 12:05:34 pv charge current: 0.683A = 0A + 0.683A (-471W @ 3p)
[lp-1  ] DEBUG 2022/11/05 12:05:34 max charge current: 6A
[ocpp  ] TRACE 2022/11/05 12:05:34 *core.RemoteStartTransactionConfirmation: &{Status:Accepted}
[lp-1  ] DEBUG 2022/11/05 12:05:34 charger enable
[lp-1  ] DEBUG 2022/11/05 12:05:34 wake-up timer: start
[site  ] DEBUG 2022/11/05 12:05:38 ----
[lp-1  ] DEBUG 2022/11/05 12:05:38 charge power: 0W
[ocpp  ] TRACE 2022/11/05 12:05:39 *core.StatusNotificationRequest: &{ConnectorId:1 ErrorCode:NoError Info: Status:Charging Timestamp:<nil> VendorId: VendorErrorCode:}
[ocpp  ] TRACE 2022/11/05 12:05:39 *core.StatusNotificationRequest: &{ConnectorId:1 ErrorCode:NoError Info:Charging Status:Charging Timestamp:<nil> VendorId: VendorErrorCode:}
[ocpp  ] TRACE 2022/11/05 12:05:39 *core.StartTransactionRequest: &{ConnectorId:1 IdTag:ACE0374641 MeterStart:197203 ReservationId:<nil> Timestamp:2022-11-05 12:05:37 +0100 CET}
[site  ] DEBUG 2022/11/05 12:05:40 pv power: 0W
[ocpp  ] TRACE 2022/11/05 12:05:41 *core.MeterValuesRequest: &{ConnectorId:1 TransactionId:0xc000629200 MeterValue:[{Timestamp:2022-11-05 12:05:38 +0100 CET SampledValue:[{Value:197.203 Context:Transaction.Begin Format: Measurand: Phase: Location: Unit:kWh}]}]}
[ocpp  ] TRACE 2022/11/05 12:05:41 TransactionID: 824640180736 
[site  ] DEBUG 2022/11/05 12:05:42 grid power: 834W
[ocpp  ] TRACE 2022/11/05 12:05:43 *core.MeterValuesRequest: &{ConnectorId:1 TransactionId:0xc00092cb98 MeterValue:[{Timestamp:2022-11-05 12:05:40 +0100 CET SampledValue:[{Value:197203.000 Context:Sample.Periodic Format: Measurand:Energy.Active.Import.Register Phase: Location:Outlet Unit:Wh} {Value:0.258 Context:Sample.Periodic Format: Measurand:Current.Import Phase:L1 Location:Outlet Unit:A} {Value:0.273 Context:Sample.Periodic Format: Measurand:Current.Import Phase:L2 Location:Outlet Unit:A} {Value:0.272 Context:Sample.Periodic Format: Measurand:Current.Import Phase:L3 Location:Outlet Unit:A}]}]}
[ocpp  ] TRACE 2022/11/05 12:05:43 TransactionID: 824643341208 
[site  ] DEBUG 2022/11/05 12:05:46 grid currents: [3.21 0.9 0.77]A
[site  ] DEBUG 2022/11/05 12:05:47 site power: -346W
[lp-1  ] DEBUG 2022/11/05 12:05:47 charger status: C
[lp-1  ] INFO 2022/11/05 12:05:47 start charging ->
[main  ] ERROR 2022/11/05 12:05:47 vehicle status: 408 Request Timeout
[lp-1  ] DEBUG 2022/11/05 12:05:47 available power 346W < 4140W min 3p threshold
[lp-1  ] DEBUG 2022/11/05 12:05:47 phase scale1p in 2m47s
[lp-1  ] DEBUG 2022/11/05 12:05:47 pv charge current: 6.5A = 6A + 0.501A (-346W @ 3p)
[lp-1  ] DEBUG 2022/11/05 12:05:47 pv timer reset
[lp-1  ] DEBUG 2022/11/05 12:05:47 pv timer inactive
[ocpp  ] TRACE 2022/11/05 12:05:47 update period with phases: 0, current: 6.501449
[... 355 lines cut ...]
[site  ] DEBUG 2022/11/05 12:09:45 site power: 598W
[lp-1  ] DEBUG 2022/11/05 12:09:45 charger status: B
[lp-1  ] DEBUG 2022/11/05 12:09:45 vehicle api refresh
[main  ] DEBUG 2022/11/05 12:09:46 vehicle status: A (Cichowóz)
[lp-1  ] WARN 2022/11/05 12:09:46 charger out of sync: expected enabled, got disabled
[ocpp  ] TRACE 2022/11/05 12:09:46 *core.RemoteStartTransactionConfirmation: &{Status:Accepted}
[lp-1  ] DEBUG 2022/11/05 12:09:46 available power -598W < 4140W min 3p threshold
[lp-1  ] DEBUG 2022/11/05 12:09:46 phase scale1p in 0s
[lp-1  ] DEBUG 2022/11/05 12:09:46 phase scale1p timer elapsed
[ocpp  ] TRACE 2022/11/05 12:09:46 *core.RemoteStopTransactionConfirmation: &{Status:Rejected}
[lp-1  ] ERROR 2022/11/05 12:09:46 charger disable: Rejected
[ocpp  ] TRACE 2022/11/05 12:09:46 *core.RemoteStopTransactionConfirmation: &{Status:Rejected}
[lp-1  ] ERROR 2022/11/05 12:09:46 charger disable: Rejected
[site  ] DEBUG 2022/11/05 12:09:48 ----
[lp-1  ] DEBUG 2022/11/05 12:09:48 charge power: 0W
[site  ] DEBUG 2022/11/05 12:09:49 pv power: 0W
[site  ] DEBUG 2022/11/05 12:09:50 grid power: 1801W

### What type of operating system are you running?

Docker container

### Version

0.106.0

andig, 2022-11-05:
What line are you referring to?

tarelda, 2022-11-05:
@andig This is not enhancement. This is bug caused by mismatch of OCPP status to EVCC status.

In OCPP, charge point when receives RemoteStartTransaction goes from state Available to Preparing. Preparing state doesn't mean that cable is plugged in and charger can be enabled.
Also (at least mine) charger:
 - accepts RemoteStartTransaction requests even if there is no vehicle currently plugged in 
 - does not send StatusNotificationRequest after StopTransactionRequest.

andig, 2022-11-05:
That is, we should tread Preparing as `A`?

tarelda, 2022-11-05:
> That is, we should treat Preparing as `A`?

It means both A and B, depending on transaction state.

andig, 2022-11-06:
> It means both A and B, depending on transaction state.

That's bad. Right now we have:

	case core.ChargePointStatusAvailable, // "Available"
		core.ChargePointStatusUnavailable: // "Unavailable"
		res = api.StatusA
	case
		core.ChargePointStatusPreparing,     // "Preparing"
		core.ChargePointStatusSuspendedEVSE, // "SuspendedEVSE"
		core.ChargePointStatusSuspendedEV,   // "SuspendedEV"
		core.ChargePointStatusFinishing:     // "Finishing"
		res = api.StatusB

Even using transaction state won't give additional information since

> accepts RemoteStartTransaction requests even if there is no vehicle currently plugged in

Any suggestion how to find out the "real" charger state in terms of evcc?

andig, 2022-11-06:
> EVCC ignores fact that EV disconnected and tries to disable disabled charger.

So how would we notice the disconnect reliably?

tarelda, 2022-11-06:
> > EVCC ignores fact that EV disconnected and tries to disable disabled charger.
> 
> So how would we notice the disconnect reliably?

Track StopTransactionsRequest transitions instead of discrete states. Also modify enable to prevent sending StartRemoteTransaction when transaction is running. Set up sanity check on timeout to request StatusNotification after StopTransaction.

I noticed two things in the log I provided:
- after receiving finishing state, there is no statusnotification with new state ,
- charger goes into preparing state because new transaction is initiated, because EVCC for some unknown reason tries to do it in following order: disable, disable, enable, disable. Then goes into infinite loop, because RemoteStopTransaction always fails when there is no ongoing transaction.

andig, 2022-11-06:
> because EVCC for some unknown reason tries to do it in following order: disable, disable, enable, disable

where in the log?

tarelda, 2022-11-06:
```
[ocpp  ] TRACE 2022/11/05 12:07:21 *core.StatusNotificationRequest: &{ConnectorId:1 ErrorCode:NoError Info: Status:Available Timestamp:<nil> VendorId: VendorErrorCode:}
[ocpp  ] TRACE 2022/11/05 12:07:22 *core.StopTransactionRequest: &{IdTag:ACE03746xx MeterStop:197267 Timestamp:2022-11-05 12:07:20 +0100 CET TransactionId:1 Reason:EVDisconnected TransactionData:[{Timestamp:2022-11-05 12:05:38 +0100 CET SampledValue:[{Value:197.203 Context:Transaction.Begin Format: Measurand: Phase: Location: Unit:kWh}]} {Timestamp:2022-11-05 12:07:20 +0100 CET SampledValue:[{Value:197.267 Context:Transaction.End Format: Measurand: Phase: Location: Unit:kWh}]}]}
[site  ] DEBUG 2022/11/05 12:07:23 grid currents: [2.59 0.67 0.65]A
[site  ] DEBUG 2022/11/05 12:07:24 site power: -469W
[lp-1  ] DEBUG 2022/11/05 12:07:24 charger status: A
[lp-1  ] INFO 2022/11/05 12:07:24 car disconnected
[lp-1  ] INFO 2022/11/05 12:07:24 vehicle updated: Cichowóz -> unknown
[lp-1  ] WARN 2022/11/05 12:07:24 charger out of sync: expected enabled, got disabled
[ocpp  ] TRACE 2022/11/05 12:07:24 *core.RemoteStartTransactionConfirmation: &{Status:Accepted}
[lp-1  ] DEBUG 2022/11/05 12:07:24 charger disable: contactor delay 3m10s
[ocpp  ] TRACE 2022/11/05 12:07:24 *core.StatusNotificationRequest: &{ConnectorId:1 ErrorCode:NoError Info: Status:Preparing Timestamp:<nil> VendorId: VendorErrorCode:}
[site  ] DEBUG 2022/11/05 12:07:28 ----
```

Actually, I was wrong. There is StatusNotification with status Available. So my diagnosis is also wrong. It is simple race condition. Neither Enabled() nor Status() are checked before Enable(). Also what is worth noting is that Enabled() is based only on transaction presence.

Btw, why Status() and Enabled() are separated? 

```
[site  ] DEBUG 2022/11/05 12:08:28 ----
[lp-1  ] DEBUG 2022/11/05 12:08:28 charge power: 0W
[site  ] DEBUG 2022/11/05 12:08:30 pv power: 0W
[site  ] DEBUG 2022/11/05 12:08:31 grid power: 744W
[site  ] DEBUG 2022/11/05 12:08:34 grid currents: [2.79 0.63 0.66]A
[site  ] DEBUG 2022/11/05 12:08:35 site power: -436W
[lp-1  ] DEBUG 2022/11/05 12:08:35 charger status: B
[main  ] DEBUG 2022/11/05 12:08:35 vehicle status: A (Cichowóz)
[lp-1  ] WARN 2022/11/05 12:08:35 charger out of sync: expected enabled, got disabled
[ocpp  ] TRACE 2022/11/05 12:08:35 *core.RemoteStartTransactionConfirmation: &{Status:Accepted} //enable
[lp-1  ] DEBUG 2022/11/05 12:08:35 available power 436W < 4140W min 3p threshold
[lp-1  ] DEBUG 2022/11/05 12:08:35 phase scale1p in 0s
[lp-1  ] DEBUG 2022/11/05 12:08:35 phase scale1p timer elapsed
[ocpp  ] TRACE 2022/11/05 12:08:35 *core.RemoteStopTransactionConfirmation: &{Status:Rejected} // disable
[lp-1  ] ERROR 2022/11/05 12:08:35 charger disable: Rejected
[ocpp  ] TRACE 2022/11/05 12:08:35 *core.RemoteStopTransactionConfirmation: &{Status:Rejected} // disable
[lp-1  ] ERROR 2022/11/05 12:08:35 charger disable: Rejected
[site  ] DEBUG 2022/11/05 12:08:38 ----
```

Actually there is one more enable before and it starts with scale1p timer ending.

rtrbt, 2022-11-07:
While working on the OCPP implementation for the WARP charger (tfocpp), I've noticed the same issue. I've tested the following:
09:39:00: Started EVCC and tfocpp
09:40:00: Simulated plugging in a car (This changes the IEC state to B and tfocpp sends a StatusNotification.req with Status Preparing)
09:40:30: Set EVCC to charge immediately
09:41:00: Set the simulated IEC state to C i.e. charging
09.41:30: Simulated unplugging the car

Relevant part of my EVCC log:
```
[site  ] DEBUG 2022/11/07 09:41:23 ----
[lp-1  ] DEBUG 2022/11/07 09:41:23 charge power: 22080W
[script] DEBUG 2022/11/07 09:41:23 /bin/sh -c echo 10000: 10000
[site  ] DEBUG 2022/11/07 09:41:23 pv power: 10000W
[script] DEBUG 2022/11/07 09:41:23 /bin/sh -c echo -9000: -9000
[site  ] DEBUG 2022/11/07 09:41:23 grid power: -9000W
[site  ] DEBUG 2022/11/07 09:41:23 site power: -9000W
[lp-1  ] DEBUG 2022/11/07 09:41:23 charger status: C
[ocpp  ] TRACE 2022/11/07 09:41:31 *core.StopTransactionRequest: &{IdTag:evcc MeterStop:1234 Timestamp:2022-11-07 09:41:30 +0100 CET TransactionId:1 Reason: TransactionData:[]}
[ocpp  ] TRACE 2022/11/07 09:41:31 *core.StatusNotificationRequest: &{ConnectorId:1 ErrorCode:NoError Info: Status:Finishing Timestamp:2022-11-07 09:41:30 +0100 CET VendorId: VendorErrorCode:}
[ocpp  ] TRACE 2022/11/07 09:41:32 *core.StatusNotificationRequest: &{ConnectorId:1 ErrorCode:NoError Info: Status:Available Timestamp:2022-11-07 09:41:30 +0100 CET VendorId: VendorErrorCode:}
[site  ] DEBUG 2022/11/07 09:41:33 ----
[lp-1  ] DEBUG 2022/11/07 09:41:33 charge power: 22080W
[script] DEBUG 2022/11/07 09:41:33 /bin/sh -c echo 10000: 10000
[site  ] DEBUG 2022/11/07 09:41:33 pv power: 10000W
[script] DEBUG 2022/11/07 09:41:33 /bin/sh -c echo -9000: -9000
[site  ] DEBUG 2022/11/07 09:41:33 grid power: -9000W
[site  ] DEBUG 2022/11/07 09:41:33 site power: -9000W
[lp-1  ] DEBUG 2022/11/07 09:41:33 charger status: A
[lp-1  ] INFO 2022/11/07 09:41:33 stop charging <-
[lp-1  ] INFO 2022/11/07 09:41:33 car disconnected
[lp-1  ] DEBUG 2022/11/07 09:41:33 set charge mode: off
[lp-1  ] DEBUG 2022/11/07 09:41:33 pv timer elapse
[lp-1  ] DEBUG 2022/11/07 09:41:33 pv timer inactive
[lp-1  ] DEBUG 2022/11/07 09:41:33 set min current: 6
[lp-1  ] DEBUG 2022/11/07 09:41:33 set max current: 32
[lp-1  ] DEBUG 2022/11/07 09:41:33 set min soc: 0
[lp-1  ] DEBUG 2022/11/07 09:41:33 set target soc: 100
[lp-1  ] WARN 2022/11/07 09:41:33 charger out of sync: expected enabled, got disabled
[ocpp  ] TRACE 2022/11/07 09:41:34 *core.RemoteStartTransactionConfirmation: &{Status:Accepted}
[ocpp  ] TRACE 2022/11/07 09:41:34 *core.StatusNotificationRequest: &{ConnectorId:1 ErrorCode:NoError Info: Status:Preparing Timestamp:2022-11-07 09:41:33 +0100 CET VendorId: VendorErrorCode:}
[ocpp  ] TRACE 2022/11/07 09:41:35 *core.RemoteStopTransactionConfirmation: &{Status:Rejected}
[lp-1  ] ERROR 2022/11/07 09:41:35 charger disable: Rejected
[site  ] DEBUG 2022/11/07 09:41:35 ----
[lp-1  ] DEBUG 2022/11/07 09:41:35 charge power: 0W
[script] DEBUG 2022/11/07 09:41:35 /bin/sh -c echo 10000: 10000
[site  ] DEBUG 2022/11/07 09:41:35 pv power: 10000W
[script] DEBUG 2022/11/07 09:41:35 /bin/sh -c echo -9000: -9000
[site  ] DEBUG 2022/11/07 09:41:35 grid power: -9000W
[site  ] DEBUG 2022/11/07 09:41:35 site power: -9000W
[lp-1  ] DEBUG 2022/11/07 09:41:35 charger status: B
[lp-1  ] INFO 2022/11/07 09:41:35 car connected
[lp-1  ] DEBUG 2022/11/07 09:41:35 pv timer elapse
[lp-1  ] DEBUG 2022/11/07 09:41:35 pv timer inactive
[lp-1  ] WARN 2022/11/07 09:41:35 charger out of sync: expected enabled, got disabled
[ocpp  ] TRACE 2022/11/07 09:41:35 *core.RemoteStartTransactionConfirmation: &{Status:Accepted}
[ocpp  ] TRACE 2022/11/07 09:41:36 *core.RemoteStopTransactionConfirmation: &{Status:Rejected}
[lp-1  ] ERROR 2022/11/07 09:41:36 charger disable: Rejected
[site  ] DEBUG 2022/11/07 09:41:43 ----
[lp-1  ] DEBUG 2022/11/07 09:41:43 charge power: 0W
[script] DEBUG 2022/11/07 09:41:43 /bin/sh -c echo 10000: 10000
[site  ] DEBUG 2022/11/07 09:41:43 pv power: 10000W
[script] DEBUG 2022/11/07 09:41:43 /bin/sh -c echo -9000: -9000
[site  ] DEBUG 2022/11/07 09:41:43 grid power: -9000W
[site  ] DEBUG 2022/11/07 09:41:43 site power: -9000W
[lp-1  ] DEBUG 2022/11/07 09:41:43 charger status: B
[lp-1  ] WARN 2022/11/07 09:41:43 charger out of sync: expected enabled, got disabled
[ocpp  ] TRACE 2022/11/07 09:41:44 *core.RemoteStartTransactionConfirmation: &{Status:Accepted}
[ocpp  ] TRACE 2022/11/07 09:41:44 *core.RemoteStopTransactionConfirmation: &{Status:Rejected}
[lp-1  ] ERROR 2022/11/07 09:41:44 charger disable: Rejected
[site  ] DEBUG 2022/11/07 09:41:53 ----
[... 51 lines cut ...]
[lp-1  ] DEBUG 2022/11/07 09:41:33 charger status: A
[lp-1  ] INFO 2022/11/07 09:41:33 stop charging <-
[lp-1  ] INFO 2022/11/07 09:41:33 car disconnected
```
but later logs
```
[lp-1  ] WARN 2022/11/07 09:41:33 charger out of sync: expected enabled, got disabled
```
and probably tries to resync the charger state by sending a RemoteStartTransaction.req. This request is accepted by tfocpp, because, as the spec states:
>  the Charge Point SHALL reply with RemoteStartTransaction.conf and a status indicating whether it has
accepted the request and will attempt to start a transaction.
And it is allowed to attempt to start a transaction even if no cable is plugged in. Otherwise OCPP's ConnectionTimeout configuration would be useless. (Note the StatusTransition from IDLE to NO_PLUG. This is sent to EVCC as a StatusNotificati [... cut ...]
@andig If you want to reproduce this on your end, I can send you a statically linked linux executable of tfocpp (or you can try to build https://github.com/Tinkerforge/tfocpp yourself, however this is currently a bit complicated). You know how to reach me by e-mail ;)

(ocpp_traffic is a zipped pcap. Seems like Github does not allow uploading pcaps, but allows zips)

[ocpp.log](https://github.com/evcc-io/evcc/files/9950219/ocpp.log)
[evcc.log](https://github.com/evcc-io/evcc/files/9950221/evcc.log)
[ocpp_traffic.zip](https://github.com/evcc-io/evcc/files/9950225/ocpp_traffic.zip)

andig, 2022-11-07:
I have no idea how to proceed. From evcc side it's very simple:
- we need to know the charger status (A..C)
- we need to enabled/disable the charger and determine the enabled status

Everything else will just work. If there's any steps we can take in that direction for Ccpp happy to take a PR. This is *not* a problem of the Loadpoint, it's one of the Ocpp charger.

andig, 2022-11-07:
> Neither Enabled() nor Status() are checked before Enable(). 

They are.

> Also what is worth noting is that Enabled() is based only on transaction presence.

That is obviously an Ocpp problem. Better ideas needed.

> EVCC then repeats to send RemoteStartTransaction.reqs and RemoteStopTransaction.reqs forever.

Because it tries to sync the charger to the expected state.

> If you want to reproduce this on your end, I can send you a statically linked linux executable of tfocpp

We already have the cmd/ocpp test app. I'm conceptually lacking a approach how to continue, another app won't help there unfortunately.

rtrbt, 2022-11-07:
>we need to know the charger status (A..C)
>we need to enabled/disable the charger and determine the enabled status

This is difficult, as you can't easily tell apart state A and B: A StatusNotification "Preparing" is sent if
1. the cable is plugged in
2. and/or an NFC tag is seen or injected via RemoteStartTransaction.req (even if no cable is plugged in)

If I understand the OCPP spec correctly, you can completely control the charger via a SmartChargingProfile: If you set a ChargePointMaxProfile that allows a current of 0A, a charger would still start a transaction if the cable is plugged in and the charge is authorized. However no energy is transmitted. You can then update the charging profile if current is available. (Also you can completely unblock a charger with a ClearChargingProfile.req)

To determine the enabled status, you can use a GetCompositeSchedule.req. If only the ChargePointMaxProfile is installed, the reported limits are only caused by the profile.

If you control a charger in this way, you don't have to keep track of transactions at all.

tarelda, 2022-11-07:
> > we need to know the charger status (A..C)
> > we need to enabled/disable the charger and determine the enabled status
> 
> This is difficult, as you can't easily tell apart state A and B: A StatusNotification "Preparing" is sent if
> 
> 1. the cable is plugged in
> 2. and/or an NFC tag is seen or injected via RemoteStartTransaction.req (even if no cable is plugged in)
> 
> If I understand the OCPP spec correctly, you can completely control the charger via a SmartChargingProfile: If you set a ChargePointMaxProfile that allows a current of 0A, a charger would still start a transaction if the cable is plugged in and the charge is authorized. However no energy is transmitted. You can then update the charging profile when current is available. (Also you can completely unblock a charger with a ClearChargingProfile.req)
> 
> To determine the enabled status, you can use a GetCompositeSchedule.req. If only the ChargePointMaxProfile is installed, the reported limits are only caused by the profile.
> 
> If you control a charger in this way, you don't have to keep track of transactions at all.

Not all chargers support composite charging profiles. Also ChargepointMaxProfile doesn't phase setting (this actually make a lot of sense in chargepoint implementation standpoint). I mean just see #4946 .

I am confused by you stance here @andig . You explicitly deleted my workarounds for loadpoint behaviour...

rtrbt, 2022-11-07:
>Also ChargepointMaxProfile doesn't do phase setting (this actually make a lot of sense from chargepoint implementation standpoint).

I can't find any reference in the spec, that a phase limitation in a ChargePointMaxProfile may be ignored. Maybe this is a quirk with your charger? (If not, I'm really interested in this, as I've said, I'm currently implementing OCPP for the WARP charger)

> Not all chargers support composite charging profiles.

Technically they have to if they claim to support the SmartCharging feature profile.

tarelda, 2022-11-07:
> I can't find any reference in the spec, that a phase limitation in a ChargePointMaxProfile may be ignored. Maybe this is a quirk with your charger? (If not, I'm really interested in this, as I've [...]

There is no prohibition, but standard does not define how to resolve conflict between transaction profiles and chargepointmax profile. 

> > Not all chargers support composite charging profiles.
> 
> Technically they have to if they claim to support the SmartCharging feature profile.

Only figuratively. Because MaxStackingLevel 0 effectively means no composite charging profiles.

rtrbt, 2022-11-07:
> but standard does not define how to resolve conflict between transaction profiles and chargepointmax profile.

You are correct, nothing is said explicitly about conflicts of **phase** limits, only the power/current limits: (3.13.3):
>At any point in time, the available power or current in the Composite Schedule, which is the result of merging the
schedules of charging profiles ChargePointMaxProfile and TxDefaultProfile (or TxProfile), SHALL be less than or
equal to lowest value of available power or current in any of the merged schedules.

However I would assume that
>"This Composite Schedule is calculated by taking the minimum value for each time interval.

(directly above in 3.13.3) implies that the minimum number of phases over all profiles is to be used. So if we have one ChargePointMaxProfile that allows 8 A on two phases (3680 W) and a TxProfile that allows 3000 W, it sounds reasonable (but is not required by the spec as far as i can see) to charge with two phases and a current of ~ 6,5 A.

Interpreting the minimum to also apply to the phases supported by Errata 4.0 3.34:
>When the Charge Point receives a Charging Profile with 'numberPhases' higher then the currently used amount of phases or maximum amount of phases this Charge Point can use, The Charge Point SHALL use the 'limit' as is given, for the amount of phases that are/can be used for charging. The Charge Point SHALL NOT calculate a new 'limit' based on the amount of phases possible and given in the 'numberPhases' field.

So it should not be possible to exceed another current/power limit with a changed phase setting.

> Only figuratively. Because MaxStackingLevel 0 effectively means no composite charging profiles.

Hmm good point. With MaxStackingLevel 0 you can't install any ChargePointMaxProfiles and TxDefaultProfiles, but only a TxProfile via RemoteStartTransaction.req. So my idea of controlling the charger with a ChargePointMaxProfile does not work in this case. Are there any chargers that set MaxStackingLevel to 0?

tarelda, 2022-11-09:
> > but standard does not define how to resolve conflict between transaction profiles and chargepointmax profile.
> 
> You are correct, nothing is said explicitly about conflicts of **phase** limits, only the power/current limits: (3.13.3):
> 
> > At any point in time, the available power or current in the Composite Schedule, which is the result of merging the
> > schedules of charging profiles ChargePointMaxProfile and TxDefaultProfile (or TxProfile), SHALL be less than or
> > equal to lowest value of available power or current in any of the merged schedules.
> 
> However I would assume that
> 
> > "This Composite Schedule is calculated by taking the minimum value for each time interval.
> 
> (directly above in 3.13.3) implies that the minimum number of phases over all profiles is to be used. So if we have one ChargePointMaxProfile that allows 8 A on two phases (3680 W) and a TxProfile that allows 3000 W, it sounds reasonable (but is not required by the spec as far as i can see) to charge with two phases and a current of ~ 6,5 A.
> 
> Interpreting the minimum to also apply to the phases supported by Errata 4.0 3.34:
> 
> > When the Charge Point receives a Charging Profile with 'numberPhases' higher then the currently used amount of phases or maximum amount of phases this Charge Point can use, The Charge Point SHALL use the 'limit' as is given, for the amount of phases that are/can be used for charging. The Charge Point SHALL NOT calculate a new 'limit' based on the amount of phases possible and given in the 'numberPhases' field.
> 
> So it should not be possible to exceed another current/power limit with a changed phase setting. **Edit:** Note that this leaves the path of "what happens in the real world" and is only my interpretation of the spec.

Offtopic: Spec vs real world sometimes doesn't align very well. I think developers that wrote my charger's firmware had simple thought process. If spec doesn't exactly specify scenario how to behave it is better to ignore it since Tx profiles allow phase number control per connector which I think is the most useful scenario.

> Hmm good point. With MaxStackingLevel 0 you can't install any ChargePointMaxProfiles and TxDefaultProfiles, but only a TxProfile via RemoteStartTransaction.req. So my idea of controlling the charger [...]

Yes, at least two as I recall: ABB Terra AC (firmwares 1.5.x < (they pulled 1.6.x series that supports it at least three times from production xD)) and Wallbox Commander.

tarelda, 2022-11-09:
> > I am confused by you stance here @andig . You explicitly deleted my workarounds for loadpoint behaviour...
> 
> @tarelda Little stance really if any. I'm happy for discussions, suggestions and PRs. If I understand them and they are beneficial PRs will get merged. If I deleted something then probably because it wasn't clear or had negative side effects. As much as I'd like to make OCPP a first class evcc client, as much is it my obligation to make sure that other users are not negatively impacted. You'll also remember that there were a number of misunderstandings (i.e. phase switching seems to work just fine as-is).
> 
> That said: happy to look at any PR. To make it simple to digest, concentrate on one thing to fix instead of a broad discussion about potential aspects. Those are much harder to follow for me.

Correct me if I am wrong: 
1. EVCC expects charger API to be synchronous (every function call has to be fully finished if it returns). 
2. OCPP object instance is implemented as singleton to support multiple OCPP chargers in EVCC.

Unfortunately, this one won't be easy and concentrated PR. Do you want me to open PR and push my WIP draft, so everyone interested can comment on it? I find code to be more informative about subject than endless discussions.

andig, 2022-11-10:
> EVCC expects charger API to be synchronous (every function call has to be fully finished if it returns).

Not quite. I would say it requires API to be idempotent. Most boxes don't finish in time, especially 1p3p is notoriously slow. If loadpoint encounters non-matching status due to slow async execution it will just re-trigger the needed action once more and issue a warning.

> OCPP object instance is implemented as singleton to support multiple OCPP chargers in EVCC.

No. The central system (CS) is a singleton and has to be, each charge point (CP) is its own instance.

> Unfortunately, this one won't be easy and concentrated PR. Do you want me to open PR and push my WIP draft, so everyone interested can comment on it? I find code to be more informative about subject than endless discussions.

Sure. I'll still need to understand it...

andig, 2022-11-11:
Then lets close the discussion here- looking forward to PR!

andig, 2022-11-11:
@tarelda would still like to welcome you on Slack for quicker exchange!

closed 2022-11-11 by andig