evcc-io/evcc #5060: Heidelberg: disable/update watchdog

andig, 2022-11-05:
Ich konnte das Problem beheben in dem ich das Watchdog Timeout in der Heidelberg im Modbus Register 257 auf 0 also deaktiviert habe:

mbpoll -m rtu -0 -t 4 -B -r 257 /dev/ttyUSB0 0

Ggf. muss vorher Programm mpoll installiert werden:
sudo apt install mpoll

_Originally posted by @orbus64 in https://github.com/evcc-io/evcc/discussions/5054#discussioncomment-4057789_

andig, 2022-11-05:
@premultiply das sollten wir Charger-seitig schon tun?

premultiply, 2022-11-05:
Das ist kein Problem sondern nur ein Missverständnis.
Sobald evcc läuft und korrekt kommuniziert hört das Blinken nach ca. 10 Sekunden auf.
Das Blinken ist somit ein Indikator dass etwas mit der Steuerung nicht stimmt.
Natürlich kann man das abstellen, ist aber eigentlich sinnvoll es so zu lassen wie default.
Wer mag kann es natürlich umkonfigurieren.

closed 2022-11-05 by premultiply

andig, 2022-11-05:
Danke Dir 😘

orbus64, 2022-11-06:
> Das ist kein Problem sondern nur ein Missverständnis.
> Sobald evcc läuft und korrekt kommuniziert hört das Blinken nach ca. 10 Sekunden auf.
> Das Blinken ist somit ein Indikator dass etwas mit der [...]

Also bei mir war das auch nach >10 Sek nicht funktional...
Habe zwei Heidelberg abwechseln versucht um Fehler auszuschließen, auch extra Waveshare RS485 USB Stick besorgt...
War schon kurz davor EVCC aufzugeben als ich durch Zufall in den Registern der Heidelberg den Wachdog Time Out gesehen hatte.

Sobald die Wallbox auf Follower Modus aktiviert wurde kam es zu dem Problem... 
Prinzipiell ist der Watchdog seitens der Wallbox auf jeden Fall sinnvoll... 

Vielleicht gibt es auch unterschiedliche Heidelberg Varianten im Feld...!??
Wenn Ihr möchtet teste ich noch einmal mit verschiedenen Timeouts...

premultiply, 2022-11-06:
Sobald die Wallbox auf Follower-Modus konfiguriert ist erwartet sie regelmäßige Kommunikation vom Master (hier evcc).
Bleibt diese aus wechselt sie in den Fallback-Modus und fängt an zu blinken um den Anwender darauf aufmerksam zu machen dass der Master fehlt.
Kommuniziert der Master wieder regelmäßig hört nach ca. 10 Sekunden das Geblinke auf und alles funktioniert wieder korrekt.

Es funktioniert daher wie erwartet.

Klappt die Kommunikation nicht durchgehend, z. B. weil noch irgendwas rumgekabelt, rumkonfiguriert oder getestet wird blinkt es halt. Alles normal.

premultiply, 2022-11-06:
Voraussetzung ist natürlich dass das Regelintervall von evcc kleiner als das konfigurierte Timeout der Box ist!
Mit der Standardeinstellung evcc-Intervall 10s und Wallbox-Timeout 15s sollte es keine Probleme geben wenn die Kommunikation richtig funktioniert.

orbus64, 2022-11-23:
Hallo zusammen,

danke für die ausführliche Erklärung. Ich bin auch der Meinung das der Timeout in der Wallbox an sich sinnvoll ist, so kann man direkt visuell nachvollziehen das es ein Problem gibt.

Ich habe die Vermutung das dies mit den 2 RCT Wechselrichtern zusammenhängt wenn einer davon nachts aus ist, vielleicht dauert die Kommunikation dann mit der Heidelberg zu lange und die Wallbox meldet dann den Fehler. 

Heute nachdem der 2. Wechselrichter ausging konnte ich beobachten wie die Wallbox den Fehler angezeigt hatte. Timeout in der Heidelberg immer noch auf 0. Nach einiger Zeit war der Fehler dann weg. 

Würde es helfen die neue Proxy Modbus Funktion für die beiden Wechselrichter im EVCC einzurichten um damit dem Problem zu begegnen? (Mehrere Modbus Verbindungen auf ein Gerät #5094)