evcc-io/evcc #5063: Cannot create charger 'template': template not found: tinkerforge-warp since 0.106.2

erikgieseler, 2022-11-05:
### Describe the bug

Hallo,
ich habe heute das Update auf die Version 0.106.2 gemacht.
Leider startet evcc danach nicht mehr.
Im Log wird folgende Fehlermeldung angezeigt:
[main  ] FATAL 2022/11/05 19:28:07 cannot create charger 'wallbox_rechts': cannot create charger 'template': template not found: tinkerforge-warp.
Nach einem Rollback auf die Version 0.105.2 hat es wieder ganz normal funktioniert.
Vielen Dank voarb für eure Hilfe !

### Steps to reproduce

Siehe oben

### Configuration details

Ist beigefügt

### Log details

[main  ] INFO 2022/11/05 19:47:05 evcc 0.106.2
[main  ] INFO 2022/11/05 19:47:05 using config file: /etc/evcc.yaml
[main  ] INFO 2022/11/05 19:47:05 starting ui and api at :7070
[mqtt  ] INFO 2022/11/05 19:47:05 connecting evcc at tcp://192.168.180.253:1883
[mqtt  ] DEBUG 2022/11/05 19:47:05 tcp://192.168.180.253:1883 connected
[main  ] FATAL 2022/11/05 19:47:05 cannot create charger 'wallbox_links': cannot create charger 'template': template not found: tinkerforge-warp
[main  ] FATAL 2022/11/05 19:47:05 will attempt restart in: 5m0s

### What type of operating system are you running?

Docker container

### Version

0.106.2

erikgieseler, 2022-11-05:
[evcc.yaml.txt](https://github.com/evcc-io/evcc/files/9944513/evcc.yaml.txt)

erikgieseler, 2022-11-05:
Ganz merkwürdig, nachdem ich den Container komplett gelöscht und nochmal neu erstellt hatte, funktionierte auch die 0.106.2.
Hat sich also erledigt :-)

erikgieseler, 2022-11-05:
Nach einem Neustart des Containers ist der Fehler wieder da !
Logdatei ist beigefügt.
[_evcc_logs.txt](https://github.com/evcc-io/evcc/files/9944587/_evcc_logs.txt)

andig, 2022-11-06:
@erikgieseler nutzt Du die Releaseversion oder das Nightly?

erikgieseler, 2022-11-06:
Ich verwende in Docker Compose die "andig/evcc:latest" Version.

andig, 2022-11-06:
Fixed in https://github.com/evcc-io/evcc/commit/b386fdd10f0a664f5797da412f82a446e703098b

closed 2022-11-06 by andig

andig, 2022-11-06:
@erikgieseler das ist immer noch komisch. Den Fehler kann ich nachvollziehen, aber nur in der nightly Version auf Basis des `master` Branches. In der .2 sollte der Code überhaupt nicht drin sein. Sehr verwirrend, das Du den Fehler bekommst.

andig, 2022-11-06:
Ich hab die 0.106.3 nochmal "released". Das ist zwar nicht best practice, aber tatsächlich hat das Docker Release fälschlicherweise Code aus der aktuellen Entwicklerversion enthalten.

erikgieseler, 2022-11-06:
Damit startet es jetzt ganz normal. Danke !

andig, 2022-11-06:
Wir freuen uns über zufriedene Sponsoren ;)