evcc-io/evcc #5073: SoC immer anzeigen

TobiasSt78, 2022-11-06:
Der Akkustand des Hausakkus ist fast so wichtig wie der des Autos. Er wird zwar per Symbol idealisiert angezeigt, ich fände es darüber Hinaus cool, eine %-Angabe neben dem Akku Symbol direkt zu haben und nicht immer das Detailmenü öffnen zu müssen. Ungefähr so, wie man es vom iPhone kennt…

Dann hätte man wieder alles auf einen Blick. 

Das UI und die UX von EVCC sind genial!!! Weiter so.

andig, 2022-11-06:
Mhhhhm. Das ist eine weitere Anfrage aus der Kategorie „mehr Daten“. Wir wollen uns ja aufs Laden konzentrieren und eher weniger als mehr anzeigen. Wofür ist der Batteriestand relevant?

TobiasSt78, 2022-11-13:
Lean Design ist ein super Ansatz! Kann ich voll nachvollziehen.

Der Hausbatteriestand ist fürs Debuggen der Automatik recht wichtig und davon hängt in der "dunklen Jahreszeit" ab, ob ich übersteuere - oder nicht. 

Ehrlich gesagt funktioniert EVCC so viel besser als alle anderen Views die ich für meine 2 Kostal WR habe, weshalb ich es auch nutze, um einen schnellen Überblick über meine PV Anlage zu bekommen. Das schlanke Design und der Farbcode ist perfekt für den schnellen Überblick. 
Spannend ist morgens allerdings auch der Blick auf den Haus-Akku: Habe ich noch 5% oder eher 15% übrig...

naltatis, 2022-11-13:
Danke für das Lob. Wir haben dieses Thema, also was zeigen wir im zugeklappten Zustand an, damals lange diskutiert. Gab auch einige Stimmen denen die PV Produktions in kW super wichtig waren und die das am Sonnensymbol stehen haben wollten. Wir haben uns dafür entschieden wirklich nur die Verbrauchsarten (Eigenverbrauch, Einspeisung und Netzbezug) zu labeln. Das sind ja für die Regelautomatik die wichtigen Werte.

Beim Hausakku haben wir zusätzlich eine Füllstandsvisualisierung über das Icon gemacht. Da gibt es fünf Zustände (leer, viertel, halb, dreiviertel, voll) die du am Icon ablesen kannst. Das ist zugegebenermaßen subtil und du kannst daran nicht erkennen ob du 5 oder 15% im Speicher hast (Abstufung in 20% Schritten). Aber für allgemeine Fragen wir "ist der Speicher mind. halbvoll?" sollte das ausreichen. Daher würde ich auch weiter davon absehen im geschlossenen Zustand mehr anzuzeigen. Auch der "Expert-Modus" der im geschlossenen Zustand ganz viel Details anzeigt war mal im Gespräch, sehe ich aber aktuell auch nicht als notwendig.

Wenn es ums Debuggen geht halt ich "aufklappen" für zumutbar :D Wenn es wirklich Szenarien gibt bei denen diese Details wichtig sind, also bspw. unterschiedliches Ladeverhalten abhängig vom Akkustand (bufferSoC, prioritySoC) würde ich alternativ lieber versuchen eine Antwort für genau diese Fragen zu finden. Das kann vielleicht  in textlicher Form geschehen (bspw. analog zu "Überschuss verfügbar, starte Ladung in 50s" oder durch einen anderen Indikator). Dafür müssen wir aber die Fragestellung aus Nutzersicht klar formulieren können.

VolkerK62, 2022-11-14:
>Beim Hausakku haben wir zusätzlich eine Füllstandsvisualisierung über das Icon gemacht. Da gibt es fünf Zustände (leer, viertel, halb, dreiviertel, voll) die du am Icon ablesen kannst. 

Cool, ist mir bisher noch nie aufgefallen :-)