evcc-io/evcc #5074: Weitere Kategorie mit Symbol

TobiasSt78, 2022-11-06:
Es wird bisher zwischen Haus, Auto und Batterie bei den Verbrauchern unterschieden. Die Smart-Meter von z.B. Heizlüftern kann man mit einbinden, sie werden aber entweder dem „Auto“ oder „Haus“ zugeordnet.
Wäre echt cool, für „Sonstiges“ ein eigenes Symbol zu bekommen. Gern auch „S1“, „S2“,…

übrigens lassen sich Heizlüfter über den Shelly Plug sehr genial wie ein BEV über EVCC steuern. Durch höhere Delays hier, haben die Autos beim Laden dann trotzdem immer Vorrang. Also eine minimalistische HEM Funktionalität.

andig, 2022-11-06:
> übrigens lassen sich Heizlüfter über den Shelly Plug sehr genial wie ein BEV über EVCC steuern. Durch höhere Delays hier, haben die Autos beim Laden dann trotzdem immer Vorrang

das ist ein Irrtum. Im dynamischen Zustand gibt es sowas wie Vorrang nicht oder nur zufällig.

andig, 2022-11-06:
Ich glaube nicht, dass wir weitere Kategorien einführen wollen- auch Heizlüfter (die energetisch eine Katastrophe sind) stellen einfach Hausverbrauch dar.

/cc @premultiply

premultiply, 2022-11-06:
Mal abgesehen davon dass sich evcc ganz bewusst funktional auf die vergleichsweise komplexe Aufgabe "Laden von Elektrofahrzeugen" beschränkt, und nicht versucht irgendwie eine halb angegarte eierlegende Wollmilchsau zu sein, ist es halt dann einfach so, dass alles was man an evcc anbindet als Ladepunkt bzw. Fahrzeug betrachtet wird.

Für das übergreifende Energiemanagement von **sonstigen** einfachen oder progammbasierten Verbrauchern gibt es andere, u. U. auch prognosebasierte Lösungen am Markt wie z. B. den Sunny Home Manager von SMA und auch diverse Hausautomationslösungen.

Heizlüfter und Heizstäbe (oder andere auf reiner Widerstandsheizung basierenden Systeme) zu betreiben ist das genaue Gegenteil von verantwortlichem Umgang mit Energie und kein Szenario dass wir auch nur irgendwie unterstützen oder gutheißen möchten.

premultiply, 2022-11-06:
"Haus" ist bei evcc ein Synonym für "Rest" oder "Sonstige" und stellt mathematisch genau dies dar.
Hausverbrauch = Netzbezug + Batterieentladung - aktuelle Gesamtladeleistung

closed 2022-11-07 by andig

TobiasSt78, 2022-11-26:
Danke für die ausführlichen Antworten und kann ich soweit akzeptieren. 
Die Einstellung zu Heizlüftern teile ich grundsätzlich, wobei es halt die dumme Ausnahme gibt, dass in meinem Haus nur ein Raum leider über die Gasheizung nicht ausreichend warm wird oder die gesamte Vorlauftemerpatur hoch gestellt werden müsste, da er am Ende des Leitungsstrangs hängt und so erst das restliche Haus inkl. Flur sinnlos erwärmt wird. 
Da dies ausgerechnet ein Teenie-Zimmer ist, ist ein Heizlüfter durchaus eine energetisch halbwegs sinnvolle und wirtschaftliche Zwischenlösung - in diesem Sonderfall. 
Ja, Auslegungsfehler der Heizungsauslegung... eine Wärmepumpe wäre für nur einen Raum jedoch ein "Overkill" und Rest des Hauses ist maximal gut isoliert, so dass sich ein Umstieg noch nicht lohnt.

**Eure Einstellung, EVCC für genau einen Zweck zu optimieren und so schlank wie möglich zu halten ist ein sehr guter UX-Ansatz!** 

Also vergesst meinen Wunsch und der oben geschilderte Fall ist ein Sonderfall der nicht berücksichtigt werden muss. :-)

premultiply, 2022-11-26:
Siehe #5200 als Kompromiss ;)