evcc-io/evcc #5078: Charge Session Data

kuboschek, 2022-11-06:
_I also speak German_

**Is your feature request related to a problem? Please describe.**
For tax reasons, we would like to count the energy charged into each of their vehicles.
Our KEBA EVSE logs the total energy and energy per charge session, and it's displayed on the evcc UI as well.

**Describe the solution you'd like**
As a developer, I would like to access the data on current and past charge sessions, including which charge point and vehicle was charged.

Another workable solution is to get reports on new charge sessions via MQTT so I can store the data away separately.

**Describe alternatives you've considered**
I understood there is an integration to write data to Influx DB. Does this include the charge session data?
The documentation (https://docs.evcc.io/docs/reference/configuration/influx/) on the scope of data loaded to Influx is very sparse currently.

I have considered pinging the KEBA station separately to pull the data with a custom script. Can evcc handle receiving the responses caused by a different program requesting data from the charge point? Based on https://github.com/evcc-io/evcc/blob/586803646172d1745096c6ef31beef6a3b2539e3/charger/keba.go#L135 this could also work.

I can contribute code in whichever direction will be good for both the project and my use-case.

andig, 2022-11-06:
Die Energy je Fzg ist doch im Sessionlog schon enthalten- was fehlt da noch?

kuboschek, 2022-11-09:
Ah danke, ich find's wohl gerade nicht, in den API-Docs steht nur etwas von State - ist das dort dabei?

andig, 2022-11-09:
Yep 👍🏻

closed 2022-11-09 by andig