evcc-io/evcc #5081: Log serial port params and modbus protocol

premultiply, 2022-11-06:
For debugging and verification purposes the serial port device (maybe also tcp?) and the applied parameters (baudrate, comset) and where applicable the used modbus protocol (tcp, rtu, ascii or raw) used for initialisation should be logged at least on trace level (also for `evcc charger` and `evcc meter`).

andig, 2022-11-06:
@premultiply das berühmte „8E1“ ist lange behoben- mir fehlt hier ehrlich gesagt der Bedarf?

premultiply, 2022-11-06:
Es ist sonst einfach nicht ersichtlich was wirklich angewandt wird.
Mir ist das erst heute wieder in mehreren Fällen aufgefallen. Ich fische da immer im Trüben bei den oftmals spärlichen Informationen.

andig, 2022-11-07:
Die Infos stehen doch glasklar in der Config?

premultiply, 2022-11-07:
Welche aber gerade wirklich auch aktiv ist sieht man nicht.

andig, 2022-11-07:
Es gibt je Connection nur eine Konfiguration. Mischen ist nicht möglich.

premultiply, 2022-11-07:
Ich bezog das auf die evcc-Konfigd.
Mal aus dem Home-Verzeichnis mit örtlicher evcc.yaml gestartet, mal als Dienst aus /etc/evcc.yaml. Mal unbekannt, mal neustart vergessen oder damit andere Konfig aktiviert usw.

andig, 2022-11-07:
Mhhhm. Beim Service hilft auch `evcc discuss` nicht wirklich. Aktuell leider nicht so gut lösbar, da wär erstmal ein Refactoring der ganzen Modbus Verbindungen fällig. Ich machs bis dahin zu, Modbus kann dauern...

closed 2022-11-07 by andig

premultiply, 2022-11-07:
ok