evcc-io/evcc #5092: Anzahl Ladevorgänge nicht korrekt mit Go-e und disconnect

bicaluv, 2022-11-07:
### Describe the bug

In der Anzeige der Ladevorgänge fehlen Einträge, wenn das Auto von der Go-e Wallbox getrennt wird, ohne dass vorher das Laden gestoppt wird. Stoppt das Laden, weil das SoC erreicht ist und bleibt das Auto dabei verbunden, werden Werte in die Liste der Ladevorgänge eingetragen.

### Steps to reproduce

Starte das Laden mit "schnell" oder "min+pv" und trenne das Auto ohne vorher das Laden zu stoppen.  
(Ob der Fehler auch dann auftritt, wenn "pv" aktiv ist konnte ich noch nicht prüfen)

### Configuration details

folgende Log-Einstellungen waren aktiv:  
```
# log settings
log: debug
levels:
  site: info
  lp-1: info
  lp-2: info
  cache: debug
  db: trace
```

### Log details

```
Nov 06 23:18:31 raspberrypi evcc[27343]: [lp-1  ] INFO 2022/11/06 23:18:31 car connected
Nov 06 23:18:31 raspberrypi evcc[27343]: [main  ] DEBUG 2022/11/06 23:18:31 vehicle status: B (CITIGOe iV)
Nov 06 23:18:31 raspberrypi evcc[27343]: [lp-1  ] INFO 2022/11/06 23:18:31 vehicle updated: unknown -> CITIGOe iV
Nov 07 07:05:50 raspberrypi evcc[27343]: [db    ] TRACE 2022/11/07 07:05:50 SELECT * FROM `sessions` WHERE charged_kwh>=0.05 ORDER BY created desc 19 <nil>
Nov 07 08:28:41 raspberrypi evcc[27343]: [lp-1  ] INFO 2022/11/07 08:28:41 start charging ->
Nov 07 08:28:41 raspberrypi evcc[27343]: [db    ] TRACE 2022/11/07 08:28:41 INSERT INTO `sessions` (`created`,`finished`,`loadpoint`,`identifier`,`vehicle`,`meter_start_kwh`,`meter_end_kwh`,`charged_kwh`) VALUES ("2022-11-07 08:28:41.29","0000-00-00 00:00:00","Carport","","CITIGOe iV",0.000000,0.000000,0.000000) RETURNING `id` 1 <nil>
Nov 07 08:38:43 raspberrypi evcc[27343]: [db    ] TRACE 2022/11/07 08:38:43 SELECT * FROM `sessions` WHERE charged_kwh>=0.05 ORDER BY created desc 19 <nil>
Nov 07 09:35:35 raspberrypi evcc[27343]: [db    ] TRACE 2022/11/07 09:35:35 SELECT * FROM `sessions` WHERE charged_kwh>=0.05 ORDER BY created desc 19 <nil>
Nov 07 10:39:41 raspberrypi evcc[27343]: [lp-1  ] INFO 2022/11/07 10:39:41 stop charging <-
Nov 07 10:39:41 raspberrypi evcc[27343]: [db    ] TRACE 2022/11/07 10:39:41 UPDATE `sessions` SET `created`="2022-11-07 08:28:41.29",`finished`="2022-11-07 10:39:41.367",`loadpoint`="Carport",`identifier`="",`vehicle`="CITIGOe iV",`meter_start_kwh`=0.000000,`meter_end_kwh`=0.000000,`charged_kwh`=0.000000 WHERE `id` = 42 1 <nil>
Nov 07 10:39:41 raspberrypi evcc[27343]: [lp-1  ] INFO 2022/11/07 10:39:41 car disconnected
Nov 07 10:39:41 raspberrypi evcc[27343]: [db    ] TRACE 2022/11/07 10:39:41 UPDATE `sessions` SET `created`="2022-11-07 08:28:41.29",`finished`="2022-11-07 10:39:41.393",`loadpoint`="Carport",`identifier`="",`vehicle`="CITIGOe iV",`meter_start_kwh`=0.000000,`meter_end_kwh`=0.000000,`charged_kwh`=0.000000 WHERE `id` = 42 1 <nil>
Nov 07 10:39:41 raspberrypi evcc[27343]: [lp-1  ] INFO 2022/11/07 10:39:41 vehicle updated: CITIGOe iV -> unknown
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.106.3 incl. latest from master

andig, 2022-11-07:
Ich wiederhole mal die Frage vom letzten Anlauf: was steht in der SessionEnergie
- während geladen wird
- danach

Also:
- evcc meter benutzen
- Go-E mit loggen

Ohne die Info zum Root Cause wird sich hier keine Lösung finden...

bicaluv, 2022-11-07:
Warum bekomme ich nicht die Chance und es wird gleich geschlossen?  
Wollte die Daten noch nachliefern.

bicaluv, 2022-11-07:
Charger zeigt natürlich etas an.

bicaluv, 2022-11-07:
Welche Log-Einstellungen hättest du denn gerne?

andig, 2022-11-07:
trace für den Go-E.

> Wollte die Daten noch nachliefern.

Dann machen wir einfach wieder auf. An der Stelle waren wir ja im anderen Issue schon...

bicaluv, 2022-11-07:
Haben mit diesen Einstellungen einen Log erstellt:  

```
log: debug
levels:
  site: info
  lp-1: info
  lp-2: info
  cache: trace
  db: trace
```

evcc -c ../etc/evcc.yaml charger:  
```
[main  ] INFO 2022/11/07 16:42:52 evcc 0.106.3_mm (877851f0)
[main  ] INFO 2022/11/07 16:42:52 using config file: ../etc/evcc.yaml
[db    ] TRACE 2022/11/07 16:42:52 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2022/11/07 16:42:52 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2022/11/07 16:42:52 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2022/11/07 16:42:52 SELECT * FROM `settings` 6 <nil>
Power:          6610W
Current L1..L3: 15.9A 15.7A 0A
Charge status:  C
Enabled:        true
Charged:        2.4kWh
Identifier:     <none>
```

Wie man sieht, wurde, über 2,4kWh geladen, im Ladevorgang-Eintrag steht aber nur 1,5kWh vom 1. Teil dees Ladens. Der abschließende Wert geht verloren. Ist also nicht ganz die Fehlersituation, dass eine 0 geschrieben wird, aber eben nur der alte Wert und nicht der korrekte neue Wert.

[mm.log.zip](https://github.com/evcc-io/evcc/files/9953322/mm.log.zip)

bicaluv, 2022-11-08:
@andig : magst du noch mal ins Log schauen?

andig, 2022-11-08:
Nur wenn Du eins erstellst :(

> trace für den Go-E.

bicaluv, 2022-11-08:
Dann bin ich zu doof. Wie mache ich den Trace für den Go-E? Oder meinst du den lp-1?

andig, 2022-11-08:
go-e: trace

cache raus.

andig, 2022-11-09:
Danke, damit lässt sich arbeiten. Die Sessionenergie kommt aus `dws`- im zweiten Log ist die zum Ende hin 0, kurz vorher gibts noch einen Wert. Jetzt bleibt die Frage wie ein guter Workaround aussehen könnte.

@premultiply haben wir eine Erwartungshaltung, bis zu welchem Zeitpunkt die Sessionenergie vom Charger lesbar sein sollte? Über den Disconnect hinaus?

bicaluv, 2022-11-09:
@andig : Habe bisher kein Sponsorentoken, da bisher nicht gebraucht. Kann aber die Abfrage IsAuthorized() "patchen", falls das ausreicht.  
Den PR könnte ich morgen testen, da das Auto heute nicht mehr da ist.  
Und ist mein Go-E v2? Eigentlich nicht, oder?

andig, 2022-11-09:
Auf jeden Fall scheint er nur das v1 API zu haben. Anderenfalls hätte er Zählerstände und es damit gäbe weitere mögliche Workarounds.

premultiply, 2022-11-10:
> @premultiply haben wir eine Erwartungshaltung, bis zu welchem Zeitpunkt die Sessionenergie vom Charger lesbar sein sollte? Über den Disconnect hinaus?

Ja, bis zum (Re)Connect. Wie bei der Fossil-Zapfsäule.

bicaluv, 2022-11-10:
@andig : Test heute war erfolgreich!

andig, 2022-11-10:
Super, danke für die Analyse!

closed 2022-11-10 by andig