evcc-io/evcc #5113: EVCC progressive webapp as installable app

Gifford47, 2022-11-09:
**Is your feature request related to a problem? Please describe.**
This is a request for a progressive web app feature for evcc. it would be very handy if the interface could be opened quickly like an app on the smartphone.

**Describe the solution you'd like**
[Here](https://developer.mozilla.org/en-US/docs/Web/Progressive_web_apps/Installable_PWAs) is an example what i mean.

**Describe alternatives you've considered**
I know that a can create a normal website shortcut to my homecreen, but that requires me to switch to the homescreen every time to open the website.

**Additional context**

andig, 2022-11-09:
Scheint mir aufgrund der HTTPS Anforderung nicht umsetzbar. Zumindest unter iOS ist es auch nicht notwendig für Appswitching auf den Homescreen zu gehen. Likely wontfix.

naltatis, 2022-11-09:
Ich würd das nicht ausschließen und sehe hier schon den Mehrwert für die Benutzung. Ja, aktuell haben wir kein HTTPS, aber da könnte man durchaus zukünftig was mit Self-Signed Certs machen. Das funktioniert ja bei ner Fritzbox, Octoprint oder Synology auch. Würd das aber aktuell eher als niedrige Priorität sehen.

andig, 2022-11-09:
Welches Problem löst das denn wirklich? Eines von Android?

StefanSchoof, 2022-11-09:
Es gibt schon ein manifest: https://demo.evcc.io/meta/site.webmanifest

Ich habe mein evcc mit [traefik](https://github.com/traefik/traefik) über https hinter auth veröffentlicht und nutzte evcc wie eine App auf meinem Android.

andig, 2022-11-10:
Sieht aus als hätten wir zumindest keinen akuten Handlungsbedarf. Ggf. Lösung falls wir https integrieren, dafür gibt es allerdings keine konkrete Pläne.

closed 2022-11-10 by andig