evcc-io/evcc #5182: Disable timer continues when leaving PV mode

premultiply, 2022-11-16:
### Discussed in https://github.com/evcc-io/evcc/discussions/5181

<div type='discussions-op-text'>

<sup>Originally posted by **VolkerK62** November 16, 2022</sup>
Ist das Verhalten so gewollt?

Es wird im Modus PV geladen. Überschuss sinkt, der disable Timer startet.
Ich schalte auf MinPV (in der Hoffnung, dass in kurzer Zeit wieder genug Überschuss vorhanden ist)
Diese Hoffnung wird nicht erfüllt, also schalte ich wieder zurück auf PV (in der Erwartung, dass der timer neu startet).
Dann wird die Ladung aber sofort beendet. Im Log steht "timer elapsed"
Offensichtlich läuft der timer im "Hintergrund" weiter.

Bug oder feature?
</div>

VolkerK62, 2022-11-20:
Beobachtung:
Kann es sein, dass gar kein timer gestartet wird, wenn beim Umschalten (von `minpv` oder `schnell`) auf `pv` die Bedingungen für PV-Laden nicht erfüllt sind?

VolkerK62, 2022-12-13:
Ich habe etwas rumprobiert. Es ist tatsächlich so, dass der timer nur dann aktiviert wird, wenn sich während `PV` Laden die Bedingungen verschlechtern und nicht mehr erfüllt sind.
Wird mit `MinPV` oder `schnell` geladen und dann bei Nichterfüllung der Bedingungen auf `PV` umgeschaltet, ist es so, als hätte man `aus` eingeschaltet.
Ich würde mir wünschen, dass auch in dem Fall der timer startet.

VolkerK62, 2022-12-20:
Falls es kein bug ist, sondern bewusst so designed, dann kann hier meinetwegen zu gemacht werden.
Ich habe jetzt einen anderen Weg für meinen Winter-Modus gefunden, bei dem ich komplett im PV-Modus bleibe und somit auch der timer richtig funktioniert.

andig, 2022-12-29:
> Dann wird die Ladung aber sofort beendet. Im Log steht "timer elapsed"

Die Frage ist ja, was das Wunschverhalten ist. Wenn ich bewusst den Modus ändere (z.B. auf "sofort" schalte) dann würde ich auch erwarten, dass bei erneutem Wechsel des Modus jegliches Hintergrundverhalten inaktiv ist. 

Analog würde ich beim Wechsel auf `minpv`- wo es keinen Timer gibt- auch erwarten, dass `pv` so reagiert, als wäre alles "frisch".

closed 2022-12-29 by andig