evcc-io/evcc #5191: Follow up issue to #5124

hlathome, 2022-11-17:
### Describe the bug

Original issue is resolved except:
`climater: json: cannot unmarshal number 42.7 into Go struct field ChargingStatus.Data.chargingStatus.chargeRate_kmph of type int`

### Steps to reproduce

see #5124

### Configuration details

```yaml
see #5124
```

### Log details

```text
/app # evcc --log debug
[main  ] INFO 2022/11/17 20:00:26 evcc 0.106.5
[main  ] INFO 2022/11/17 20:00:26 using config file: /etc/evcc.yaml
[main  ] INFO 2022/11/17 20:00:26 listening at :7070
[mqtt  ] INFO 2022/11/17 20:00:26 connecting evcc-483225113 at tcp://fhemprod:1883
[mqtt  ] DEBUG 2022/11/17 20:00:26 tcp://fhemprod:1883 connected
[mqtt  ] DEBUG 2022/11/17 20:00:26 tcp://fhemprod:1883 subscribe hlathome/DOIF_Energiebilanz/power.gridBalance
[site  ] INFO 2022/11/17 20:00:29 site config:
[site  ] INFO 2022/11/17 20:00:29   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2022/11/17 20:00:29     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/11/17 20:00:29     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2022/11/17 20:00:29   vehicles:
[site  ] INFO 2022/11/17 20:00:29     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.106.5

andig, 2022-11-17:
Welches Auto?

hlathome, 2022-11-17:
Audi eTron

andig, 2022-11-17:
Ist im nightly ebenfalls gefixt.

closed 2022-11-17 by andig

hlathome, 2022-11-17:
danke!