evcc-io/evcc #5197: Seit update auf 107 kein Fahrzeug mehr in der Ansicht.

Bluppy, 2022-11-18:
### Describe the bug

Seit dem Update auf 0.107.0 werden die Fahrzeuge nicht mehr erkannt.

Die Fahrzeuge stehen bei der Auswahl:
unavailable
unavailable

Bei der Abfrage via Vehicle sind die Fahrzeuge sauber drin. Beim Neustart ist das letzte Fahrzeug sauber mit Name drin.

Allgemein dauert der Start des evcc Dienst mega lang.

### Steps to reproduce

Fehler besteht ganze zeit

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval
tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.3335 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.09 # [currency]/kWh
log: error

  #influx:
  #url: http://localhost:8086
  #database: evcc

meters:
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.1.150  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv1
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.1.150  
  port: 1502  
  usage: battery  
  modbus: tcpip  
  name: battery2
- type: template
  template: kostal-ksem-inverter 
  id: 71  
  host: 192.168.1.150  
[... 174 lines cut ...]
[lp-1  ] DEBUG 2022/11/18 12:18:35 vehicle soc: 45%
[lp-1  ] DEBUG 2022/11/18 12:18:35 vehicle range: 178km
[lp-1  ] DEBUG 2022/11/18 12:18:35 vehicle target soc: 100%
[lp-1  ] DEBUG 2022/11/18 12:18:35 pv charge current: 3.33A = 0A + 3.33A (-2297W @ 3p)
[lp-1  ] DEBUG 2022/11/18 12:18:38 set target soc: 100
[site  ] DEBUG 2022/11/18 12:18:38 ----
[lp-1  ] DEBUG 2022/11/18 12:18:38 charge power: 0W
[site  ] DEBUG 2022/11/18 12:18:39 pv power: 2911W
[site  ] DEBUG 2022/11/18 12:18:39 battery power: -2410W
[site  ] DEBUG 2022/11/18 12:18:39 grid power: -7W
[site  ] DEBUG 2022/11/18 12:18:39 battery soc 0: 94%
[site  ] DEBUG 2022/11/18 12:18:39 site power: -2317W
[lp-1  ] DEBUG 2022/11/18 12:18:39 charger status: B
[lp-1  ] DEBUG 2022/11/18 12:18:39 next soc poll remaining time: 59m56s
[lp-1  ] DEBUG 2022/11/18 12:18:39 pv charge current: 3.36A = 0A + 3.36A (-2317W @ 3p)
^C[main  ] DEBUG 2022/11/18 12:18:50 telemetry: charge: Δ2/2Wh @ 0W
```

### What type of operating system are you running?

Linux

### Version

0.107.0

andig, 2022-11-18:
[lp-1  ] ERROR 2022/11/18 12:18:24 vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'kia': cannot get vehicles: Get "https://prd.eu-ccapi.kia.com:8080/api/v1/spa/vehicles": net/http: request canceled (Client.Timeout exceeded while awaiting headers)

andig, 2022-11-18:
Ich mache nochmal auf. Auffällig ist, dass der Erste (root cause) Fehler nicht im Log steht. SoC ist ja nur ein Folgefehler.

andig, 2022-11-18:
Gehts denn wenn Du auf 106 zurück gehst? Falls nein liegts nicht an evcc...

Bluppy, 2022-11-18:
> Gehts denn wenn Du auf 106 zurück gehst? Falls nein liegts nicht an evcc...

Gibt es einenn Weg ein downgrade per apt befehl umzusetzen? Vor den Update lief es noch mit 106.

Bluppy, 2022-11-18:
Okay nach downgrade auf 106.5 geht der Neustart zwar schneller aber fahrzeuge bleiben unbekannt. Okay dann muss ich schauen welches andere Paket evcc mit den update komplett gegangen ist.

Bluppy, 2022-11-18:
Mh kein wirkliches Paket enthalten was das Verhalten erklärt:

Start-Date: 2022-11-18  11:52:19
Commandline: apt-get upgrade -y
Requested-By: ubuntu (1000)
Upgrade: libgdk-pixbuf2.0-bin:arm64 (2.42.8+dfsg-1ubuntu0.1, 2.42.8+dfsg-1ubuntu0.2), evcc:arm64 (0.106.5, 0.107.0), gir1.2-gdkpixbuf-2.0:arm64 (2.42.8+dfsg-1ubuntu0.1, 2.42.8+dfsg-1ubuntu0.2), libgdk-pixbuf-2.0-0:arm64 (2.42.8+dfsg-1ubuntu0.1, 2.42.8+dfsg-1ubuntu0.2), libgdk-pixbuf-2.0-dev:arm64 (2.42.8+dfsg-1ubuntu0.1, 2.42.8+dfsg-1ubuntu0.2), libgdk-pixbuf2.0-common:arm64 (2.42.8+dfsg-1ubuntu0.1, 2.42.8+dfsg-1ubuntu0.2)
End-Date: 2022-11-18  11:52:29

Bluppy, 2022-11-18:
Mh okay sehr komisches verhalten. Ich habe einfach mit den advanced script nochmal eine neue Konfiguration yaml erzeugt. Nach aktivieren dieser Konfiguration läuft es. Logindaten sind aber eins zu eins gleich da sogar kopiert.

andig, 2022-11-18:
Closed in https://github.com/evcc-io/evcc/commit/99264e4cf9a12cecc520a97e04ec42fd68a38bd3

closed 2022-11-18 by andig