evcc-io/evcc #5198: Encrypt VIN and password in evcc.yaml

igotsoul, 2022-11-18:
**Is your feature request related to a problem? Please describe.**
Passwords and user critical data shouldn't stored in plain text, especially if they're used for vehicle status access.

**Describe the solution you'd like**
On first startup of evcc ensures that the plain text credentials in the config will be encrypted

**Describe alternatives you've considered**
-

**Additional context**
Add any other context or screenshots about the feature request here.

andig, 2022-11-18:
Hast Du eine Idee wie eine technische Lösung dazu aussehen sollte und möchtest Dich an der Implementierung beteiligen? Sonst hat das keine Chance auf Umsetzung.

igotsoul, 2022-11-18:
Das hier sieht doch ganz brauchbar dafür aus: https://github.com/palantir/go-encrypted-config-value

andig, 2022-11-18:
In der Tat, das ist spannend. Aber- für so wichtig ich das Thema auch halte- momentan keine Priorität. Wir werden uns zunächst auf das Thema Config UI konzentrieren um die Zugangshürden zu evcc möglichst weiter zu reduzieren. Daher vorerst wontfix. Trotzdem vielen Dank, dass Du das Thema auf die Agenda gebracht hast!