evcc-io/evcc #5199: Climate friendly charging with adapt.sh

thorstenwagner, 2022-11-18:
It's not always possible to fully charge via solar, especially in the winter months. I think it would be great to take the predictions of https://adapt.sh/  into account. Like: Only charge if it's at least comfortable for the climate.

What do you think?  Is it out of scope of evcc?

andig, 2022-11-18:
We're working on a "winter" mode. Climate friendliness can pretty well be approximated with PV availability imho which is what we're gonna improve next. May revisit later, wontfix for time being. Thank you for the suggestion and interesting link!

thorstenwagner, 2022-12-12:
I comment here: Maybe you could also use the current carbon intensity from electricity maps ( API: https://app.electricitymaps.com/zone/DE )

andig, 2022-12-12:
See https://github.com/evcc-io/evcc/pull/5454. Do you want to provide a PR on that basis?

thorstenwagner, 2022-12-16:
Asked around on Twitter: https://mobile.twitter.com/pixelbaumarkt/status/1603811295958122517

Maybe Adrien from adapt.sh can help us.

andig, 2022-12-29:
Seems we're not getting support from adapt.sh, too bad.

closed 2022-12-29 by andig

andig, 2022-12-30:
Siehe https://github.com/evcc-io/evcc/discussions/4554#discussioncomment-4556573

thorstenwagner, 2022-12-31:
Hmm, that came via twitter: https://www.co2signal.com/
An alternative l?

andig, 2022-12-31:
Nope- its missing the forecast that we need for planning :(

premultiply, 2022-12-31:
https://gruenstromindex.de/ ?

andig, 2022-12-31:
Guter Tipp: https://github.com/evcc-io/evcc/pull/5635. Da brauchts nur noch den ZIP Code den man sich fast noch per IP-Lookup besorgen könnte. Damit wäre es zero-config!