evcc-io/evcc #5202: PV überschuss je Ladevorgang

PVJan, 2022-11-19:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

andig, 2022-11-19:
Wäre prinzipiell machbar, ist aber ein sehr spezielles Szenario. Das hat bei den aktuellen Themen nur eine Chance wenn sich jemand findet das zu bauen.

/cc @naltatis

naltatis, 2022-11-19:
Was wir aber auf der Liste haben ist, dass wir den Eigenenergieanteil im Ladevorgang erfassen wollen. Auf der Basis könntest du ja abrechnen.

PVJan, 2022-11-19:
Ist der eigenenergieanteil nicht automatisch der pV Überschuss bzw. pv Anteil ?

manny4566, 2022-11-19:
Hi,

ich knüpfe an der Stelle auch noch einmal an mit meiner Bitte, dass pro Ladevorgang auch der Sonnenanteil in Prozent berechnen und dargestellt wird, z.b. unter den Sessions als weitere Information. Ggf. könnte man dort generell den Ladevorgang mit weiteren Informationen der entsprechenden Session erweitern. Sonnenenergie in %, Unterteilung der Gesamtladung in Netzbezug, Batteriezuschuss, PV Bezug. Danke
Meine ursprüngliche Anfrage habe ich hier gestellt:

[#3503 ](3503)

naltatis, 2023-02-04:
👆 Der PR schafft jetzt einige Vorbedingungen mit denen wir mehr Sessiondaten erfassen können (%Eigenstrom, co2, Preis).

fobi87, 2023-02-13:
Wenn ihr hinbekommt dass man PV und Netzbezug differenzieren kann je Ladevorgang, vielleicht mit Tages- , Wochen-, Monats-, Jahres- und Gesamtübersicht, habt ihr definitiv einen Sponsor mehr und ihr würdet euch von der Konkurrenz deutlich abheben. 
Weiter so Jungs !!

andig, 2023-02-13:
@fobi87 von welcher Konkurrenz ;) Scherz beiseite- ich hab mir den Markt nie angeschaut- gibts vergleichbare Lösungen?

fobi87, 2023-02-13:
@andig Konkurrenz belebt ja das Geschäft oder besser gesagt Produkt. Hab zwar noch keine PV, diese wird aber in den nächsten 2 - 3 Monaten realisiert. 
Habe mich daher mit einigen evcc alternativen beschäftigt. Aber bei allen fehlt mir eben diese PV/Netz differenzierung. 
Alle haben Nachteile/Vorteile.
Alles über Cloud-API, zuviel bastelei, reiner Onlinedienst, etc ...
Ich kann mir vorstellen wie intensiv dieses Projekt ist, evcc könnte sich ja zu einem Allrounder entwickeln, mit Wärmepumpenanbindung etc ...

andig, 2023-02-15:
> Habe mich daher mit einigen evcc alternativen beschäftigt

OT: kannst Du verraten, welche Du Dir angeschaut hast?

fobi87, 2023-02-15:
@andig ev-autocharge, solaranzeige, clever-pv, solarmanager und bestimmt noch 2-3 andere die ich mir nicht gemerkt habe, die meinen Wechselrichter oder Wallbox teilweise  nicht unterstützen oder aber auf eigene Energiezähler setzen.

hoermto, 2023-02-21:
Gibts dann auch ein Ladelog mit Persistenz? Das wäre cool.

andig, 2023-02-21:
Das gibts doch schon!

hoermto, 2023-02-21:
> Das gibts doch schon!

Oh, ist mir entgangen. Ich laufe aktuell noch mit der 0.109.
Mal ne ggf komische Frage: gibts eigentlich sowas wie ein Changelog oder Release notes, die ggf auch in der Gui angezeigt werden, wo man die letzten Änderungen sieht? Es gibt ja die Info über die aktuelle Version unten links, da wäre sowas als Link ggf gut aufgehoben?

andig, 2023-02-21:
Ist hier völlig OT, aber Du musst ja nur auf den Link drauf klicken und gelangst zu https://github.com/evcc-io/evcc/releases

hoermto, 2023-02-21:
ist mir irgendwie entgangen dass das da drinsteht ...

kuboschek, 2023-03-21:
Hi, würde hier gerne Code beisteuern. Vue sowie Golang kenne ich, evcc-Code noch nicht.

Ich sehe hier zwei vielleicht drei Tasks:

1. PV-Anteil pro Ladung berechnen und persistieren
2. Das ganze auch anzeigen

Use-Cases wie Abrechnung oder Preis-Berechnung sind dann per CSV-Export und weiterer Verarbeitung möglich. Da bin ich aus raus.

andig, 2023-03-21:
Sehr gerne :) Siehe site.go!

kuboschek, 2023-03-26:
Danke, hab' mich etwas umgeschaut.

So wie ich verstehe, schreibt allein der Loadpoint aktuell die Charging Sessions in die DB.
Zugriff auf die anderen Meter (PV, Grid und so) bzw. greenShare hat aber nur die Site.

Beides brauche ich an einer Stelle, um den PV-Anteil pro Session zu ermitteln.
Dem Loadpoint auf die Site Zugriff zu geben erscheint mir falsch, normal geht's ja andersherum.
Aus der Site ein Event mit der aktuellen Erzeugung zu senden wäre wohl auch möglich.

Am Ende braucht's etwas wie "pvMeterStart" und "pvMeterEnd", natürlich summiert bei mehreren, aber halt im Loadpoint verfügbar.

Bin hier unsicher was die richtige Richtung angeht.

hoermto, 2023-03-26:
Update() könnte dem LP als weiteren Parameter den aktuellen Greenshare mitgeben, damit LP das loggen kann.

andig, 2023-03-26:
Das wird nicht reichen. Der muss ja pber den Ladevorgang averaged werden.

kuboschek, 2023-03-26:
Dann geht's wohl nur über die absoluten Energy Werte über den Zeitraum der Ladung.

premultiply, 2023-03-26:
Summen aller Energiequellen von (EnergyTotalEnd-EnergyTotalStart) über den Zeitraum der jeweiligen Ladeslots.

premultiply, 2023-03-27:
Ist das Momentanverhältnis wirklich relevant?
Reicht es nicht aus die Summenenergien von und zu den relevanten Quellen und Senken pro Ladevorgangs ins Verhältnis zu setzen?

Ich bin mir auch nicht sicher - muss man in Ruhe drüber nachdenken. Dafür fehlt mir gerade Ruhe :(

naltatis, 2023-03-28:
Das ist auf jeden Fall relevant. Ladevorgang ist ja bei uns vom Anstecken bis zum Abstöpseln. Wenn ich zwei Ladepunkte habe die gleichzeitig anfangen und enden, das eine Fahrzeug lädt aber bspw. seine Energie am Tag (Anfang des Ladevorgangs) und das andere in der Nacht (Ende des Ladevorgangs), dann sollten beide Ladevorgänge unterschiedliche Werte haben. Auch wenn sie in Summe vielleicht die gleiche Energiemenge genommen haben.

Will sagen: Es ist relevant zu welchem Zeitpunkt im Ladevorgang wie viel geladen wurde.

premultiply, 2023-03-28:
Bei jedem jedem aktiven Ladeslot-Anfang/Ende (Wechsel zu/von Status C) müsste natürlich neu abgelesen werden und dann über jeden Ladeslot eine neue Summe pro LP gebildet werden und diese dann über die gesamte Session addiert bzw. der Mittelwert gebildet werden.

Oder bin ich damit immer noch auf dem Holzweg?

naltatis, 2023-03-28:
Das wäre mir auch zu grob. Selbst wenn wir es nicht pro Ladevorgang (An-/Abstecken) sondern pro aktivem Laden (Status C) machen sprechen wir hier ja potentiell über große Zeiträume (Stunden) in denen sich das Verbrauchsverhältnis zwischen Haus und den einzelnen LPs ja durchaus ändert.

juja123, 2023-04-05:
Hallo,
ich las bisher begeistert mit. Nun bin ich zufällig über diese Diskussion gestolpert und würde gerne meine Meinung beitragen. Mit der Programmierung in Go kenne ich mich leider nicht aus (andere Sprachen schon), aber mit der Bilanzierung von Energien, insbesondere der aus einer PV-Anlage.
Und im Kontext der Nutzungsanteile der PV-Energie ist immer die Problemstellung, dass man möglichst hochaufgelöst die einzelnen Leistungen in ein Verhältnis setzt und dann über die Zeit summiert (integeriert), so wie das @naltatis ja schon beschrieben hat. Ansonsten würde man annehmen, dass die Leistung aus PV, Charger und Gebäude über den Zeitraum konstant wären.
Beispiel: Im Gebäude befindet sich ein Großverbraucher wie eine Wärmepumpe, wenn diese anspringt steigt der Verbrauch stark an. Es wird Strom aus dem Netz bezogen. Evcc regelt je nach Modus runter. Damit bezieht vorallem das Gebäude aus dem Netz, was aber in einer Betrachtung über den gesamten Zeitraum nicht berücksichtigt wird. Selbiges kann man mit der schwankenden PV-Erzeugung durchgehen.

premultiply, 2023-04-05:
Das Integral von Leistung über Zeit nennt sich Energie 😁

naltatis, 2023-04-05:
Ich bau morgen mal was über das wir dann diskutieren können :D

PR #7380 (2023-05-02): Add session infos (co2, price, solar %)
- ☀️💰🌱 collect solar %, price and co2 data per charging session.
- 📱UI shows available data in the top right field of the loadpoint (previously time).
- 👉 Toggle through values by tapping/clicking. Choice is remembered per Loadpoint.
- 📜 data is persistent and available in charging session log and csv exports.
- Introduced `EnergyMetrics` to encapsulate energy stats. This can be reused for future data collection (e.g. `dailyEnergy` in mini loadpoint).

Todos

- [x] add tests
- [x] handle null data (db) in sessions UI (json)
- [x] pointer formatting in CS [... cut]
closed 2023-05-02 by naltatis