evcc-io/evcc #5205: Bluelink: Vehicle soc: timeout führt zu ungewollter Ladung

MathysDev, 2022-11-19:
### Describe the bug

Ich erhalte wiederholt timeout-Meldungen bei der Abfrage des SOC via bluelink.
Zusammen mit minSOC führt dies dazu, dass im PV-Modus geladen wird obwohl SOC (nicht geladen) > minSOC wäre

Spontan zwei Lösungsansätze (noch ungetestet)
- Verlängerung Timeout
- Wenn SOC unbekannt, Mindestladung nicht starten

### Steps to reproduce

1. EV mit bluelink konfigurieren
2. Restart EVCC

### Configuration details

```yaml
vehicles:
- type: template
  template: hyundai
  title: IQ5
  user: ****
  password: ****
  capacity: 72
  cache: 30m
  phases: 3
  language: de
  mode: pv
  minSoC: 5
  targetSoC: 90
  minCurrent: 6
  maxCurrent: 16
  name: IQ5
```

### Log details

```text
-- Journal begins at Mon 2022-04-04 16:41:41 CEST, ends at Sat 2022-11-19 09:08:13 CET. --
Nov 19 09:03:16 raspi evcc[631]: [lp-1  ] INFO 2022/11/19 09:03:16 start charging ->
Nov 19 09:03:16 raspi evcc[631]: [lp-1  ] ERROR 2022/11/19 09:03:16 vehicle soc: timeout
Nov 19 09:04:56 raspi evcc[631]: [lp-1  ] ERROR 2022/11/19 09:04:56 vehicle soc: timeout
Nov 19 09:06:36 raspi evcc[631]: [lp-1  ] ERROR 2022/11/19 09:06:36 vehicle soc: timeout
```

### What type of operating system are you running?

Linux

### Version

0.107.0

VolkerK62, 2022-11-19:
Lösung ist `minSoC: 0` 
https://docs.evcc.io/docs/guides/charging#warum-wird-mit-voller-leistung-geladen-wenn-der-soc-des-fahrzeug-nicht-erkannt-wird

MathysDev, 2022-11-19:
Das verringert aber die Funktionalität, wenn bluelink eben doch funktioniert. Die Timeouts sind ja nur Zeitweise. Manchmal ist bluelink "schnell genug"

andig, 2022-11-19:
@MathysDev das wurde schon dutzende male diskutiert. Wir entscheiden uns immer gleich: im Zweifel (und das ist so ein Fall) lieber fahrbereites Auto als 2€ gespart!

MathysDev, 2022-11-19:
Schade. Bluelink ist dermassen instabil, dass es durchaus auch mal den Akku auffüllt, da angenommen wird dass SoC = 0 wenn timeout oder anderer Fehler.

andig, 2022-11-19:
Schade Bluelink, ja. Ich fände leeres Auto aber noch schlimmer wenn ich los wollte...

VolkerK62, 2022-11-19:
@MathysDev  du hast ja auch einen Ioniq5. Magst du dir mal #4097  anschauen und mir dort evtl Feedback geben, ob es bei dir auch so ist.
Problem ist, dass der Ioniq5 beim anstecken von evcc nicht erkannt wird, bzw, erst, wenn geladen wird.