evcc-io/evcc #5206: Animationsfehler?

Kai9555, 2022-11-19:
### Describe the bug

Hallo zusammen,

zunächst danke für das tolle Projekt, welches ich gern durch ein kleines Sponsoring unterstütze.

Ist es normal, dass die Animation der Klammer beim aktualisieren so springt (siehe Video unten am Haus rechts). Das verhalten habe bei jeder Aktualisierung und sieht irgendwie unschön aus.
[https://www.dropbox.com/s/hp9i6ftrj1cwcr3/Bildschirmaufnahme%202022-11-19%20um%2009.12.21.mov?dl=0](url)

Vielen Dank

Mit freundlichen Grüßen
Kai

### Steps to reproduce

Jederzeit

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: a5b95c34f0109b329f7dff611dafd198c8e0b8ed415180ee2ac7013c8731ee5c

interval: 5s # control cycle interval

sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJleHAiOjE3NjI2MzM2MTMsImlhdCI6MTY2ODAyNTYxMywiaXNzIjoiZXZjYy5pbyIsInN1YiI6Ikthxxxxxxxxxxxxxxxxxxxxxxxxxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: sungrow-inverter 
  id: 1  
  host: 192.168.2.xxx
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv1
- name: pv2
  type: template
  template: shelly-1pm
  usage: pv
  host: 192.168.2.xxx
- name: tasmota
  type: custom
  power:
     source: http
     uri: http://192.168.2.xxx/cm?cmnd=Status%208
     jq: .StatusSNS.MT175.P

chargers:
- type: template
  template: go-e-gemini 
  host: 192.168.2.xxxx
  name: wallbox4

loadpoints:
- title: Carport
  charger: wallbox4
  mode: off
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false

site:
  title: Mein Zuahuse
  meters:
    grid: tasmota
    pvs: 
    - pv1
    - pv2
```

### Log details

```text
[main  ] INFO 2022/11/19 09:24:51 evcc 0.107.0
[main  ] INFO 2022/11/19 09:24:51 using config file: /etc/evcc.yaml
[main  ] INFO 2022/11/19 09:24:51 starting ui and api at :7070
[lp-1  ] WARN 2022/11/19 09:24:51 locking phase config to 3p for switchable charger
[site  ] INFO 2022/11/19 09:24:51 site config:
[site  ] INFO 2022/11/19 09:24:51   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2022/11/19 09:24:51     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/11/19 09:24:51     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2022/11/19 09:24:51     pv 2:      power ✓ energy ✗ currents ✗
[lp-1  ] INFO 2022/11/19 09:24:51 loadpoint 1:
[lp-1  ] INFO 2022/11/19 09:24:51   mode:        off
[lp-1  ] INFO 2022/11/19 09:24:51   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2022/11/19 09:24:51   meters:      charge ✓
[lp-1  ] INFO 2022/11/19 09:24:51     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2022/11/19 09:24:51 phase timer inactive
[lp-1  ] DEBUG 2022/11/19 09:24:51 pv timer inactive
[site  ] DEBUG 2022/11/19 09:24:51 ----
[server] FATAL 2022/11/19 09:24:51 unlinkat /tmp/evcc: operation not permitted
```

### What type of operating system are you running?

Linux

### Version

107

premultiply, 2022-11-19:
/cc @naltatis

andig, 2022-11-19:
Das sieht eigentlich aus als würde etwas mit den Meßwerten nicht stimmen- jedenfalls hab ich den Effekt immer beim testen wenn etwas krumm ist. Sind denn alle Werte plausibel?

Kai9555, 2022-11-19:
> Das sieht eigentlich aus als würde etwas mit den Meßwerten nicht stimmen- jedenfalls hab ich den Effekt immer beim testen wenn etwas krumm ist. Sind denn alle Werte plausibel?

Ja die Werte sind plausibel und nachvollziehbar korrekt. Auf jeden Fall schön zu hören, dass es kein normales Verhalten ist. Das einzige was noch nicht eingebunden ist, ist ein E-Auto (da ich es noch nicht habe). Bitte gebt mir bescheid wenn weitere Infos zum ergründen nötig sind.

Vielen Dank

naltatis, 2022-12-06:
@Kai9555 der Dropbox Link funktioniert nicht mehr. Magst du den noch mal aktualisieren oder das Video direkt hier hochladen?

naltatis, 2022-12-06:
Ok, danke.

naltatis, 2022-12-29:
> Das sieht eigentlich aus als würde etwas mit den Meßwerten nicht stimmen- jedenfalls hab ich den Effekt immer beim testen wenn etwas krumm ist. Sind denn alle Werte plausibel?

Das wäre jetzt auch meine Hypothese. Ich hab es nicht hinbekommen den Versatz in der Animation zu provozieren. @Kai9555 Tritt das noch immer bzw. regelmäßig bei dir auf? Wenn ja, hast du einen Log-Auszug dazu?

Kai9555, 2022-12-29:
> Das wäre jetzt auch meine Hypothese. Ich hab es nicht hinbekommen den Versatz in der Animation zu provozieren. @Kai9555 Tritt das noch immer bzw. regelmäßig bei dir auf? Wenn ja, hast du einen [...]

Ja der Fehler besteht noch immer und war mit allen Versionen der letzten Monate immer gleich. Ich schaue mal die Tage das ich ein Log bereitstelle. In der mobilen Ansicht ist es übrigens immer versetzt.
[image]

andig, 2022-12-29:
@Kai9555 mit ziemlicher Sicherheit stimmen Deine Messwerte einfach nicht. Mach bitte eine neue Discussion auf. Wenn wir überzeugt sind, dass die Messwerte zueinander passen und der Fehler dann immer noch auftritt dann können wir uns das hier weiter anschauen. DANKE!

Kai9555, 2022-12-29:
> @Kai9555 mit ziemlicher Sicherheit stimmen Deine Messwerte einfach nicht. Mach bitte eine neue Discussion auf. Wenn wir überzeugt sind, dass die Messwerte zueinander passen und der Fehler dann immer noch auftritt dann können wir uns das hier weiter anschauen. DANKE!

Was meinst du denn mit "Messwerte stimmen nicht" - die Werte die gelesen werden stimmen mit denen der nativen Apps auf jeden Fall überein. Hier kann ich keine Abweichung feststellen. Was für einen Ansatz kann ich denn verfolgen um das Problem einzugrenzen?

premultiply, 2022-12-29:
`evcc meter`?

premultiply, 2022-12-29:
Die Frage ist nicht was irgendwelche Apps anzeigen sondern ob die Leistungsbillanz deines Geräteparks stimmt.
Dafür ist z. B. wichtig ob die Geräte richtig messen und elektrisch an der vorgesehenen und erwarteten Stelle richtig herum eingebaut sind.

Kai9555, 2022-12-29:
> Die Frage ist nicht was irgendwelche Apps anzeigen sondern ob die Leistungsbillanz deines Messpunktzoos stimmt.
> Dafür ist z. B. wichtig ob die Geräte richtig messen und elektrisch an der vorgesehenen und erwarteten Stelle richtig herum eingebaut sind.

Ich habe gerade alles aus der ivcc.yaml rausgehauen und nur den gridmeter (Hichi Tasmota Lesekopf am Zähler) konfiguriert 

meters:
- name: tasmota
  type: custom
  power:
     source: http
     uri: http://192.168.2.65/cm?cmnd=Status%208
     jq: .StatusSNS.MT175.P

Der Lesekopf misst definitiv korrekt, da ich live die Werte mit dem Zähler abgleichen kann und dennoch springt die Klammer.

premultiply, 2022-12-29:
> `evcc meter`?

Kai9555, 2022-12-29:
> `evcc meter`?

Wie meinst du das? Ich verstehe nicht ganz.

premultiply, 2022-12-29:
Na auf der Konsole einfach `evcc meter` eingeben und die Ausgabe hier einfügen.

Kai9555, 2022-12-29:
> Na auf der Konsole einfach `evcc meter` eingeben und die Ausgabe hier einfügen.

Natürlich:

[main  ] INFO 2022/12/29 20:28:41 evcc 0.110.1
[main  ] INFO 2022/12/29 20:28:41 using config file: /etc/evcc.yaml
[db    ] INFO 2022/12/29 20:28:41 using sqlite database: /home/pi/.evcc/evcc.db
Power: 2259W

Wie gesagt ist jetzt nur der Stromzähler konfiguriert und die Klammer springt hin und her.

Kai9555, 2022-12-30:
> @Kai9555 mit ziemlicher Sicherheit stimmen Deine Messwerte einfach nicht. Mach bitte eine neue Discussion auf. Wenn wir überzeugt sind, dass die Messwerte zueinander passen und der Fehler dann immer noch auftritt dann können wir uns das hier weiter anschauen. DANKE!

Hallo @andig soll ich nun ein neues Thema aufmachen, oder reaktivieren wir das hier? Ich habe alles neu aufgesetzt und nur den Hichi Lesekopf vom Stromzähler wie folgt konfiguriert:

```
- name: tasmota
  type: custom
  power:
     source: http
     uri: http://192.168.2.65/cm?cmnd=Status%208
     jq: .StatusSNS.MT175.P
```

Dennoch Springt die Klammer. Die Werte sind definitiv korrekt und werden wie folgt ausgeben:

```
{
  "StatusSNS": {
    "Time": "2022-12-30T07:47:31",
    "MT175": {
      "E_in": 858.283,
      "P": 343,
      "L1": 94,
      "L2": 73,
      "L3": 174,
      "E_out": 57.175,
      "Meter_id": ""
    }
  }
}
```