evcc-io/evcc #5212: Tweak community data to show only whole numbers, without numbers after the decimal point "Charge power kW" and "solar power %"

andig, 2022-11-19:
I think it would be good to tweak community data to show only whole numbers, without numbers after the decimal point "Charge power kW" and "solar power %"
As more and more users join, it does not make sense to show such small values, and they keep changing like crazy.

PS : Also the change is happening too slow on the web site https://evcc.io/, it takes forever to count down or up, ir there was a larger change in charging power for example. 
Maybe it would look OK, if no decimal point were shown too?

_Originally posted by @RTTTC in https://github.com/evcc-io/evcc/issues/5001#issuecomment-1320864750_

andig, 2022-11-19:
I would also like to rethink the update visuals. When you open the website after longer time, you'll see rotating numbers for a long time if the change has been big. Imho a bit much visual overload.

flushbug, 2022-11-19:
Removing the decimal point would also fix an issue I have observered on small screens like my iPhone 13 mini. It seems like specific numbers in the font are wide and forcing the textbox with the number of charging users switching from 2 lines to three lines. If the total charge power is changing a lot, the text right next to it is toggeling from 2 to 3 lines over and over again. I was only able to make a screenshot of both states an no video.
[image]
[image]

another option would be to limit the charging power Display to 3 digits. Below 100 kw one decimal point could be used, from 100 kW 3 digits without decimal point and above 1000 kW it switches to gW with one or two decimal points.

andig, 2022-11-19:
We could also save space removing „mit/über evcc“- it‘s redundant.

RTTTC, 2022-11-28:
Or just remove animation - to make change fast? https://github.com/evcc-io/evcc/pull/3779
I still think that decimal points are not needed anymore, with larger user count.

PR #5495 (2022-12-18): Chore: Optimized JavaScript bundle
- Reduced JS bundle by 70kb (uncompressed)
- Use Bootstrap tree shaking
- Removed vue-number-animation & gsap dependencies. Using lighter countup instead
- Faster community number animations. Removed decimal from charging power. fixes #5212
- Add rollup-plugin-visualizer to see bundle contents (see asset-stats.html after UI build)
- Dependency maintenance (axios upgrade, remove dependency/devDependency distinction since we are no library)
closed 2022-12-18 by andig