evcc-io/evcc #5214: indent wrong in "evcc/templates/definition/meter/sungrow-hybrid.yaml"

elepete, 2022-11-19:
### Describe the bug

User of a Sungrow SH10RT where in the web IF the battery Watt data is always 0
Please help me to understand if the indent is perhaps wrong in file "evcc/templates/definition/meter/sungrow-hybrid.yaml"
in lines 60, 68 and 76 I would expect the indent of 2, as in all other lines.
Also register 13000 reads always 0, so I suspect this may be an issue as well, as the multiplier would be always 0.
Is there any way I can simply test, I did not succeed in generating a custom yaml without the sungrow hybrid template.
Anybody that has this working with a Hybrid of the SHxxRT series?

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
meters:
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.x.y (just to hide my private network)  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.x.y.  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.x.y  
  port: 502  
  usage: battery  
  modbus: tcpip  
  name: battery3
```

### Log details

```text
[main  ] INFO 2022/11/19 14:26:39 evcc 0.106.3
[main  ] INFO 2022/11/19 14:26:39 using config file: /home/peter/evcc.yaml
[main  ] INFO 2022/11/19 14:26:39 listening at :7070
[site  ] WARN 2022/11/19 14:26:42 vehicle 'generic 2-phase' does not support automatic detection
[site  ] INFO 2022/11/19 14:26:42 site config:
[site  ] INFO 2022/11/19 14:26:42   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2022/11/19 14:26:42     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2022/11/19 14:26:42     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2022/11/19 14:26:42     battery 1: power ✓ energy ✓ currents ✗ soc ✓
[site  ] INFO 2022/11/19 14:26:42   vehicles:
[site  ] INFO 2022/11/19 14:26:42     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗
[site  ] INFO 2022/11/19 14:26:42     vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2022/11/19 14:26:42 loadpoint 1:
[lp-1  ] INFO 2022/11/19 14:26:42   mode:        minpv
[lp-1  ] INFO 2022/11/19 14:26:42   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2022/11/19 14:26:42   meters:      charge ✗
[lp-1  ] DEBUG 2022/11/19 14:26:42 phase timer inactive
[lp-1  ] DEBUG 2022/11/19 14:26:42 pv timer inactive
[site  ] DEBUG 2022/11/19 14:26:42 ----
[lp-1  ] DEBUG 2022/11/19 14:26:42 charge power: 0W
[site  ] DEBUG 2022/11/19 14:26:42 pv power: 6612W
[site  ] DEBUG 2022/11/19 14:26:42 battery power: 0W
[site  ] DEBUG 2022/11/19 14:26:42 grid power: -3520W
[site  ] DEBUG 2022/11/19 14:26:42 battery soc 0: 99%
[site  ] DEBUG 2022/11/19 14:26:42 site power: -3520W
[lp-1  ] DEBUG 2022/11/19 14:26:42 charger status: A
[lp-1  ] INFO 2022/11/19 14:26:42 car disconnected
```

### What type of operating system are you running?

Linux

### Version

0.106.3

premultiply, 2022-11-19:
Use only the LAN-Port on the inverter itself.
Do NOT use WiFi or LAN on dongle!
The Dongle Firmware still has many bugs including the battery power direction is always zero. Well known bug.

closed 2022-11-19 by premultiply

elepete, 2022-11-19:
Thank you - works - and sorry that was not clear  to me