evcc-io/evcc #5225: "Estimate" Batterieladestand läuft weiter

markus80d, 2022-11-20:
### Describe the bug

Hallo,

ich nutzte EVCC unter Homeassistant in der aktuellsten Version 107.1 zusammen mit einem go-e-v3 und einem Seat Mii.

Mir ist nun schon mehrfach aufgefallen, auch mit vorigen evcc-Versionen, dass der angezeigte Ladestand des Autos nach dem Laden oder nach einer längeren Unterbrechung nicht passt und deutlich höher angezeigt wird als real in der Autoanzeige.

Das Auto blieb dabei z.B. über Nacht angesteckt. Anzeige in EVCC z.B. 92% obwohl nur gut 60% drin waren. Ich bilde mir auch ein mich zu erinnern, dass es auch während einem aktiven Ladevorgang mal auftrat und das Laden dann beendet wurde da der Zielladestand 90% überschritten war, aber real nur eben gut 60% drin waren.

Kann es sein dass die "ESTIMATE-Funktion" evtl. nicht korrekt funktioniert/rechnet oder sogar weiterläuft?

Es gab doch letztes auch mal ein anderes Problem mit den go-e's, vielleicht kann das ja auch damit zusammen hängen?

Ich habe mittlerweile den API-Abruf des Autos auf "connected" alle 120min. gestellt und dann hat es den Ladestand irgendwann wenigstens wieder runterkorrigiert von 90 auf 60% und ich konnte Weiterladen.

Solange aber der Ladestand falsch (zu hoch über/gleich Soll) ist wird aktuell natürlich dann das Laden beendet, obwohl das Auto bei weitem nicht nicht so weit geladen ist.

Könnte es evtl. sein, dass das auch durch das Update von 107.0 auf 107.1 kommt und dabei irgendwas passiert?

Danke für eure Hilfe!

VG Markus

### Steps to reproduce

1. Laden
2. Stoppen z.B. weil PV nix mehr bringt
3. Auto bleibt z.B. über Nacht angesteckt
4. Morgens auf EVCC gesehen, dass Ladestand von ca. 52 auf über 80% angestiegen ist ohne Laden

### Configuration details

```yaml
network:
  schema: http
  host: 000.000.00.00
  port: 7070

interval: 15s

log: info

chargers:
- name: garage
  type: template
  template: go-e-v3
  host: 000.000.000.00

loadpoints:
- title: Garage 
  charger: garage
  mode: pv
  vehicle: mii
  phases: 0
  enable:
    delay: 1m
    threshold: -1000
  disable:
    delay: 3m
    threshold: 700
  guardduration: 5m
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
  soc:
    poll:
      mode: connected
      interval: 120m
    estimate: true

meters:
- name: grid
  type: movingaverage
  decay: 0.25
  meter:
    type: template
    template: powerfox-poweropti
    usage: grid
    user: *****
    pass *****

site:
  title: Laderegelung E-Auto
  meters:
    grid: grid
  residualPower: 0

vehicles:
  - name: mii
    type: template
    template: seat
    title: Seat Mii electric
    user: marku***
    password: ****
    phases: 2
    capacity: 32
    minSoC: 15
    targetSoC: 90

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.28
  feedin:
    type: fixed
    price: 0.145

sponsortoken: ****

telemetry: true
plant: ****
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'evcc --config /config/evcc.yaml --sqlite /data/evcc.db'
[main  ] INFO 2022/11/20 04:44:02 evcc 0.107.1
[main  ] INFO 2022/11/20 04:44:02 using config file: /config/evcc.yaml
[main  ] INFO 2022/11/20 04:44:02 starting ui and api at :7070
[lp-1  ] WARN 2022/11/20 04:44:04 poll mode '{connected 2h0m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[site  ] INFO 2022/11/20 04:44:04 site config:
[site  ] INFO 2022/11/20 04:44:04   meters:      grid ✓ pv ✗ battery ✗
[site  ] INFO 2022/11/20 04:44:04     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/11/20 04:44:04   vehicles:
[site  ] INFO 2022/11/20 04:44:04     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗
[lp-1  ] INFO 2022/11/20 04:44:04 loadpoint 1:
[lp-1  ] INFO 2022/11/20 04:44:04   mode:        pv
[lp-1  ] INFO 2022/11/20 04:44:04   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2022/11/20 04:44:04   meters:      charge ✓
[lp-1  ] INFO 2022/11/20 04:44:04     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2022/11/20 04:44:04 vehicle updated: unknown -> Seat Mii electric
[lp-1  ] INFO 2022/11/20 04:44:06 car connected
[lp-1  ] INFO 2022/11/20 10:07:50 start charging ->
```

### What type of operating system are you running?

Linux

### Version

0.107.1

premultiply, 2022-11-20:
Ohne aussagekräftiges Logfile nicht zu analysieren. Eine grundsätzliche Fehlfunktion des Estimators halte ich aber für nicht naheliegend.
Ggf. `estimator: false` setzen.

closed 2022-11-20 by premultiply

markus80d, 2022-12-05:
Hi premultiply nochmal, der estimator produziert bei mir eigentlich jede Woche mal grob falsche Ladestände. Aber ich bekomme noch nicht so richtig zu fassen warum, wann und wie...

Darf ich mich hier nochmal melden wenn ich's konkretisieren kann?

Im Moment habe ich daher leider estimate=false gesetzt wie vorgeschlagen. Möchte das aber in Zukunft schon wieder aktivieren...

VG Markus

premultiply, 2022-12-05:
Sicher.
Der Estimator verlässt sich auf die von der Wallbox gemessene Ladeenergiemenge (sofern verfügbar).
Wenn damit alles in Ordnung ist bin ich relativ sicher dass der Estimator korrekt funktioniert.

Da du aber einen go-e-Charger hast ist die Wahrscheinlichkeit hoch, dass genau dort das Problem liegt und das Teil auch bei dir nicht richtig zählt, rebootet, interne Überläufe produziert oder nicht erreichbar ist.

markus80d, 2022-12-06:
OK, interessant. Das kann ich leider nicht beurteilen ob der go-e korrekte Daten liefert. Die Leistung liefert er das sieht man ja auf der Oberfläche, aber ob sonst alles korrekt läuft... Da hast du mehr Erfahrung :-)... Belassen wir es erst mal dabei. VG Markus

premultiply, 2022-12-06:
Die Wahrheit steckt wie immer in einem passenden Logfile. ;)

markus80d, 2022-12-06:
Da ist bei mir aktuell der Haken. Entweder bin ich zu doof, aber ich weiß nicht wo das Logfile beim Homeassistant EVCC-Addon liegt. Im Addon-Bereich im HA werden immer nur die letzten Einträge unter Log angezeigt, aber kein komplettes File. Kannst du mir da weiterhelfen wo das ist? Dann könnte ich das mit dem Estimator evtl. mal tracken/loggen. Log=debug reicht, oder?