evcc-io/evcc #5233: EVCC 107.1 does not run under Homeassistant anymore

ThomasWollstadt, 2022-11-21:
106.5 did run. automatic update to 107.1 casuses EVCC to not start. No change to Yaml or other files...

log inside homeassitant:

Fehler beim Starten
cannot create charger 'go-e-charger'
cannot create charger 'template'
cannot create charger 'go-e'
sponsorship required, see https://github.com/evcc-io/evcc#sponsorship

premultiply, 2022-11-21:
Steht alles da:

> cannot create charger 'go-e'
sponsorship required, see https://github.com/evcc-io/evcc#sponsorship

closed 2022-11-21 by premultiply

premultiply, 2022-11-21:
https://github.com/evcc-io/evcc/discussions/5122

ThomasWollstadt, 2022-11-21:
Ok. Das hatte ich nicht verfolgt und daher nicht mitbekommen...Gerne werde ich ein Sponsor da EVCC das verdient hat.Holen Sie sich Outlook für Android