evcc-io/evcc #5243: Sessions: odometer is empty

andig, 2022-11-21:
Wenn ich das richtig verstanden habe, dann sollte in der CSV-Datei der Ladevorgänge doch der Kilometerstand mit aufgeführt sein.
Die Spalte "Odometer" ist vorhanden, aber bei mir steht immer  0  drin.
Version 0.107.1

_Originally posted by @VolkerK62 in https://github.com/evcc-io/evcc/discussions/5240_

andig, 2022-11-21:
Mhhm- working for me when testing. That's bad :O

andig, 2022-11-22:
cache: trace

wäre gut, da sieht man die aktuellen Werte fürs Odometer. Spannend wäre auch mal in die SQlite Datei reinzuschauen wenn Auto anschlossen und km-Stand ausgelesen wurde- steht dann der richtige Wert drin? Dann wüssten wir schonmal ob er gar nicht rein kommt oder später überschrieben wird.

VolkerK62, 2022-11-22:
wenn du mir sagst, wo ich die SQlite Datei finde ;-)

VolkerK62, 2022-11-22:
Also, ich habe es genau so https://github.com/evcc-io/evcc/discussions/5159#discussioncomment-4145827 gemacht.
Allerdings wird mir nach "select" nichts angezeigt.

auch mit `sudo nano evcc.db` bekomme ich nur eine leere Datei angezeigt

andig, 2022-11-22:
ls -la ~/.evcc

andig, 2022-11-22:
Da liegt eine Datei und hat die richtige Größe. Also:

    sqlite3 ~/.evcc/evcc.db

Dann

    select * from sessions;

RTTTC, 2022-11-22:
Same here, Enyaq provides odometer data,  but it is not in CSV file, (0)

VolkerK62, 2022-11-22:
das ist der Outcome
```
pi@raspberrypi:~ $ sqlite3 ~/.evcc/evcc.db
SQLite version 3.34.1 2021-01-20 14:10:07
Enter ".help" for usage hints.
sqlite> select * from sessions;
sqlite> 
```

VolkerK62, 2022-11-22:
Ich habe im loglevel `db: trace` und `cache: trace` eingetragen, evcc neu gestartet und die Ladung gestartet. 

Das kommt bei `start charging`
```
Nov 22 14:32:20 raspberrypi evcc[4625]: [db    ] TRACE 2022/11/22 14:32:20 INSERT INTO `sessions` (`created`,`finished`,`loadpoint`,`identifier`,`vehicle`,`odometer`,`meter_start_kwh`,`meter_end_kwh`,`charged_kwh`) VALUES ("2022-11-22 14:32:20.492","0000-00-00 00:00:00","Carport","","(T)Raumschiff",0.000000,1148.599976,0.000000,0.000000) RETURNING `id` 1 <nil>
```
und das bei `stop charging`
```
Nov 22 14:34:40 raspberrypi evcc[4625]: [db    ] TRACE 2022/11/22 14:34:40 UPDATE `sessions` SET `created`="2022-11-22 14:32:20.492",`finished`="2022-11-22 14:34:40.493",`loadpoint`="Carport",`identifier`="",`vehicle`="(T)Raumschiff",`odometer`=0.000000,`meter_start_kwh`=1148.599976,`meter_end_kwh`=1148.699951,`charged_kwh`=0.099976 WHERE `id` = 50 1 <nil>
```

im gesamten zeitraum gibt es diese `odometer`Einträge im log:
```
Nov 22 14:31:30 raspberrypi evcc[4625]: [lp-1  ] DEBUG 2022/11/22 14:31:30 vehicle odometer: 11242km
Nov 22 14:31:30 raspberrypi evcc[4625]: [cache ] TRACE 2022/11/22 14:31:30 lp-1/vehicleOdometer: 11241.8
Nov 22 14:31:44 raspberrypi evcc[4625]: [lp-1  ] DEBUG 2022/11/22 14:31:44 vehicle odometer: 11242km
Nov 22 14:31:44 raspberrypi evcc[4625]: [cache ] TRACE 2022/11/22 14:31:44 lp-1/vehicleOdometer: 11241.8

```
log-file ist gespeichert. Kann ich problemlos komplett einstellen.

thierolm, 2022-11-22:
Hi, kann es sein, dass in
https://github.com/evcc-io/evcc/blob/c87df312c5475304b0ce091cf2ae13e39ad30219/core/db/session.go#L35
das Setzen des Odometer-Wertes fehlt?

ammann-ruchti, 2022-11-27:
Meine evcc.db ist 12288 Bytes gross und enthält keine sessions Tabelle. Auch bei mir ist im csv-Export die odometer-Kolonne weiterhin mit Nullen gefüllt statt mit dem km-Stand. Die Fahrzeugabfrage meines Enyaq z. B. mittels evcc vehicle --name ev4 liefert den korrekten odo-Wert. Stand evcc V0.108.2.

andig, 2022-11-27:
Wenn keine Sessions in der Tabelle sind, Du im UI aber welche siehst dann schaust Du in die falsche Datei.

ammann-ruchti, 2022-11-27:
Ok, danke. Ich habe die "richtige" unter /root/.evcc/ gefunden. Die Sessions sind drin, offensichtlich ohne ODO-Meter Eintrag. Hier z.B. die neueste Zeile:
19|2022-11-26 23:38:40.476836317+01:00|2022-11-26 23:40:20.529597541+01:00|Carport||Enyaq iV 80|0.0|0.0|0.183867491552833|0.0

RTTTC, 2022-11-27:
@ammann-ruchti This issue is simply not fixed yet.

VolkerK62, 2022-11-27:
Ah, ok. Wenn man am richtigen Ort guckt ;-)
Aber, Wunderheilung? Seit heute morgen ist ODO vorhanden, auch in CSV.
Version ist immer noch (seit 2 Tagen) = 0.108.0

```
50|2022-11-22 14:32:20.492229914+01:00|2022-11-22 15:31:30.309758622+01:00|Carport||(T)Raumschiff|1148.59997558594|1148.80004882813|0.2000732421875|0.0
51|2022-11-23 04:51:40.226937448+01:00|2022-11-23 15:14:30.473572772+01:00|Carport||(T)Raumschiff|1148.80004882813|1157.30004882813|8.5|0.0
52|2022-11-24 09:37:57.078750584+01:00|2022-11-24 15:26:37.160526719+01:00|Carport||(T)Raumschiff|1157.30004882813|1167.80004882813|10.5|0.0
53|2022-11-25 14:00:52.96866155+01:00|2022-11-25 14:10:24.696947815+01:00|Carport||(T)Raumschiff|1167.80004882813|1167.90002441406|0.0999755859375|0.0
54|2022-11-26 09:11:39.959605852+01:00|2022-11-26 11:21:37.355765103+01:00|Carport||(T)Raumschiff|1167.90002441406|1169.80004882813|1.9000244140625|0.0
55|2022-11-26 13:16:10.651176001+01:00|2022-11-27 10:34:40.957661049+01:00|Carport||(T)Raumschiff|1169.80004882813|1170.30004882813|0.5|11295.6
```

thierolm, 2022-11-27:
Faszinierend ... :-)

ammann-ruchti, 2022-11-27:
Ja, leider sieht das bei mir nicht so positiv aus (keine Wundheilung bei mir). Es fehlen in der evcc.db weiterhin die Daten über kWh-Stand am Anfang und Ende der Session wie auch der ODO-Meter-Stand. Hier der letzte Eintrag nach Neuverbinden des Fz und Neustart von evcc: 21|2022-11-27 11:22:54.740524571+01:00|2022-11-27 11:23:44.645679144+01:00|Carport||Enyaq iV 80|0.0|0.0|0.137576168159333|0.0

VolkerK62, 2022-11-27:
zu früh gefreut...
Auto abgesteckt, neu angesteckt, log level geändert, neu gestartet und dann kurz mal geladen .... und schon ist kein mehr ODO drin

```
55|2022-11-26 13:16:10.651176001+01:00|2022-11-27 11:18:40.455825059+01:00|Carport||(T)Raumschiff|1169.80004882813|1170.90002441406|1.0999755859375|11295.6
56|2022-11-27 11:24:54.558592866+01:00|2022-11-27 11:27:12.255589209+01:00|Carport||(T)Raumschiff|1170.90002441406|1171.09997558594|0.199951171875|0.0
```

thierolm, 2022-11-27:
Wieviele Loadpoints habt ihr denn definiert? >1?

VolkerK62, 2022-11-27:
ich hab nur einen.

thierolm, 2022-11-27:
OK, dann liegt die Ursache nicht am SQLite Modul.

ammann-ruchti, 2022-11-27:
Auch ich habe nur einen Ladepunkt...

VolkerK62, 2022-11-27:
ist schon irgendwie verhext ... mal isser drin und mal nicht
```
53|2022-11-25 14:00:52.96866155+01:00|2022-11-25 14:10:24.696947815+01:00|Carport||(T)Raumschiff|1167.80004882813|1167.90002441406|0.0999755859375|0.0
54|2022-11-26 09:11:39.959605852+01:00|2022-11-26 11:21:37.355765103+01:00|Carport||(T)Raumschiff|1167.90002441406|1169.80004882813|1.9000244140625|0.0
55|2022-11-26 13:16:10.651176001+01:00|2022-11-27 11:18:40.455825059+01:00|Carport||(T)Raumschiff|1169.80004882813|1170.90002441406|1.0999755859375|11295.6
56|2022-11-27 11:24:54.558592866+01:00|2022-11-27 12:03:14.767843601+01:00|Carport||(T)Raumschiff|1170.90002441406|1171.09997558594|0.199951171875|0.0
57|2022-11-27 12:14:48.562046602+01:00|2022-11-27 13:03:02.546702037+01:00|Carport||(T)Raumschiff|1171.09997558594|1171.5|0.4000244140625|0.0
58|2022-11-27 13:03:11.112025885+01:00|2022-11-27 13:03:38.29465079+01:00|Carport||(T)Raumschiff|1171.5|1171.5|0.0|11295.6
59|2022-11-27 13:04:12.402413518+01:00|2022-11-27 13:10:38.73241158+01:00|Carport||(T)Raumschiff|1171.5|1171.5|0.0|11295.6
60|2022-11-27 13:11:05.766973243+01:00|2022-11-27 13:12:50.172710998+01:00|Carport||unavailable|1171.5|1171.5|0.0|0.0
61|2022-11-27 13:12:59.07407612+01:00|2022-11-27 13:17:18.106340705+01:00|Carport||(T)Raumschiff|1171.5|1171.5|0.0|11295.6
62|2022-11-27 13:17:26.546925532+01:00|2022-11-27 14:01:06.80289779+01:00|Carport||(T)Raumschiff|1171.5|1171.59997558594|0.0999755859375|11295.6
``

int5749, 2022-11-27:
@andig 
Wann  wird denn 'odometer' in die DB geschrieben?
Ich habe mir gerade mal die 'db.go' und 'session.go' angeschaut, dort wird 'odometer' definiert, aber nach meinem Verständnis nicht in die DB geschrieben? Oder erfolgt dies an anderer Stelle?

Viele Grüße

andig, 2022-11-27:
Im Loadpoint nach Fahrzeugwechsel

andig, 2022-11-27:
Idee: wir können da ja mal Logging rein machen um zu verstehen, wo die Werte verschwinden?

int5749, 2022-11-27:
> Idee: wir können da ja mal Logging rein machen um zu verstehen, wo die Werte verschwinden?

Ich kann ja leider nicht selber kompilieren :-/ werde aber heute Nacht das Auto noch laden.
Derzeit ist unser Fiat 500 angeschlossen, MQTT zeigt odometer Werte an, aber im Session Table auch bei mir auf 0
Mittlerweile läuft evcc bei mir manuell installiert auf einem Linux (Debian)

Wenn ich irgendwie unterstützen kann, sehr gerne.

ammann-ruchti, 2022-11-27:
> Idee: wir können da ja mal Logging rein machen um zu verstehen, wo die Werte verschwinden?

Wenn du mir angeben kannst, wie ich das gewünschte Logging aktiviere, mache ich das morgen gerne.

int5749, 2022-11-28:
@andig 
Kann es evtl. mit der Kennung zusammenhängen? Denn diese ist z.B. beim Fiat auch leer und bei Dir hatte es ja in den Tests funktioniert (weiß aber nicht, welches Auto Du nutzt).

ammann-ruchti, 2022-11-28:
@andig 
Wenn ich es richtig verstehe, müsste ich in der evcc.yaml unter log / levels noch den level enyac: trace einfügen? Oder reicht debug schon?

andig, 2022-11-28:
Wir haben hier leider keinen Erkenntnisgewinn. Was wir brauchen ist ein Logfile 

    db: trace
    lp-1: debug

 aus dem ersichtlich wird, ob und wo der Kilometerstand geschrieben und schlimmstenfalls wieder auf 0 gesetzt wird. Im Nightly ab morgen sind dafür nochmal drei Ergänzungen drin wo man Session Start/Stop/Update sieht.

ammann-ruchti, 2022-11-28:
ok, mach ich morgen mit dem neuen Nightly

int5749, 2022-11-28:
> Wir haben hier leider keinen Erkenntnisgewinn. Was wir brauchen ist ein Logfile
> 
> ```
> levels
>   db: trace
>   lp-1: debug
> ```
> 
> aus dem ersichtlich wird, ob und wo der Kilometerstand geschrieben und schlimmstenfalls wieder auf 0 gesetzt wird. Im Nightly ab morgen sind dafür nochmal drei Ergänzungen drin wo man Session Start/Stop/Update sieht.

u got mail ;-)

ammann-ruchti, 2022-11-29:
@thierolm 
Hast Du dieses Problem heute morgen gelöst? Dann wäre es im Nightly noch nicht behoben...

int5749, 2022-11-29:
> @thierolm Hast Du dieses Problem heute morgen gelöst? Dann wäre es im Nightly noch nicht behoben...

Laut Post vor 14h, wurde aber noch nicht ge-merged. Somit wird es noch nicht enthalten sein.
Somit macht ein weiterer Test offensichtlich erst nach dem merge Sinn.

int5749, 2022-12-01:
Kurzes Update: Fix ist noch nicht ge-merged, daher auch noch nicht in der 108.3 enthalten.

@andig 
Werden noch weitere Logs it dem "erweiterten" logging benötigt? Nicht, das Du auf weitere Logs wartest, und wir keine erstellen, da es evtl. schon gefixed aber nicht gemerged ist ;-)

Bitte nicht falsch verstehen: Ist nicht als "pushy" zu werten, möchte nur verhindern, dass der Eine auf den Anderen wartet ;-)

andig, 2022-12-01:
Keine Logs notwendig, Problem ist dank @thierolm erkannt.

andig, 2022-12-03:
Closed in https://github.com/evcc-io/evcc/commit/42ca83d3c0b5d6e4ccc67a5754b7889bad8ca14a. Hoffe es funktioniert jetzt, Nightly build ist angestossen und in 15min fertig.

closed 2022-12-03 by andig

int5749, 2022-12-03:
> Closed in https://github.com/evcc-io/evcc/commit/42ca83d3c0b5d6e4ccc67a5754b7889bad8ca14a. Hoffe es funktioniert jetzt, Nightly build ist angestossen und in 15min fertig.

Werde wahrscheinlich morgen testen können.

int5749, 2022-12-04:
> Closed in [42ca83d](https://github.com/evcc-io/evcc/commit/42ca83d3c0b5d6e4ccc67a5754b7889bad8ca14a). Hoffe es funktioniert jetzt, Nightly build ist angestossen und in 15min fertig.

Getestet und => bei mir hat dies nun funktioniert 👍 

[image]
Nach dem anschließen (ohne Ladevorgang) war der Wert für Odometer bereits gefüllt

#off-topic Erstaunlicherweise wurde an dem EV Charger sogar direkt der Fiat 500 erkannt, nicht "nur" ein Gastfahrzeug, später mal sehen ob dies an der WARP Pro ebenfalls so ist.

[image]
Nach Abschluss des Ladevorgangs war dieser im CSV mit dem Kilometerstand eingetragen. 💯 

DANKE

andig, 2022-12-04:
Die Blumen gebühren @thierolm!

VolkerK62, 2022-12-04:
>Getestet und => bei mir hat dies nun funktioniert 

confirm

Fleurop für @thierolm  🥇

ammann-ruchti, 2022-12-04:
Ich bin noch unterwegs und werde am Dienstag testen. Das wird sicher bei mir auch laufen.

Bereits jetzt allen Beteiligten herzlichen Dank für eure Mithilfe beim Lösen des Problems.