evcc-io/evcc #5253: HEMS SMA doesn't enable evcc

Percy2Live, 2022-11-23:
### Describe the bug

I configured my evcc so it shows up in the SMA sunny portal, in order to allow evcc to load with a shared ration of grid-energy.
I used this configuration in my evcc.yaml:

```
hems:
  type: sma
  allowcontrol: true
  vendorid: ABCAABAA
  deviceid: ABCBBBABBBBB
```
Afterwards I was able to connect evcc in the SMA sunny portal.
Unfortunately SMA does never enable the wallbox. I always see this infobox in evcc, even when there should be enough energy:

[image]

I tried to change the energy ration in the sunny portal, but even with 100% grid energy the infobox in evcc doesn't change:

[image]

evcc is listed as a non-manually controlable consumer in the sunny portal. It seems that evcc does not inform sunny portal when it has a need to charge, and therefore the sunny portal is never sending the release signal.

### Steps to reproduce

Follow the instructions about using HEMS SMA

### Configuration details

```yaml
uri: 192.168.xxx.xxx:7070 # uri for ui
interval: 10s # control cycle interval
sponsortoken: xxx

log: debug

meters:

- name: Homemanager
  type: sma
  serial: xxx# insert your own serialnumber
  
#- name: Homemanager
#  type: template
#  template: sma-home-manager
#  usage: grid
#  host: 192.168.xxx.xxx # IP-Adresse oder Hostname 
    
- name: Wechselrichter
  type: template
  template: sma-inverter
  usage: pv
  host: 192.168.xxx.xxx # IP-Adresse oder Hostname
  password: xxx# Anlagenpasswort / Installateurspasswort # Optional 

chargers:
- type: template
  template: go-e-v3 
  host: xxx
  name: wbPascal
- type: template
  template: go-e-v3 
  host: xxx
  name: wbSandra

vehicles:
- type: template
  template: nissan 
  title: Leaf Pascal 
  user: xxx  
  password: xxx
  vin: xxx
  capacity: 40  
  name: ev1 

loadpoints:
- title: Ladepunkt Sandra
  charger: wbSandra
  vehicles: 
  - ev1
  mode: pv
#  mode: minpv #In Kombination mit den Sunny Portal Einstellungen kann in diesem Modus auch mit reduziertem PV Anteil geladen werden, z.B. mit 30% Netzbezug
  resetOnDisconnect: true
  phases: 3 # ev phases (default 3)
  enable: # pv mode enable behavior
    delay: 5m # threshold must be exceeded for this long
    threshold: -1600 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 10m # threshold must be exceeded for this long
    threshold: 300 # maximum import power (W)
  guardDuration: 10m # switch charger contactor not more often than this (default 5m)
  minCurrent: 6 # minimum charge current (default 6A)
  maxCurrent: 32 # maximum charge current (default 16A)
  soc:
    poll:
      mode: charging
      interval: 60m
    min: 0
    target: 100
    estimate: true

- title: Ladepunkt Pascal
  charger: wbPascal
  vehicles: 
  - ev1
  mode: pv
  resetOnDisconnect: true
  phases: 3 # ev phases (default 3)
  enable: # pv mode enable behavior
    delay: 5m # threshold must be exceeded for this long
    threshold: -1600 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 10m # threshold must be exceeded for this long
    threshold: 300 # maximum import power (W)
  guardDuration: 10m # switch charger contactor not more often than this (default 5m)
  minCurrent: 6 # minimum charge current (default 6A)
  maxCurrent: 32 # maximum charge current (default 16A)
  soc:
    poll:
      mode: charging
      interval: 60m
    min: 0
    target: 100
    estimate: true

site:
  title: Home Ladestation # display name for UI
  meters:
    grid: Homemanager # grid meter
    pvs:
      - Wechselrichter # list of pv inverters/ meters
    battery: # battery meter
  prioritySoC: # give home battery priority up to this soc (empty to disable)
  bufferSoC: # ignore home battery discharge above soc (empty to disable)
  residualPower: #Wintermodus: Wie viel Netzbezug ist ok, damit trotzdem geladen wird (z.B. -1000W)

hems:
  type: sma
  allowcontrol: true
  vendorid: ABCAABAA
  deviceid: ABCBBBABBBBB

# tariffs are the fixed or variable tariffs
# cheap (tibber/awattar) can be used to define a tariff rate considered cheap enough for charging
tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    # either static grid price
    type: fixed
    price: 0.264 # EUR/kWh
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.127 # EUR/kWh
influx:
  url: http://xxx:8086
  database: evcc
  user: xxx
  password: xxx
```

### Log details

```text
/app # evcc --log debug
[main  ] INFO 2022/11/23 11:21:56 evcc 0.106.2
[main  ] INFO 2022/11/23 11:21:56 using config file: /etc/evcc.yaml
[main  ] WARN 2022/11/23 11:21:56 `uri` is deprecated and will be ignored. Use `network` instead.
[main  ] INFO 2022/11/23 11:21:56 starting ui and api at :7070
[lp-1  ] WARN 2022/11/23 11:22:05 Configuring soc.target at loadpoint is deprecated and must be applied per vehicle
[lp-1  ] WARN 2022/11/23 11:22:05 vehicles option is deprecated
[lp-1  ] WARN 2022/11/23 11:22:05 locking phase config to 3p for switchable charger
[lp-2  ] WARN 2022/11/23 11:22:05 Configuring soc.target at loadpoint is deprecated and must be applied per vehicle
[lp-2  ] WARN 2022/11/23 11:22:05 vehicles option is deprecated
[semp  ] WARN 2022/11/23 11:22:05 SEMP_BASE_URL unspecified, using http://192.168.101.30:7070 instead
[site  ] INFO 2022/11/23 11:22:05 site config:
[site  ] INFO 2022/11/23 11:22:05   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2022/11/23 11:22:05     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2022/11/23 11:22:05     pv 1:      power ✓ energy ✓ currents ✓
[site  ] INFO 2022/11/23 11:22:05   vehicles:
[site  ] INFO 2022/11/23 11:22:05     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
[lp-1  ] INFO 2022/11/23 11:22:05 loadpoint 1:
[lp-1  ] INFO 2022/11/23 11:22:05   mode:        pv
[lp-1  ] INFO 2022/11/23 11:22:05   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2022/11/23 11:22:05   meters:      charge ✓
[lp-1  ] INFO 2022/11/23 11:22:05     charge:    power ✓ energy ✓ currents ✓
[lp-2  ] INFO 2022/11/23 11:22:05 loadpoint 2:
[lp-2  ] INFO 2022/11/23 11:22:05   mode:        pv
[lp-2  ] INFO 2022/11/23 11:22:05   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-2  ] INFO 2022/11/23 11:22:05   meters:      charge ✓
[lp-2  ] INFO 2022/11/23 11:22:05     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] DEBUG 2022/11/23 11:22:05 phase timer inactive
[lp-1  ] DEBUG 2022/11/23 11:22:05 pv timer inactive
[lp-1  ] DEBUG 2022/11/23 11:22:05 max charge current: 6A
[lp-2  ] DEBUG 2022/11/23 11:22:05 phase timer inactive
[lp-2  ] DEBUG 2022/11/23 11:22:05 pv timer inactive
[influx] ERROR 2022/11/23 11:22:06 write failed (attempts 0): 400 Bad Request: field type conflict
[lp-2  ] ERROR 2022/11/23 11:22:06 charger: Get "http://192.168.101.153/status": dial tcp 192.168.101.153:80: connect: no route to host
[site  ] DEBUG 2022/11/23 11:22:06 ----
[main  ] FATAL 2022/11/23 11:22:06 listen tcp :7070: bind: address already in use
```

### What type of operating system are you running?

Docker container

### Version

0.106.2

premultiply, 2022-11-23:
You can verify what evcc offers to SHM by calling http://evcc.local:7070/semp

Percy2Live, 2022-11-23:
This is the output of the simple energy management protocol:

```
<Device2EM xmlns="http://www.sma.de/communication/schema/SEMP/v1">
<DeviceInfo>
<Identification>
<DeviceId>F-ABCAAAAA-abcbbbbbbbbb-00</DeviceId>
<DeviceName>Ladepunkt Sandra</DeviceName>
<DeviceType>EVCharger</DeviceType>
<DeviceSerial>123456781122-0</DeviceSerial>
<DeviceVendor>github.com/evcc-io/evcc</DeviceVendor>
</Identification>
<Characteristics>
<MinPowerConsumption>1380</MinPowerConsumption>
<MaxPowerConsumption>7360</MaxPowerConsumption>
</Characteristics>
<Capabilities>
<CurrentPower>
<Method>Measurement</Method>
</CurrentPower>
<Timestamps>
<AbsoluteTimestamps>false</AbsoluteTimestamps>
</Timestamps>
<Interruptions>
<InterruptionsAllowed>true</InterruptionsAllowed>
</Interruptions>
<Requests>
<OptionalEnergy>true</OptionalEnergy>
</Requests>
</Capabilities>
</DeviceInfo>
<DeviceInfo>
<Identification>
<DeviceId>F-ABCAAAAA-abcbbbbbbbbc-00</DeviceId>
<DeviceName>Ladepunkt Pascal</DeviceName>
<DeviceType>EVCharger</DeviceType>
<DeviceSerial>123456781122-1</DeviceSerial>
<DeviceVendor>github.com/evcc-io/evcc</DeviceVendor>
</Identification>
<Characteristics>
<MinPowerConsumption>1380</MinPowerConsumption>
<MaxPowerConsumption>22080</MaxPowerConsumption>
</Characteristics>
<Capabilities>
<CurrentPower>
<Method>Measurement</Method>
</CurrentPower>
<Timestamps>
<AbsoluteTimestamps>false</AbsoluteTimestamps>
</Timestamps>
<Interruptions>
<InterruptionsAllowed>true</InterruptionsAllowed>
</Interruptions>
<Requests>
<OptionalEnergy>true</OptionalEnergy>
</Requests>
</Capabilities>
</DeviceInfo>
<DeviceStatus>
<DeviceId>F-ABCAAAAA-abcbbbbbbbbb-00</DeviceId>
<EMSignalsAccepted>true</EMSignalsAccepted>
<Status>Off</Status>
<PowerConsumption>
<PowerInfo>
<AveragePower>0</AveragePower>
<Timestamp>0</Timestamp>
<AveragingInterval>60</AveragingInterval>
</PowerInfo>
</PowerConsumption>
</DeviceStatus>
<DeviceStatus>
<DeviceId>F-ABCAAAAA-abcbbbbbbbbc-00</DeviceId>
<EMSignalsAccepted>false</EMSignalsAccepted>
<Status>Off</Status>
<PowerConsumption>
<PowerInfo>
<AveragePower>0</AveragePower>
<Timestamp>0</Timestamp>
<AveragingInterval>60</AveragingInterval>
</PowerInfo>
</PowerConsumption>
</DeviceStatus>
<PlanningRequest>
<Timeframe>
<DeviceId>F-ABCAAAAA-abcbbbbbbbbb-00</DeviceId>
<EarliestStart>0</EarliestStart>
<LatestEnd>86400</LatestEnd>
<MinEnergy>0</MinEnergy>
<MaxEnergy>37095</MaxEnergy>
<MaxPowerConsumption>7360</MaxPowerConsumption>
<MinPowerConsumption>1380</MinPowerConsumption>
</Timeframe>
</PlanningRequest>
</Device2EM>
```

premultiply, 2022-11-23:
Es muss der Min+PV-Modus gewählt sein damit der SHM steuern kann.

Percy2Live, 2022-11-24:
Das ist mir bekannt, unabhängig vom gewählten Modus bleibt die Infobox aber immer auf folgendem Wert stehen, auch im Min+PV Modus:

[image]

premultiply, 2022-11-24:
Da hilft ggf. mal ein tracelog der Kommunikation mit dem SHM.
In der Regel ist es aber wirklich so dass der SHM die Ladung sperrt.
Warum auch immer.
Manchmal hilft auch ein Reboot des SHM.
Das ist das gleiche Problem an dem auch der SMA EV Charger krankt und er nicht lädt als lädt.
Da evcc das alleine ohnehin quasi alles besser macht ist unsere Empfehlung `allowcontrol: false` zu belassen.
Es gibt nur sehr wenige Szenarien (z. B. Verbraucherprioritäten) bei denen eine Mitsteuerung durch den SHM sinnvoll sein kann.
Die Bedarfe werden von evcc ja immer korrekt gemeldet. Das ist das Wesentliche.

koaschten, 2022-11-24:
eventuell hilft die reset prozedur wie hier beschrieben? https://github.com/evcc-io/evcc/discussions/4305#discussioncomment-3609840

Percy2Live, 2022-11-24:
Danke für den Hinweis mit dem Reboot, ich werde das mal versuchen und die Tage bei entsprechendem Wetter beobachten. Kann es sein, dass SMA die Verbraucherplanung 24h im Voraus durchführt? Das könnte erklären, warum ich beim Testen keinerlei Effekt am jeweiligen Tag feststellen kann.
Nutzen wollte ich die Verbrauchersteuerung um jetzt in der kalten Jahreszeit die PV/Grid Ratio beim Laden anzupassen. Das geht zwar auch in evcc über den ResidualPower Parameter, allerdings muss ich hier bei einer Anpassung evcc rebooten. Über das Sunny-Portal kann ich eine Änderung der Ratio einfach im laufenden Betrieb vornehmen

premultiply, 2022-11-24:
Du kannst diese Einstellungen auch bei evcc jederzeit ohne Neustart durchführen.
Das geht über die API z.B. per HTTP oder MQTT.

closed 2022-12-03 by andig