evcc-io/evcc #5257: Vehicle error: invalid character 'u' looking for beginning of object key string

schenlap, 2022-11-23:
### Describe the bug

Vehicle SOC wird nicht mehr gelesen. Getestet mit audi & enyaq.

### Steps to reproduce

1. Fehler ist permanent seit update von 0.106.3 auf 0.107.1
2. (datum ist dieses mal richtig, daran liegt es nicht :-)  )
3.
...

### Configuration details

```yaml
vehicles:
- name: audi
  type: audi
  title: A3 e-tron
  capacity: 7.5 # kWh
  user: xxx
  password: xxx
  vin: xxx
  cache: 5m
  onIdentify:
    minSoC: 10 # charge to at least 10% independent of charge mode
    targetSoC: 90 # limit charge to 90%
    mode: pv
- name: enyaq
  type: enyaq
  title: Enyaq iV
  capacity: 77 # kWh
  user: xxx
  password: xxx
  vin: xxx
  cache: 30m
  onIdentify:
    minSoC: 15 # charge to at least 10% independent of charge mode
    targetSoC: 80 # limit charge to 90%
```

### Log details

```text
[main  ] INFO 2022/11/23 19:15:32 evcc 0.107.1
[main  ] INFO 2022/11/23 19:15:32 using config file: /etc/evcc.yaml
[main  ] ERROR 2022/11/23 19:15:33 creating vehicle audi failed: cannot create vehicle 'audi': invalid character 'u' looking for beginning of object key string
[main  ] ERROR 2022/11/23 19:15:33 creating vehicle enyaq failed: cannot create vehicle 'enyaq': invalid character 'u' looking for beginning of object key string
audi
----
.SoC:     vehicle not available: cannot create vehicle 'audi': invalid character 'u' looking for beginning of object key string
Capacity: 0.0kWh

enyaq
-----
.SoC:     vehicle not available: cannot create vehicle 'enyaq': invalid character 'u' looking for beginning of object key string
Capacity: 0.0kWh
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.107.1

ABBolle, 2022-11-24:
Gilt ebenso für VW: https://github.com/evcc-io/evcc/discussions/5258#discussion-4595619

Zugangsdaten sind unterwegs.

StevieC121176, 2022-11-24:
Bei mir seit der neuesten Nightly heute morgen auch bei Skoda und Audi

```pi@raspberrypi:~ $ evcc vehicle
[main  ] INFO 2022/11/24 07:09:27 evcc 0.107.1 (c87df312)
[main  ] INFO 2022/11/24 07:09:27 using config file: /etc/evcc.yaml
[main  ] ERROR 2022/11/24 07:09:28 creating vehicle ev5 failed: cannot create vehicle 'template': cannot create vehicle 'audi': invalid character 'u' looking for beginning of object key string
[main  ] ERROR 2022/11/24 07:09:28 creating vehicle ev6 failed: cannot create vehicle 'template': cannot create vehicle 'enyaq': invalid character 'u' looking for beginning of object key string
[main  ] ERROR 2022/11/24 07:09:28 creating vehicle ev7 failed: cannot create vehicle 'template': cannot create vehicle 'skoda': invalid character 'u' looking for beginning of object key string
ev8
---
SoC:          0%
Capacity:     0.8kWh
OnIdentified: Mode:off, MinCurrent:0.2, MaxCurrent:0.2, MinSoC:0, TargetSoC:100

ev5
---
.SoC:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'audi': invalid character 'u' looking for beginning of object key string
Capacity: 0.0kWh

ev6
---
.SoC:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'enyaq': invalid character 'u' looking for beginning of object key string
Capacity: 0.0kWh

ev7
---
.SoC:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'skoda': invalid character 'u' looking for beginning of object key string
Capacity: 0.0kWh

```

StevieC121176, 2022-11-24:
Ich habe parallel auf dem Raspi noch iobroker laufen und dort den VW Connect Adapter installiert. Bei diesem werden alle Werte von VW, Audi und Skoda noch angezeigt. Sollte VW etwas geändert haben sollte, würde es bei dem Adapter sicher auch zu Problemen kommen.

ABBolle, 2022-11-24:
> Ich habe parallel auf dem Raspi noch iobroker laufen und dort den VW Connect Adapter installiert. Bei diesem werden alle Werte von VW, Audi und Skoda noch angezeigt. Sollte VW etwas geändert haben, würde es bei dem Adapter sicher auch zu Problemen kommen.

Oder ioBroker kommt mit dem übertragenen Wert (der evtl irgendein Sonderzeichen enthält) besser zurecht. Die App läuft ja auch, die greift ja auch nur auf die API zu… aber ich habe Abend Downgrades bis 0.105.x gemacht und hatte auch mit den Versionen Probleme, die früher problemlos liefen. Für mich bleibt hier letztlich auch nur die Schnittstelle EVCC <—> VW/Audi/Skoda-Server…

andig, 2022-11-24:
Es ist ganz einfach: VW hat die SSt geändert...

roto11r, 2022-11-24:
same thing for me yesterday configuring my e-Up! .

rj3n53n, 2022-11-24:
Ich hab das Problem mit dem CUPRA Born auch.
Mir ist es gestern erstmalig aufgefallen.

sneakeroid, 2022-11-24:
Seit heute Nachmittag ca. 15 Uhr bei mir das gleiche Problem am ID.4.

RTTTC, 2022-11-24:
I think we have more than enough data points :DD 
All VAG cars are afected.

Takero1978, 2022-11-24:
Bei mir das gleiche Problem! Skoda Octavia IV Plug-In. Mir ist schon seit ein paar Tagen aufgefallen das die Vorklimatisierung nicht mehr funktionierte da die Wallbox nicht Freigegeben wurde. 

`creating vehicle ev1 failed: cannot create vehicle 'template': cannot create vehicle 'skoda': invalid character 'u' looking for beginning of object key string
.SoC:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'skoda': invalid character 'u' looking for beginning of object key string
`

aBua-MUC, 2022-11-24:
Ebenso bei mir mit Audi etron Q4. Wobei mir aufgefallen ist, dass die App heute den ganzen Tag Verundungsprobleme hatte und seit heute Abend wieder geht, allerdings evcc leider nicht.

panzerdev, 2022-11-24:
Habe den Fehler gefunden. Pull Request kommt gleich

andig, 2022-11-24:
Hatte ich auch gesehen, reicht das schon? Ich sehe „in echt“ auch überall neue Client IDs. Ist das damit wirklich schon behoben (ist zu spät heute nochmal rein zu schauen)?

panzerdev, 2022-11-24:
Ich teste das gleich nach dem bauen nochmal

panzerdev, 2022-11-24:
Mein VW ID3 geht mit der Änderung wieder. Zu anderen kann ich leider nix sagen.

AES-256-GCM, 2022-11-24:
Das kann ich bestätigen (ID.4):

master (https://github.com/evcc-io/evcc/commit/31965b0059e6829e89be456e6a2ebd0900bad09c):
```
tony@Tony-Lenovo ~/repos/evcc_test
 % sudo docker run -it -v "$(pwd)/evcc.yaml:/app/evcc.yaml" f55882d9aa7f evcc vehicle
[main  ] INFO 2022/11/24 23:06:27 evcc 0.107.1 (31965b00)
[main  ] INFO 2022/11/24 23:06:27 using config file: /app/evcc.yaml
SoC:            43%
Charge status:  B
Range:          184km
Odometer:       not available
Finish time:    2022-11-24 22:05:00 +0100 CET
Climate active: false
Outside temp:   20.0°C
Target temp:    20.0°C
Target SoC:     100%
Capacity:       77.0kWh
```
vs. Version 0.107.1:
```
tony@Tony-Lenovo ~/repos/evcc_test
 % sudo docker run -it -v "$(pwd)/evcc.yaml:/app/evcc.yaml" andig/evcc:0.107.1 evcc vehicle
Unable to find image 'andig/evcc:0.107.1' locally
0.107.1: Pulling from andig/evcc
9621f1afde84: Already exists 
8e14fcd9805e: Pull complete 
d01c494b0b01: Pull complete 
6f2aa47ac284: Pull complete 
a9fea25db8dc: Pull complete 
98fbfe8ac5d3: Pull complete 
Digest: sha256:ff27f4e7855fc5b6428ac1df89c1d0e007c471372eaa9d5883b879b0e8ac7a92
Status: Downloaded newer image for andig/evcc:0.107.1
[main  ] INFO 2022/11/24 23:07:08 evcc 0.107.1
[main  ] INFO 2022/11/24 23:07:08 using config file: /app/evcc.yaml
[main  ] ERROR 2022/11/24 23:07:09 creating vehicle id4 failed: cannot create vehicle 'template': cannot create vehicle 'id': invalid character 'u' looking for beginning of object key string
.SoC:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'id': invalid character 'u' looking for beginning of object key string
Capacity: 0.0kWh

```

StevieC121176, 2022-11-25:
mit der neuesten Version werden die Fahrzeuge wieder erkannt.

```
</details>
Version: `0.107.1 (31965b00)`

pi@raspberrypi:~ $ evcc vehicle
[main  ] INFO 2022/11/25 07:04:36 evcc 0.107.1 (31965b00)
[main  ] INFO 2022/11/25 07:04:36 using config file: /etc/evcc.yaml
[mqtt  ] INFO 2022/11/25 07:04:36 connecting evcc-229751274 at tcp://192.168.178.77:1883
ev5
---
SoC:            69%
Charge status:  A
Range:          29km
Odometer:       21656km
Finish time:    not available
Climate active: false
Outside temp:   0.0°C
Target temp:    20.0°C
Position:       51.798713,12.247609
Capacity:       15.0kWh
OnIdentified:   Mode:off, MinCurrent:6, MaxCurrent:16, MinSoC:40, TargetSoC:99

ev6
---
SoC:           48%
Charge status: A
Range:         185km
Odometer:      12370km
Finish time:   not available
Target SoC:    80%
Capacity:      82.0kWh
Identifiers:   [*****]
OnIdentified:  TargetSoC:99, Mode:off, MinCurrent:6, MaxCurrent:16, MinSoC:30

ev8
---
SoC:          0%
Capacity:     0.8kWh
OnIdentified: Mode:off, MinCurrent:0.2, MaxCurrent:0.2, MinSoC:0, TargetSoC:100

ev7
---
SoC:            35%
Charge status:  A
Range:          67km
Odometer:       26468km
Finish time:    2022-11-25 09:09:00 +0100 CET
Climate active: false
Outside temp:   0.0°C
Target temp:    22.0°C
Position:       51.903877,12.211787
Capacity:       38.0kWh
Identifiers:    [*****]
OnIdentified:   Mode:off, MinCurrent:6, MaxCurrent:16, MinSoC:40, TargetSoC:99```

andig, 2022-11-25:
Danke @panzerdev da hab ichs wohl zu kompliziert gemacht. Oder: jetzt ist noch eine Kompatibilitätsperiode und das dicke Ende kommt demnächst... wir werden sehen.

closed 2022-11-25 by andig

panzerdev, 2022-11-25:
Gerne, mal schauen ob sie noch was ändern in Zukunft. 
Ich nutze das Projekt hier https://github.com/tillsteinbach/WeConnect-mqtt
Fürs aufzeichnen der Daten extra und da hat noch alles funktioniert. Daher wusste ich es kann nicht alles kaputt sein.

RTTTC, 2022-11-25:
With the fix implemented now I get this error:

`ERROR 2022/11/25 10:14:01 creating vehicle ev1 failed: cannot create vehicle 'enyaq': cannot get vehicles: Get "https://api.connect.skoda-auto.cz/api/v3/garage": context deadline exceeded (Client.Timeout exceeded while awaiting headers)`

`evcc vehicle ` - sometimes works, sometimes gives same error

panzerdev, 2022-11-25:
> With the fix implemented now I get this error:
>
> `ERROR 2022/11/25 10:14:01 creating vehicle ev1 failed: cannot create vehicle 'enyaq': cannot get vehicles: Get [...]

The fix has no influence on the request process to the API so the error you are seeing us either you internet connection or network at home or the Skoda server having issues. 100% not related to this issue and fix here.

schenlap, 2022-11-25:
Login on skoda website or app is currently not possible. Probably not an evcc issue.

RTTTC, 2022-11-25:
Hmm, for me Skoda app works, but unable to login to skoda web site.
I guess now it’s a waiting game :)
How are you guys with different VAG brands doing today?

ABBolle, 2022-11-25:
My VW WeConnect ID App is working fine, but I cannot reach the login to myVolkswagen website at the moment.

andig, 2022-11-25:
Da hat sich gar nichts geändert und ist hier auch off-topic ;)

andig, 2022-12-02:
Die Frage verstehe ich nicht- warum nimmst Du nicht die neue Version?

andig, 2022-12-04:
Da bin ich sprachlos. Ich mach hier zu da Thema erledigt.