evcc-io/evcc #5268: Add support for Polestar

hyperbart, 2022-11-25:
**Is your feature request related to a problem? Please describe.**
Polestar car's are not yet configurable in EVCC

**Describe the solution you'd like**
Add the Polestar brand as a configurable vehicle.

**Describe alternatives you've considered**
Use Polestar as guest vehicle

**Additional context**
Afaicau you can contact Polestar or use the Android Automotive to query necessary parameters.
https://www.polestar.com/de/developer/get-started/

Be advised. Android Automotive is not the same as Android Auto.

andig, 2022-11-25:
Polestar has no API. Can't fix.

closed 2022-11-25 by andig

hyperbart, 2022-11-25:
And Android Automotive is also not a good way to extract info for other vehicles?

andig, 2022-11-25:
Not if you don't want to develop an app running in the car. And even that would not have access to the soc. Blame Polestar or better don't buy one...

hyperbart, 2022-11-25:
> 

I'll keep that in mind when I have to pick a company car here :) . Luckily it's now only a visitors car 😃

andig, 2023-12-17:
Finally- Polestar is coming in https://github.com/evcc-io/evcc/pull/11125

hyperbart, 2023-12-18:
@andig What a beast you are. Christmas is early this year. 🎄 
End of Q1 2024 loadbalancing 🚀 ? 😏🤟