evcc-io/evcc #5271: Ich habe heute Morgen zum ersten Mal Zielladen eingestellt.

andig, 2022-11-25:
Ich habe heute Morgen zum ersten Mal Zielladen eingestellt. 
Dann kam der Hinweis auf eine neue Version (0.108.0). 
Nach dem update, waren die Einstellungen beim Zielladen weg.
Die werden noch nicht gespeichert?

_Originally posted by @VolkerK62 in https://github.com/evcc-io/evcc/discussions/1433#discussioncomment-4237441_

pauxus, 2022-11-26:
Geht nicht direkt ums Zielladen, aber ich denke, man könnte das hier noch etwas weiter fassen und auch andere Settings kurzzeitig speichern, in meinem Fall konkret: Welche Autos sind angesteckt. Ich habe hier zwei Ladepunkte und es stehen eben auch zwei Fahrzeuge da. Normalerweise wird das auch sauber erkannt, nur wenn bei einem Neustart beide angeschlossen sind, scheitert die Erkennung. Ich muss also dran denken, die Autos wieder zu konfigurieren.

Schön wäre es, wenn auch diese Information Kurzzeit persistiert wird - vielleicht einfach mit einem Zeitstempel. Hier könnte ja sogar einen einfach Json Datei reichen. Wenn beim Neustart der Zeitstempel der Settings nicht älter als 30 min oder so ist, wird sie angezogen und gelöscht, ansonsten eine Warnung ausgegeben.

PeterPablo, 2023-01-30:
Noch ein use case zu bedenken: Manchmal möchte ich bewusst nur einmalig (!) 5kWh nachladen, um eine gewisse kurze Wegstrecke sicher zurücklegen zu können. Ein Schalter zum "toggeln", ob die eingestellte Zielladung "einmalig" oder "dauerhaft"  zu sehen ist.
Mir ist es schon passiert, dass ich vergaß die Begrenzung wieder aufzuheben und dann war am darauffolgenden Morgen der Akku leider nicht voll.

andig, 2023-02-15:
> Mir ist es schon passiert, dass ich vergaß die Begrenzung wieder aufzuheben und dann war am darauffolgenden Morgen der Akku leider nicht voll.

Du könntest die Beschränkung in die Push Notifications mit aufnehmen. Noch mehr Settings im UI würde ich gerne vermeiden. Zielladung ist immer einmalig zum Termin, wäre also eine weitere Lösungsmöglichkeit.

Ist hier aber OT ;)

steve0564, 2023-03-01:
> Ich habe heute Morgen zum ersten Mal Zielladen eingestellt. Dann kam der Hinweis auf eine neue Version (0.108.0). Nach dem update, waren die Einstellungen beim Zielladen weg. Die werden noch nicht gespeichert?
> 
> _Originally posted by @VolkerK62 in [#1433 (comment)](https://github.com/evcc-io/evcc/discussions/1433#discussioncomment-4237441)_

Nicht nur das!
Das Fatale, wie ich finde, ist die Tatsache, dass das Zielladen nach einem Update noch unverändert erhalten bleibt. Man denkt nun: Super, alles in Ordnung!
[image]

Aktualisiert man aber die UI-Seite, dann sieht man erst, dass das mit dem Zielladen leider doch nicht klappen wird, da erst jetzt erkennbar ist, dass das Zielladen wieder zurückgesetzt wurde.
[image]

andig, 2023-03-01:
Guter Hinweis. Die (nicht vorhandene) Zeit zu löschen können wir schneller nachreichen als die Speicherung.

/cc @hoermto hier wäre direkt ein schöner Anwendungsfall für "settings persistieren".

torstenwiese, 2023-04-01:
Die Überlegung, bzw Bedarf hatte ich auch schon, aber anderes Scenario:
mein Kona verfügt leider nicht über Bluelink, demnach erhält EVCC nicht automatisch den SoC der Batterie.
Nun komme ich von der Arbeit heim und stelle ein, dass z.B. 40% für den nächsten morgen geladen werden sollen. Meine Frau nimmt das Auto, ich bekomme das nicht mit, und morgens ist kein Strom drin. 

Mein Gedanke war, die reset Funktion, welche in der Yaml hinterlegt ist, per API bis zu einem anzugebenden Zeitpunkt abzuschalten. 
Button, „no reset“ und Abfrage für wieviele Stunden oder bis Uhrzeit hh:mm.

schenlap, 2023-04-21:
Ich habe folgenden Workaround am Laufen, vielleicht hilft es euch bis zur Lösung.
Über evcc push-message wird ein script aufgerufen welches die Zielzeit (6:00 UTC am nächsten Tag) beim Anstecken setzt. 

```
#!/bin/bash

# configure push messages in evcc.yaml
#messaging:
#  events:
#    connect: # vehicle connect event
#      title: Car connected
#      msg: "Car connected, mode:${mode}"
#    guest:
#      title: Guest connected
#      msg: "Guest connected, mode:${mode}"
#  services:
#    - type: script
#      cmdline: /home/steve/prj/evcc/evcc_message
#      timeout: 50s

ip=localhost:7070
now=$(date)

echo "message received: $1 - $now"
if [[ "$2" == *"connected, mode:pv"* ]] # guest and car
then
        targettime=$(date -d "+1 day" -u +"%Y-%m-%d")T06:00:00Z

        curl -X POST http://$ip/api/loadpoints/1/target/time/$targettime > /dev/null
        echo "Setting targettime: " $targettime
else
        echo "Wrong message: " $2
fi

```

torstenwiese, 2023-04-21:
> Ich habe folgenden Workaround am Laufen, vielleicht hilft es euch bis zur Lösung. Über evcc push-message wird ein script aufgerufen welches die Zielzeit (6:00 UTC am nächsten Tag) beim Anstecken [...]

Vielen Dank für den Tipp! Hatte ich noch nicht auf dem Radar ein Script auch beim Anstecken zu etablieren.

Zielladen an sich ist aber nicht ganz fehlerfrei, von der Logik her nicht ganz fertig, von EVCC aus. Ich hatte lange, seit einiger Zeit sehr selten, das Problem, dass trotz Begrenzung auf 20A mit 32A geladen wird. Wenn dies geschieht, ist das EV zwar zur Zielzeit geladen, aber zu früh für die Vorklimatisierung. 
Es fehlt eventuell ein Nachsteuern des Ladestroms, wenn IST und SOLL über Zeitraum X abweichend sind? 🤔

andig, 2023-04-21:
> Zielladen an sich ist aber nicht ganz fehlerfrei, von der Logik her nicht ganz fertig, von EVCC aus. Ich hatte lange, seit einiger Zeit sehr selten, das Problem, dass trotz Begrenzung auf 20A mit 32A geladen wird. Es fehlt eventuell...

V.a. fehlt ein neues Issue in dem man das nachvollziehen kann. Bitte mit Logfile, hier ist das off-topic.

diver77, 2023-04-28:
#!/bin/bash

# configure push messages in evcc.yaml
#messaging:
#  events:
#    connect: # vehicle connect event
#      title: Car connected
#      msg: "Car connected, mode:${mode}"
#    guest:
#      title: Guest connected
#      msg: "Guest connected, mode:${mode}"
#  services:
#    - type: script
#      cmdline: /home/steve/prj/evcc/evcc_message
#      timeout: 50s

ip=localhost:7070
now=$(date)

echo "message received: $1 - $now"
if [[ "$2" == *"connected, mode:pv"* ]] # guest and car
then
        targettime=$(date -d "+1 day" -u +"%Y-%m-%d")T06:00:00Z

        curl -X POST http://$ip/api/loadpoints/1/target/time/$targettime > /dev/null
        echo "Setting targettime: " $targettime
else
        echo "Wrong message: " $2
fi

sorry kenne mich überhaupt nicht aus mit scripten.
Wo führst du das script denn genau aus? und wohin muss ich die Messages genau schicken.

Danke für Eure Hilfe

Gruß Diver

schenlap, 2023-04-28:
das Scipt legst du irgendwo ab wo evcc läuft und macht es ausführbar `chmod +x scriptname`
Oben das auskommentierte gehört in die evcc.yaml (natürlich nicht auskommentiert). Den Namen unter cmdline musst du anpassen an das verzeichnis/sciptname. Danach evcc neu starten.

diver77, 2023-05-01:
Hallo Schenlap,
danke für dein Antwort. so jetzt fängt es schon an das ich es nicht mehr kapiere. :-)
ich habe homeassistant und evcc auf einem Raspberry installiert. evcc läuft dort als addon.
wenn ich mich jetzt auf den Rapberry über meine ip verbinde sehen ich zwar alles was ich auf homeassistant installiert habe aber leider nichts von evcc.
muss ich das irgendwie über das terminal in homeassistant machen und wenn ja, wie?
und was meinst du ich muss es ausführbar machen?

den rest habe ich glaube ich verstanden. :-) ist dann ja aber auch nicht mehr so viel. 

Sorry aber bin doch ein ziemlicher Anfänger in solchen Sachen. Bekomme es zwar irgendwie meistens hin, aber nur dank Leuten wie Dir. :-)

Leider regnerische Grüße und einen schönen Restfeiertag noch.

Gruß Dirver

schenlap, 2023-05-01:
Sorry, mit Homeassistant kenne ich mich 0 aus, da kann ich leider nicht unterstützen.

diver77, 2023-05-02:
Danke für deine Antwort.

Kann mir vielleicht sonst jemand helfen, wie ich das Skript in HA zum laufen bekommen?

Danke

Gruß Diver

maximinus333, 2023-06-19:
Ich habe versucht das nachzubauen. Als Linux Anfänger kann ich mit dieser Fehlermeldung nichts anfangen:
`Fehler beim Starten
While parsing config
yaml: line 82: could not find expected ':'

Bitte überprüfe deine Konfigurationsdatei.`

Wie korrigiere ich das?

Gerhard

PR #8115 (2023-08-02): Persist targetSoc/Energy, minSoc, targetTime per vehicle, add minSoc to UI
Vehicle based settings now survive server restart and are carried over when manually changing vehicles at a loadpoint.

fixes #5271
replaces #6740

- 💽 storing `targetSoc`, `targetEnergy`, `minSoc`, `targetTime` per vehicle in database
- 🔄 stored settings are used on startup, vehicle change (UI) and physical vehicle (dis-)connect
- 🧪 removed "experimental" flag for `minSoc` UI (see screenshot)
  - 🛫 target time charging is now in the departure tab (default)
  - 🛬 additional onIdentify options (min/maxCurrent, mode) will [... cut]
closed 2023-08-02 by naltatis