evcc-io/evcc #5274: charger logic error: disabled but charging

mdkeil, 2022-11-26:
### Describe the bug

Habe dieses Verhalten mindestens schon zwei mal gehabt, dass ich das Auto anschließe und direkt mit maximaler Geschwindigkeit geladen wird, obwohl evcc defaultmäßig auf PV läuft. Es scheint wohl, dass evcc die Wallbox nicht mehr ansteuern kann, da sich das Laden nur über das Webinterface der Wallbox stoppen lässt. 

### Steps to reproduce

Auto an die Wallbox anschließen.. weiter kann ich es nicht eingrenzen, wird aber wohl abhängig von der Wallbox sein. Ich nutze eine hardy barth cpμ2.

### Configuration details

```yaml
network:
  schema: http
  host: evcc
  port: 7070
interval: 10s

influx:
  url: http://192.168.1.2:8086
  database: evcc
  user: evcc
  password: ####€

log: info

vehicles:
  - name: my_peugeot
    type: template
    template: peugeot
    title: e-208
    user: keil_stefan@gmx.de
    password: #######
    vin: ######
    capacity: 46
    #  minSoC: 20 # immediately charge to 0% regardless of mode unless "off" (disabled)
    targetSoC: 80 # limit charge to 80%

chargers:
  - name: my_charger
    type: template
    template: hardybarth-salia
    host: 192.168.1.11

meters:
  - name: my_grid
    type: template
    template: fronius-solarapi-v1
    usage: grid
    host: 192.168.1.10
  - name: my_pv
    type: template
    template: fronius-solarapi-v1
    usage: pv
    host: 192.168.1.10

site:
  title: Home
  meters:
    grid: my_grid
    pvs:
      - my_pv

loadpoints:
  - title: Carport
    charger: my_charger
    mode: pv
    phases: 3
    mincurrent: 6
    maxcurrent: 16
    resetOnDisconnect: true
    enable:
      threshold: -2500
      delay: 1m
    disable:
      threshold: 1000
      delay: 5m

tariffs:
  grid:
    type: fixed
    price: 0.3355
  feedin:
    type: fixed
    price: 0.0736
```

### Log details

```text
[main  ] INFO 2022/11/24 10:44:52 evcc 0.107.1
[main  ] INFO 2022/11/24 10:44:52 using config file: /etc/evcc.yaml
[main  ] INFO 2022/11/24 10:44:52 starting ui and api at :7070
[site  ] INFO 2022/11/24 10:44:55 site config:
[site  ] INFO 2022/11/24 10:44:55   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2022/11/24 10:44:55     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/11/24 10:44:55     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/11/24 10:44:55   vehicles:
[site  ] INFO 2022/11/24 10:44:55     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗
[lp-1  ] INFO 2022/11/24 10:44:55 loadpoint 1:
[lp-1  ] INFO 2022/11/24 10:44:55   mode:        pv
[lp-1  ] INFO 2022/11/24 10:44:55   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2022/11/24 10:44:55   meters:      charge ✓
[lp-1  ] INFO 2022/11/24 10:44:55     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2022/11/24 10:44:55 car connected
[lp-1  ] INFO 2022/11/24 10:44:55 vehicle updated: unknown -> e-208
[lp-1  ] INFO 2022/11/24 10:54:05 start charging ->
[lp-1  ] INFO 2022/11/24 11:01:55 stop charging <-
[lp-1  ] INFO 2022/11/24 11:02:35 start charging ->
[salia ] ERROR 2022/11/24 11:23:30 heartbeat: Put "http://192.168.1.11/api/secc": dial tcp 192.168.1.11:80: i/o timeout
[lp-1  ] INFO 2022/11/24 14:10:55 stop charging <-
[lp-1  ] INFO 2022/11/24 14:30:15 car disconnected
[lp-1  ] INFO 2022/11/24 14:30:15 vehicle updated: e-208 -> unknown
[salia ] ERROR 2022/11/24 20:35:07 heartbeat: Put "http://192.168.1.11/api/secc": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:15 charge meter: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:15 charge meter: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:15 meter energy: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:15 charger: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused
[salia ] ERROR 2022/11/24 20:35:38 heartbeat: Put "http://192.168.1.11/api/secc": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:38 charge meter: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:38 charge meter: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:38 meter energy: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:38 charger: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:39 charge meter: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:39 charge meter: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:39 meter energy: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:39 charger: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:45 charge meter: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:45 charge meter: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:45 meter energy: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:45 charger: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused
[main  ] INFO 2022/11/25 10:44:55 new version available: 0.108.0
[salia ] ERROR 2022/11/25 15:07:21 heartbeat: Put "http://192.168.1.11/api/secc": EOF
[main  ] INFO 2022/11/25 16:44:55 new version available: 0.108.0
[main  ] INFO 2022/11/25 22:44:55 new version available: 0.108.0
[main  ] INFO 2022/11/26 04:44:55 new version available: 0.108.0
[lp-1  ] INFO 2022/11/26 08:28:45 car connected
[lp-1  ] INFO 2022/11/26 08:28:45 start charging ->
[lp-1  ] INFO 2022/11/26 08:28:46 vehicle updated: unknown -> e-208
[lp-1  ] WARN 2022/11/26 08:28:46 charger logic error: disabled but charging
[lp-1  ] WARN 2022/11/26 08:28:46 charger logic error: disabled but charging
[lp-1  ] WARN 2022/11/26 08:28:55 charger logic error: disabled but charging
[lp-1  ] WARN 2022/11/26 08:29:05 charger logic error: disabled but charging
[lp-1  ] WARN 2022/11/26 08:29:15 charger logic error: disabled but charging
[lp-1  ] INFO 2022/11/26 08:29:25 stop charging <-
[lp-1  ] INFO 2022/11/26 08:29:35 start charging ->
[lp-1  ] WARN 2022/11/26 08:29:35 charger logic error: disabled but charging
[lp-1  ] WARN 2022/11/26 08:29:45 charger logic error: disabled but charging
[lp-1  ] WARN 2022/11/26 08:29:55 charger logic error: disabled but charging
[lp-1  ] WARN 2022/11/26 08:30:05 charger logic error: disabled but charging
[lp-1  ] WARN 2022/11/26 08:30:15 charger logic error: disabled but charging
[lp-1  ] WARN 2022/11/26 08:30:25 charger logic error: disabled but charging
[lp-1  ] WARN 2022/11/26 08:30:35 charger logic error: disabled but charging
[lp-1  ] WARN 2022/11/26 08:30:45 charger logic error: disabled but charging
[lp-1  ] WARN 2022/11/26 08:30:55 charger logic error: disabled but charging
[lp-1  ] WARN 2022/11/26 08:31:05 charger logic error: disabled but charging
[lp-1  ] WARN 2022/11/26 08:31:15 charger logic error: disabled but charging
[lp-1  ] WARN 2022/11/26 08:31:25 charger logic error: disabled but charging
[lp-1  ] WARN 2022/11/26 08:31:35 charger logic error: disabled but charging
[lp-1  ] WARN 2022/11/26 08:31:45 charger logic error: disabled but charging
[lp-1  ] WARN 2022/11/26 08:31:55 charger logic error: disabled but charging
[lp-1  ] WARN 2022/11/26 08:32:05 charger logic error: disabled but charging
[lp-1  ] WARN 2022/11/26 08:32:15 charger logic error: disabled but charging
[lp-1  ] WARN 2022/11/26 08:32:23 charger logic error: disabled but charging
[lp-1  ] WARN 2022/11/26 08:32:25 charger logic error: disabled but charging
[lp-1  ] WARN 2022/11/26 08:32:35 charger logic error: disabled but charging
[lp-1  ] WARN 2022/11/26 08:32:45 charger logic error: disabled but charging
[lp-1  ] WARN 2022/11/26 08:32:55 charger logic error: disabled but charging
[lp-1  ] WARN 2022/11/26 08:33:05 charger logic error: disabled but charging
[lp-1  ] WARN 2022/11/26 08:33:15 charger logic error: disabled but charging
[lp-1  ] WARN 2022/11/26 08:33:25 charger logic error: disabled but charging
[lp-1  ] INFO 2022/11/26 08:33:35 stop charging <-
```

### What type of operating system are you running?

Docker container

### Version

0.107.1

andig, 2022-11-26:
Verbindung zur Wallbox fehlr -> Netzwerkproblem.

mdkeil, 2022-11-26:
Das ist deutlich zu kurz gedacht, da zu 95% alles wunderbar funktioniert (Sowohl PV als auch Min+PV) wie man im Log sieht, wird das Fahrzeug auch im "Fehlerfall" wunderbar erkannt.. es wird auch korrekt erkannt, dass geladen wird.. angezeigte Ladeleistung passt ebenfalls.. Außerdem dürfte das Laden trotzdem nicht automatisch starten, wenn es ein Netzwerkproblem geben sollte, da ja die Wallbox auf "manuell" stehen müsste.

andig, 2022-11-26:
OK. Dann mach mal bitte ein trace log so dass man sieht was da mit der Wallbox gesprochen wird. Ohne ist da nix zu holen.

mdkeil, 2022-11-26:
Kann ich gerne machen. Ich gehe davon aus, dass ich ein trace aus dem evcc-docker-container zur Wallbox liefern soll. welche (bordeigenen) Tools setz man da am besten ein. Habe damit noch nicht so viel praktische Erfahrung.

Wann es klappt ist einfach.. wann es nicht klappt eher ein Glücksspiel.. ich würde den Trace dann dauerhaft aufzeichnen und dann im Fehlerfall hier abliefern.

VolkerK62, 2022-11-26:
https://docs.evcc.io/docs/reference/configuration/log

```
log: info
levels:
  site: debug
  salia: trace
  lp-1: trace
```

Micha281, 2022-11-26:
Hallo, das gleiche Problem habe ich auch mit meiner daheimladen Wallbox. Sehr oft kommt die Meldung dass die Box weiter lädt, obwohl sie abgeschaltet sein soll. Aber auch umgekehrt. Sie ladt noch nicht, obwohl EVCC sie gestartet hat. 
Nach einer kurzen Zeit hat sich das Problem aber wieder erledigt, da dann die Wallbox den Befehlen mit Verzögerung gefolgt ist.
Deshalb sollte ein Toleranzzeit eingebaut werden, die der Wallbox Zeit gibt zu reagieren.

mdkeil, 2022-11-26:
@VolkerK62 Danke!

@Micha281 Bei mir tritt es halt nur auf (zum Glück eher selten) direkt nachdem ich das Auto an die WB stecke.. Da brauch es auch keine "Toleranzzeit", denn egal warum, die WB garnicht starten darf, da sie auf manuell steht und evcc entweder auf PV oder Aus.. wie gesagt, die WB lässt sich bei mir dann über evcc erst wieder bedienen, wenn ich den Ladevorgang über das Webinterface der WB gestoppt habe.. 

Ich habe bei meiner Wallbox mal eben "Enable Heartbeat" rausgenommen, da ich im log den ein oder anderen Fehler gesehen habe..

[salia ] ERROR 2022/11/25 15:07:21 heartbeat: Put "http://192.168.1.11/api/secc": EOF

VolkerK62, 2022-11-26:
@Micha281  Lösung ist hier beschrieben https://docs.evcc.io/docs/guides/charging#fehler-charger-out-of-sync-expected-disabled-got-enabled--charger-logic-error-disabled-but-charging

premultiply, 2022-11-26:
Speziell nach dem Neuanstecken kann es aber vielleicht auch an einer Einstellung der Box liegen dass diese einfach loslädt ohne zu beachten dass sie zuvor im gesperrten Zustand war.
Könnte auch ein Firmware-Bug sein oder auch ein Bug in der evcc-Implementierung zu dieser Wallbox.
Da hilft tatsächlich nur eine genaue Analyse der Kommunikation (Tracelog wie schon beschrieben) um eingrenzen zu können wo ein genauerer Blick lohnen könnte.

mdkeil, 2022-11-27:
Habe das logging schon umgestellt. Heute muss das Auto auf jeden Fall nochmal an die Wallbox. Wann ich den Trace liefern kann, hängt halt davon ab, wann der Bug auftritt.

mdkeil, 2022-11-27:
ca. 09:43 habe ich das Fahrzeug angeschlossen und der Ladevorgang hat direkt begonnen.. habe kurze Zeit später über das Webinterface der Wallbox die Ladung gestoppt.. so gegen 09:54 habe ich auf Min+PV gestellt und die Ladung wurde zeitnah wieder aufgenommen.

[evcc.log@google.drive](https://drive.google.com/file/d/1wix19zF6-Q5NMPN5yy2R_bGIezl51Pyv/view?usp=share_link)

andig, 2022-11-27:
Hast du nochmal einen funktionierenden Ansteckvorgang? Das wäre ja der Vergleich.

mdkeil, 2022-11-27:
anbei ein log, wo es funktioniert hat // meine Vermutung, da es wohl immer dann passiert, wenn lange keine Auto angeschlossen war.. irgend ein "deep-sleep"-Modus der Wallbox?

- Ladung durch "Aus" unterbrochen
- Auto abgesteckt
- EVCC manuell auf PV gestellt
- Auto angeschlossen

[log@google.drive](https://drive.google.com/file/d/17tFQRZiwB0JOE4IoWP-ZkxkBh6znLr6V/view?usp=share_link)

mdkeil, 2022-11-28:
Was zumindest in den Logs ersichtlich ist, dass die Wallbox wohl dauerhaft auf "manuell" steht in beiden Fällen..

premultiply, 2022-11-28:
Wir brauchen ein Log vom Fehlerfall.
Und zwar dort auch vor dem Anstecken.

mdkeil, 2022-11-28:
hier mal ein kompletter Trace, der alles enthalten sollte.. es müsste ein connection error zur WB enthalten sein.. da habe ich die Wallbox rebootet.

[evcc_trace.log](https://drive.google.com/file/d/1B1GKUVg4R1b4AxJe1KyyRLFSL9rnoau5/view?usp=share_link)

andig, 2022-11-28:
Hättest Du noch die Timestamps auf die wir schauen müssen?

mdkeil, 2022-11-28:
[lp-1  ] INFO 2022/11/26 13:30:14 car disconnected // zwischen diesem Timestamp und dem folgenden muss es ein Ereignis geben, was das fehlerhafte Ereignis auslöst.. Sollte dies nicht der Fall sein, werde ich diesbezüglich mal an den Herstellersupport wenden.

[lp-1  ] INFO 2022/11/27 09:42:15 car connected // anschließend wurde der Ladevorgang mit max begonnen

[lp-1  ] INFO 2022/11/27 10:49:35 car connected // evcc bzw. walbox reagiert wie erwartet

mdkeil, 2022-12-05:
Da es noch keine weitere Antwort gibt, habe ich mir nochmal alle Logs angeschaut und zumindest eine Übereinstimmung finden können-- Nachdem das Auto von der WB getrennt wird und es dann beim folgenden Anstecken zum direkten Laden kommt, hab ich in allen Logs folgende Fehlermeldung:

`[salia ] ERROR 2022/11/24 20:35:07 heartbeat: Put "http://192.168.1.11/api/secc": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:15 charge meter: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:15 charge meter: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:15 meter energy: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:15 charger: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused
[salia ] ERROR 2022/11/24 20:35:38 heartbeat: Put "http://192.168.1.11/api/secc": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:38 charge meter: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:38 charge meter: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:38 meter energy: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:38 charger: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:39 charge meter: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:39 charge meter: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:39 meter energy: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:39 charger: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:45 charge meter: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:45 charge meter: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:45 meter energy: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused
[lp-1  ] ERROR 2022/11/24 20:35:45 charger: Get "http://192.168.1.11/api": dial tcp 192.168.1.11:80: connect: connection refused`  

Was mich wundert, dass dann der Folge-Trace wieder unauffällig ist. Die Frage ist nun, was dieses Problem verursacht? Meiner Meinung nach ist es kein Netzwerkproblem. Ich log dennoch mal einen Dauerping auf die WB.

andig, 2022-12-05:
@mdkeil passiert das immer *genau* bevor das Auto unberechtigt zu laden anfängt? Falls ja würde ich das Issue schließen- denn dann muss erstmal der Grundsatzfehler weg. Wer weiß, ob nicht die Wallbox in dem Zeitraum einen Neustart oder sonstwas macht :O

mdkeil, 2022-12-05:
Nein, das passiert zwischen dem letzen Abstecken und dem nächsten Anstecken und lässt sich zeitlich nicht einordnen. Ich vermute aber auch, dass die WB sich hin und wieder mal aufhängt bzw. einen softreboot macht. (daher auch mal Testweise der Dauerping)  Komischerweise läuft alles normal weiter nachdem ich den "fehlerhaften" Ladevorgang im UI der WB stope.

andig, 2022-12-05:
Wir machen für den Charger 2 Schritte bei der Initialisierung:

	// set chargemode manual
	res, err := wb.get()
	if err == nil && res.Secc.Port0.Salia.ChargeMode != echarge.ModeManual {
		if err = wb.post(salia.ChargeMode, echarge.ModeManual); err == nil {
			res, err = wb.get()
		}

		if err == nil && res.Secc.Port0.Salia.ChargeMode != echarge.ModeManual {
			err = errors.New("could not change chargemode to manual")
		}
	}

	if err == nil {
		go wb.heartbeat()

		wb.pause(false)

		res, err := wb.get()
		if err == nil && res.Secc.Port0.Metering.Meter.Available > 0 {
			return decorateSalia(wb, wb.currentPower, wb.totalEnergy, wb.currents), nil
		}
	}

Wenn die Box natürlich wirklich neu bootet (wenns keine Netzwerkprobleme sind) dann ist die Initialisierung weg. In dem Fall können wir auch nichts tun und wir müssen leider an den Hersteller verweisen.

Aus meiner Sicht daher wontfix und Ticket zu schließen. Meld Dich gerne nochmal wenns eine Antwort von Hardybarth gibt oder ob der Dauerping was erreicht (ich denke eher nicht- der evcc `heartbeat()` is ja auch nix Anderes).

closed 2022-12-05 by andig

mdkeil, 2022-12-06:
was mir auch aufgefallen ist, unabhängig vom "connection error", das ca. einmal pro Tag noch folgender Fehler im Log steht:

`[salia ] ERROR 2022/12/04 18:11:39 heartbeat: Put "http://192.168.1.11/api/secc": EOF`

mdkeil, 2022-12-08:
Seit der Dauerping läuft, gibt es keine Probleme mehr.. komisch