evcc-io/evcc #5280: Vehicle: startup error confuses UI

andig, 2022-11-26:
### Describe the bug

When vehicle is not initialised as expected, it should be displayed similar to the Gastfahrzeug. That's currently not the case.

### Steps to reproduce

Gastfahrzeug:

[image]

Vehicle with startup error:

[image]

### Configuration details

```yaml
n/a
```

### Log details

```text
n/a
```

### What type of operating system are you running?

Docker container

mmrazovic, 2022-12-10:
Yes, a restart solved it on my system.

andig, 2022-12-29:
@naltatis hier gibts 2 Probleme:
- Ladeanimation wird nicht analog Gastfahrzeug angezeigt
- Soc Slider wird angezeigt

Ich schaue mal, ob ich das im Backend lösen kann indem wir dann einfach ein offline-Fahrzeug erzeugen

closed 2022-12-29 by andig