evcc-io/evcc #5289: Soc von Audi kann nicht abgerufen werden

schenlap, 2022-11-27:
### Describe the bug

Von Fahrzeug Audi wird der SoC in evcc nicht angezeigt.
evcc vehicle funktioniert aber

### Steps to reproduce

1. Audi konfugurieen
2. Fahrzeug einstecken
3.
...

### Configuration details

```yaml
vehicles:
- name: audi
  type: audi
  title: A3 e-tron
  capacity: 7.5 # kWh
  user: xxx
  password: xxx
  vin: xxx
  cache: 5m
  onIdentify:
    minSoC: 10 # charge to at least 10% independent of charge mode
    targetSoC: 90 # limit charge to 90%
    mode: pv
```

### Log details

```text
Nov 27 14:22:40 raspberrypi evcc[18908]: [lp-1  ] INFO 2022/11/27 14:22:40 car connected
Nov 27 14:22:40 raspberrypi evcc[18908]: [lp-1  ] DEBUG 2022/11/27 14:22:40 pv timer elapse
Nov 27 14:22:40 raspberrypi evcc[18908]: [lp-1  ] DEBUG 2022/11/27 14:22:40 pv timer inactive
Nov 27 14:22:40 raspberrypi evcc[18908]: [lp-1  ] ERROR 2022/11/27 14:22:40 vehicle soc: vehicle not available: cannot create vehicle 'audi': Get "https://identity.vwgroup.io/oidc/v1/author
Nov 27 14:22:40 raspberrypi evcc[18908]: [lp-1  ] DEBUG 2022/11/27 14:22:40 pv charge current: 1.73A = 0A + 1.73A (-398W @ 1p)
Nov 27 14:22:40 raspberrypi evcc[18908]: [lp-1  ] DEBUG 2022/11/27 14:22:40 pv enable timer reset
Nov 27 14:22:40 raspberrypi evcc[18908]: [lp-1  ] DEBUG 2022/11/27 14:22:40 pv timer inactive
Nov 27 14:22:59 raspberrypi evcc[18908]: [site  ] DEBUG 2022/11/27 14:22:59 ----
Nov 27 14:22:59 raspberrypi evcc[18908]: [lp-1  ] DEBUG 2022/11/27 14:22:59 charge power: 0W
Nov 27 14:22:59 raspberrypi evcc[18908]: [site  ] DEBUG 2022/11/27 14:22:59 pv power: 477W
Nov 27 14:22:59 raspberrypi evcc[18908]: [site  ] DEBUG 2022/11/27 14:22:59 grid power: -325W
Nov 27 14:22:59 raspberrypi evcc[18908]: [site  ] DEBUG 2022/11/27 14:22:59 site power: -385W
Nov 27 14:22:59 raspberrypi evcc[18908]: [lp-1  ] DEBUG 2022/11/27 14:22:59 charge currents: [0 0 0]A
Nov 27 14:23:00 raspberrypi evcc[18908]: [lp-1  ] DEBUG 2022/11/27 14:23:00 charger status: B
Nov 27 14:23:00 raspberrypi evcc[18908]: [lp-1  ] DEBUG 2022/11/27 14:23:00 next soc poll remaining time: 2h59m40s
Nov 27 14:23:00 raspberrypi evcc[18908]: [lp-1  ] DEBUG 2022/11/27 14:23:00 pv charge current: 1.67A = 0A + 1.67A (-385W @ 1p)

$ evcc vehicle audi
[main  ] INFO 2022/11/27 14:27:07 evcc 0.108.2
[main  ] INFO 2022/11/27 14:27:07 using config file: /etc/evcc.yaml
SoC:            7%
Charge status:  B
Range:          1km
Odometer:       74048km
Finish time:    2022-11-27 14:32:00 +0100 CET
Climate active: false
Outside temp:   0.0°C
Target temp:    22.0°C
Position:       48.103774,13.158103
Capacity:       7.5kWh
OnIdentified:   MinSoC:10, TargetSoC:90, Mode:pv
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.108.2

andig, 2022-11-27:
Achherrje, geht das wieder los? Dann kommen jetzt wohl die Client IDs wie befürchtet. 

/cc @panzerdev

panzerdev, 2022-11-28:
@andig ich bin ein wenig verwirrt. Also zum einen funktioniert noch alles bei meinen ID3. Zum anderen sagt @schenlap cli command vehicle funktioniert aber evcc Server will nicht mit Error beim `https://identity.vwgroup.io/oidc/v1/author` GET call. Da weiß ich auch nicht genau was der Fehler wirklich ist.

Kannst du mir kurz sagen was du mit den Client IDs meinst?

rivengh, 2022-11-28:
Hatte gestern mit dem e-up das gleiche Problem. Heute aber bisher alles in Ordnung. Mal weiter beobachten...

andig, 2022-11-28:
...dann lasst uns wieder auf machen falls das reproduzierbar ein Dauerproblem wird. Temporäre Ausfälle sind immer möglich.

@schenlap im Eingangspost fehlt leider die vollständige Fehlermeldung- die brauchen wir bitte auch wenns wieder passiert.

closed 2022-11-28 by andig