evcc-io/evcc #5291: Integrate with ENTSO-E Day ahead pricing

bsmout, 2022-11-27:
Lots of dynamic tarifs are based on day ahead pricing for the specific region (country). It would be great if tariffs could be based on those day ahead prices. For example ENTSO-E is an Europe funded service that provides a free API. 

Clients for the API are available on Github.

More info: https://transparency.entsoe.eu/content/static_content/Static%20content/web%20api/Guide.html

andig, 2022-11-28:
Why would you want to do that?

andig, 2022-11-28:
Sure. But then you would integrate that tariff as we‘re already doing, not Entso.

andig, 2022-12-12:
When https://github.com/evcc-io/evcc/pull/5445 is finished, we can add additional plan data for target charging. Same approach as discussed in https://github.com/evcc-io/evcc/pull/5445#issuecomment-1346809709.

RTTTC, 2022-12-12:
OK, after more consideration looks like ENTSO-E Day ahead pricing would be the best one to use, and pricing matches my local market pricing data exactly, using mac/safari
(For some reason it showed wrong pricing when I used iphone to browse it)
So Nordpool data api is not needed.

RTTTC, 2022-12-13:
Could we reopen this?

andig, 2022-12-27:
/cc @thorstenwagner

RTTTC, 2022-12-27:
"Smart rate" is supposedly most popular utility option in Scandinavia, where customers have smart meters and electricity price is different every hour, as per Nordpool local area day ahead pricing (plus margin and any taxes)

bsmout, 2022-12-27:
> Now that we have a planner it would be nice to add minimal-CO2 planning. I'm not so keen on ENTSO prices as that's not what the user pays anyway, but would prefer CO2 load. Would appreciate suggestions for finding the relevant ENTSO reports and example data interpretation.

Based on what are you stating that ? I'm paying ENTSO pricing (with a very small margin and taxes).

andig, 2022-12-27:
At least not in Germany.

RTTTC, 2023-01-12:
> More info: https://transparency.entsoe.eu/content/static_content/Static%20content/web%20api/Guide.html

section 12.1.d for day-ahead pricing API

RTTTC, 2023-02-11:
This integration would be such a great evolution of evcc!

scholdan, 2023-03-20:
There is missing a lot of tariffs providers.
Maybe look at this https://github.com/MTrab/energidataservice and convert it? or just pull the data form homeassistant?

andig, 2023-09-03:
@duckfullstop if you want to take a stab? Entso prices would be nice but the API is a complex.

duckfullstop, 2023-09-03:
> 4.2.10. Day Ahead Prices [12.1.D] for day-ahead pricing API

This looks quite achievable (even if it does mean working with XML), I can take a bit more of a look into it in the coming week hopefully?

I'm not sure about CO2 forecasting, I don't think specific data on that is provided by ENTSO-E and we'd probably have to calculate estimates ourselves, which is a pretty big task, arguably out of scope for evcc, and unfortunately probably done best at the continent level by services like electricitymaps, etc.

andig, 2023-09-03:
Year, CO2 means mangling production and consumption plus specific CO2 per production type. I wouldn't try that honestly.

duckfullstop, 2023-09-03:
Spent a little time reading the documentation for ENTSO-E in more detail, and section 2. has some specific requirements for requesting an API key:

> To request access to the Restful API, please register on the Transparency Platform ([link](https://transparency.entsoe.eu/)) and send an email to transparency@entsoe.eu with “Restful API access” in the subject line. Indicate the email address you entered during registration in the email body. The ENTSO-E Helpdesk will make their best efforts to respond to your request within 3 working days.

https://transparency.entsoe.eu/content/static_content/Static%20content/web%20api/Guide.html#_authentication_and_authorisation

This feels like they don't intend this API for use by end users - I'm still happy to request a key and implement it, but it will substantially limit the ease of use.

RTTTC, 2023-09-03:
Maybe we could all use one key ? evcc as an organisation?

duckfullstop, 2023-09-03:
> Maybe we could all use one key ? evcc as an organisation?

Doesn't seem to be the case, these keys seem to be private use only:

> If you suspect that the token has been compromised, you must reset the token by yourself.
> If the token is missing or invalid (including suspended account), the system will respond with Http 401 Unauthorized.

I'm pretty sure some of these calls could be quite heavy (like repeatedly calling for decades of complex data at once), so it's not a surprise this restriction exists.

duckfullstop, 2023-09-06:
I have received an API key, starting work on implementing this now. Should have something rudimentary by next week at the latest (my health's a bit up and down!).

duckfullstop, 2023-09-08:
@bsmout @RTTTC Hey there! I'm busy implementing this for you now in #9794, but as someone that's never really dealt with european tariffs, it would be beneficial to know how you would expect this to work once it's finished.

Does your provider give you a specific ENTSO-E domain code (i.e `10YDE-ENBW-----N` for DETransnetW, for example), or would you expect to just define something like `DETransnetW` or `NL` for example?

And when it comes to how you get charged based on these rates, is it a simple multiplier (say 115% of the ENTSO-E cost), or would more advanced math be needed (say `(rate * 1.15) + 0.02`)?

Many thanks in advance!

djfanatix, 2023-09-08:
Hi,

Speaking for Belgium, but I think for the netherlands, germany and some other countries it is the same.
We have only 1 spot price, so BE,NL,DE,.. will be sufficient.

The different providers use a calculation which is complicated:

(((Entsoespot* 1.15) + 0.02)*tax) + government costs)

andig, 2023-09-08:
Tbo, we don‘t necessarily need the exact costs. I‘d prefer to keep an identical model across tariffs, currently in embed.go.

RTTTC, 2023-09-08:
I would expect to use country code (LT) There are a couple of countries in EU that have few regions per country, such as Sweden, UK, Norway. In such case there is NO1, NO2 etc. (As per ENTSO-E)

For me` price = (rate + fixed addon fees)*1,21` (VAT)
https://github.com/evcc-io/evcc/pull/7437 integration works perfectly for me (may be helpful to check out it's implementation?)

andig, 2023-09-08:
Think that is https://github.com/evcc-io/evcc/blob/master/tariff/embed.go

PR #9794 (2023-10-03): Feature: ENTSO-E "Day Ahead Pricing" tariff provider
This Pull Request adds support for the European Network of Transmission System Operators for Electricity's Transparency API, as documented at https://transparency.entsoe.eu/content/static_content/Static%20content/web%20api/Guide.html.

Still a bit of a WIP, wanted to get something down even if it's a bit scrappy at the moment. Still undecided whether I want to refactor to use POST instead of GET, but that's a bit of a pain to do with Go without a lot of self-repetition for core elements. For our use case, it's probably fine to stick to GET.
 [... cut]
closed 2023-10-03 by andig