evcc-io/evcc #5298: Fälschliches Laden startet nach evcc update

fabsenet, 2022-11-28:
### Describe the bug

Nach einem Update von evcc bekomme ich durchweg nur Timeouts zu sehen beim Fahrzeug (Tesla Model 3) abfragen und er beginnt die Mindestladung fälschlicherweise. Ich komme da nur raus, in dem ich evcc nochmal neu starte.
[image]
[image]

### Steps to reproduce

1. Tesla Model 3 ist angesteckt zum Laden, ist aber seit 1+ Tagen im Tiefschlaf. Der Akku steht zwischen Mindestladung und Maxladung. Evcc steht auf "PV". Alles ist soweit richtig.
2. Ich mache ein update von evcc auf die neueste Version in Home Assistant > Addons.
3. Evcc startet, bekommt von der Tesla API (mutmaßlich) keine Antwort und entscheidet dann, die Wallbox zu starten und die Mindestladung erreichen zu müssen.
4. Ich beobachte z.B. über Teslamate, dass das Auto aufhört mit Tiefschlaf, da es ja geladen wird, aber auch 10 Minuten später oder so berappelt sich evcc nicht von alleine.
5. Ich starte evcc neu, alles wird sofort erkannt und ist korrekt und er schaltet die Wallbox mit der normalen konfigurierten Wartezeit ab, da ja die Mindestladung in Wahrheit erreicht ist und PV Überschuss nicht existiert.

### Configuration details

```yaml
Tesla Model 3. MinSOC auf 50, max auf 85

HomeFix+ wallbox

...kann die Config hier rein kopieren, aber kann mir aktuell nicht vorstellen, dass da die Ursache ist?
```

### Log details

```text
Könnte ich versuchen, nächstes mal zu machen.
```

### What type of operating system are you running?

Docker container

### Version

Mehrere

andig, 2022-11-28:
Dass das laden aufgrund minSoc bei fehlendem Soc startet ist erwartet, also alles richtig.

closed 2022-11-28 by andig

fabsenet, 2022-11-28:
bitte reopen.

auto war im tiefschlaf:
[image]

ich habe log: debug in die config gemacht und dann evcc einfach nur neugestartet, es beginnt zu laden.

```log
[site  ] DEBUG 2022/11/28 14:54:49 ----
[lp-1  ] DEBUG 2022/11/28 14:54:49 charge power: 11260W
[lp-2  ] DEBUG 2022/11/28 14:54:49 charge power: 2557W
[site  ] DEBUG 2022/11/28 14:54:51 pv power: 152W
[site  ] DEBUG 2022/11/28 14:54:51 battery power: 3W
[site  ] DEBUG 2022/11/28 14:54:51 grid power: 14370W
[site  ] DEBUG 2022/11/28 14:54:51 grid currents: [20.4 21.2 20.5]A
[site  ] DEBUG 2022/11/28 14:54:51 battery soc 0: 8%
[site  ] DEBUG 2022/11/28 14:54:51 site power: 14373W
[lp-2  ] DEBUG 2022/11/28 14:54:51 charge currents: [4.6 4.06 4.28]A
[lp-2  ] DEBUG 2022/11/28 14:54:51 charger status: B
[lp-2  ] DEBUG 2022/11/28 14:54:51 next soc poll remaining time: 49m19s
[lp-2  ] DEBUG 2022/11/28 14:54:51 pv charge current: 0A = 0A + -20.8A (14373W @ 3p)
[site  ] DEBUG 2022/11/28 14:54:59 ----
[lp-1  ] DEBUG 2022/11/28 14:54:59 charge power: 11300W
[lp-2  ] DEBUG 2022/11/28 14:54:59 charge power: 2563W
[site  ] DEBUG 2022/11/28 14:55:00 pv power: 152W
[site  ] DEBUG 2022/11/28 14:55:01 battery power: 3W
[site  ] DEBUG 2022/11/28 14:55:01 grid power: 14360W
[site  ] DEBUG 2022/11/28 14:55:01 grid currents: [20.4 21.1 20.5]A
[site  ] DEBUG 2022/11/28 14:55:01 battery soc 0: 8%
[site  ] DEBUG 2022/11/28 14:55:01 site power: 14363W
[lp-1  ] DEBUG 2022/11/28 14:55:02 charge currents: [16.1 16.1 16.1]A
[lp-1  ] DEBUG 2022/11/28 14:55:02 detected phases: 3p
[lp-1  ] DEBUG 2022/11/28 14:55:02 charger status: C
[lp-1  ] DEBUG 2022/11/28 14:55:02 next soc poll remaining time: 59m39s
[lp-1  ] ERROR 2022/11/28 14:55:02 vehicle soc: 408 Request Timeout
[lp-1  ] DEBUG 2022/11/28 14:55:02 pv timer elapse
[lp-1  ] DEBUG 2022/11/28 14:55:02 pv timer inactive
[site  ] DEBUG 2022/11/28 14:55:09 ----
[lp-1  ] DEBUG 2022/11/28 14:55:09 charge power: 11240W
[lp-2  ] DEBUG 2022/11/28 14:55:09 charge power: 2564W
[site  ] DEBUG 2022/11/28 14:55:10 pv power: 152W
[site  ] DEBUG 2022/11/28 14:55:11 battery power: 3W
[site  ] DEBUG 2022/11/28 14:55:11 grid power: 14340W
[site  ] DEBUG 2022/11/28 14:55:11 grid currents: [20.4 21.1 20.5]A
[site  ] DEBUG 2022/11/28 14:55:11 battery soc 0: 8%
[site  ] DEBUG 2022/11/28 14:55:11 site power: 14343W
[lp-2  ] DEBUG 2022/11/28 14:55:11 charge currents: [4.61 4.06 4.29]A
[lp-2  ] DEBUG 2022/11/28 14:55:11 charger status: B
[lp-2  ] DEBUG 2022/11/28 14:55:11 next soc poll remaining time: 48m59s
[lp-2  ] DEBUG 2022/11/28 14:55:11 pv charge current: 0A = 0A + -20.8A (14343W @ 3p)
[site  ] DEBUG 2022/11/28 14:55:19 ----
[lp-1  ] DEBUG 2022/11/28 14:55:19 charge power: 11250W
[lp-2  ] DEBUG 2022/11/28 14:55:19 charge power: 2561W
[site  ] DEBUG 2022/11/28 14:55:20 pv power: 151W
[site  ] DEBUG 2022/11/28 14:55:20 battery power: 0W
[site  ] DEBUG 2022/11/28 14:55:20 grid power: 14350W
[site  ] DEBUG 2022/11/28 14:55:20 grid currents: [20.3 21.2 20.4]A
[site  ] DEBUG 2022/11/28 14:55:21 battery soc 0: 8%
[site  ] DEBUG 2022/11/28 14:55:21 site power: 14350W
[lp-1  ] DEBUG 2022/11/28 14:55:21 charge currents: [16 16.1 16]A
[lp-1  ] DEBUG 2022/11/28 14:55:21 detected phases: 3p
[lp-1  ] DEBUG 2022/11/28 14:55:21 charger status: C
[lp-1  ] DEBUG 2022/11/28 14:55:21 next soc poll remaining time: 59m40s
[lp-1  ] ERROR 2022/11/28 14:55:21 vehicle soc: 408 Request Timeout
[lp-1  ] DEBUG 2022/11/28 14:55:21 pv timer elapse
[lp-1  ] DEBUG 2022/11/28 14:55:21 pv timer inactive
[site  ] DEBUG 2022/11/28 14:55:29 ----
[lp-1  ] DEBUG 2022/11/28 14:55:29 charge power: 11260W
[lp-2  ] DEBUG 2022/11/28 14:55:29 charge power: 2567W
[site  ] DEBUG 2022/11/28 14:55:30 pv power: 149W
[... 54 lines cut ...]
[site  ] DEBUG 2022/11/28 15:03:19 ----
[lp-1  ] DEBUG 2022/11/28 15:03:19 charge power: 4300W
[lp-2  ] DEBUG 2022/11/28 15:03:19 charge power: 2601W
[site  ] DEBUG 2022/11/28 15:03:20 pv power: 129W
[site  ] DEBUG 2022/11/28 15:03:21 battery power: 3W
[site  ] DEBUG 2022/11/28 15:03:21 grid power: 7360W
[site  ] DEBUG 2022/11/28 15:03:21 grid currents: [10.5 11.2 10.5]A
[site  ] DEBUG 2022/11/28 15:03:21 battery soc 0: 7%
[site  ] DEBUG 2022/11/28 15:03:21 site power: 7363W
[lp-1  ] DEBUG 2022/11/28 15:03:21 charge currents: [6.1 6.1 6.1]A
[lp-1  ] DEBUG 2022/11/28 15:03:21 detected phases: 3p
[lp-1  ] DEBUG 2022/11/28 15:03:21 charger status: C
[lp-1  ] DEBUG 2022/11/28 15:03:21 next soc poll remaining time: 59m39s
[lp-1  ] DEBUG 2022/11/28 15:03:21 soc estimated: 56.32% (vehicle: 56.00%)
[lp-1  ] DEBUG 2022/11/28 15:03:21 vehicle soc: 56%
[lp-1  ] DEBUG 2022/11/28 15:03:21 vehicle range: 319km
[lp-1  ] DEBUG 2022/11/28 15:03:21 vehicle target soc: 100%
[lp-1  ] DEBUG 2022/11/28 15:03:21 available power -3063W < 4140W min 3p threshold
[lp-1  ] DEBUG 2022/11/28 15:03:21 phase scale1p in 1m0s
[lp-1  ] DEBUG 2022/11/28 15:03:21 pv charge current: 0A = 6A + -10.7A (7363W @ 3p)
```

unterm strich hat er jetzt 20 minuten geladen ohne not.

Teslamate wusste bereits 20s nach start des Ladevorgangs, dass das auto nicht mehr schläft, sondern lädt. ggf. kann man das soc pollen auch einfach schneller machen nach dem start des ladevorgangs?

fabsenet, 2022-11-28:
Es scheint wichtig zu sein, dass das Auto schläft beim Start von evcc. Hier nochmal im Überblick:

```text
Using config file: /config/evcc.yaml
starting evcc: 'evcc --config /config/evcc.yaml --sqlite /data/evcc.db'
[main  ] INFO 2022/11/28 15:13:46 evcc 0.108.2
[main  ] INFO 2022/11/28 15:13:46 using config file: /config/evcc.yaml
[main  ] INFO 2022/11/28 15:13:46 starting ui and api at :7070
[site  ] WARN 2022/11/28 15:13:48 vehicle 'Wärmepumpe' does not support automatic detection
[site  ] INFO 2022/11/28 15:13:48 site config:
[site  ] INFO 2022/11/28 15:13:48   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2022/11/28 15:13:48     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2022/11/28 15:13:48     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/11/28 15:13:48     pv 2:      power ✓ energy ✓ currents ✗
[site  ] INFO 2022/11/28 15:13:48     battery 1: power ✓ energy ✗ currents ✗ soc ✓
[site  ] INFO 2022/11/28 15:13:48   vehicles:
[site  ] INFO 2022/11/28 15:13:48     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2022/11/28 15:13:48     vehicle 2: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
[lp-1  ] INFO 2022/11/28 15:13:48 loadpoint 1:
[lp-1  ] INFO 2022/11/28 15:13:48   mode:        pv
[lp-1  ] INFO 2022/11/28 15:13:48   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2022/11/28 15:13:48   meters:      charge ✓
[lp-1  ] INFO 2022/11/28 15:13:48     charge:    power ✓ energy ✓ currents ✓
[lp-2  ] INFO 2022/11/28 15:13:48 loadpoint 2:
[lp-2  ] INFO 2022/11/28 15:13:48   mode:        pv
[lp-2  ] INFO 2022/11/28 15:13:48   charger:     power ✓ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-2  ] INFO 2022/11/28 15:13:48   meters:      charge ✓
[lp-2  ] INFO 2022/11/28 15:13:48     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2022/11/28 15:13:48 vehicle updated: unknown -> Model 3
[lp-2  ] INFO 2022/11/28 15:13:48 vehicle updated: unknown -> Wärmepumpe
[lp-1  ] ERROR 2022/11/28 15:13:59 vehicle odometer: 408 Request Timeout
[lp-1  ] INFO 2022/11/28 15:14:00 car connected
[lp-1  ] ERROR 2022/11/28 15:14:09 vehicle soc: 408 Request Timeout
[lp-1  ] ERROR 2022/11/28 15:14:11 vehicle odometer: 408 Request Timeout
[lp-2  ] INFO 2022/11/28 15:14:13 car connected
[lp-1  ] INFO 2022/11/28 15:14:20 start charging ->
[lp-1  ] ERROR 2022/11/28 15:14:20 vehicle soc: 408 Request Timeout
[lp-1  ] ERROR 2022/11/28 15:14:40 vehicle soc: 408 Request Timeout
[lp-1  ] ERROR 2022/11/28 15:14:59 vehicle soc: 408 Request Timeout
[lp-1  ] ERROR 2022/11/28 15:15:20 vehicle soc: 408 Request Timeout
[lp-1  ] ERROR 2022/11/28 15:15:40 vehicle soc: 408 Request Timeout
[lp-1  ] ERROR 2022/11/28 15:16:00 vehicle soc: 408 Request Timeout
[lp-1  ] ERROR 2022/11/28 15:16:19 vehicle soc: 408 Request Timeout
[lp-1  ] ERROR 2022/11/28 15:16:40 vehicle soc: 408 Request Timeout
[lp-1  ] ERROR 2022/11/28 15:17:00 vehicle soc: 408 Request Timeout
[lp-1  ] ERROR 2022/11/28 15:17:20 vehicle soc: 408 Request Timeout
[lp-1  ] ERROR 2022/11/28 15:17:40 vehicle soc: 408 Request Timeout
[lp-1  ] ERROR 2022/11/28 15:17:59 vehicle soc: 408 Request Timeout
```

fabsenet, 2022-11-28:
Ich dachte, ich kann vielleicht mehr zu den `vehicle soc: 408 Request Timeout` rausloggen, aber bisher nicht. Mein letzter Versuch war:

```yaml
levels:
  site: info
  lp-1: trace
  lp-2: error
  teslaModel3: trace
```

fabsenet, 2022-11-28:
Eine Lösung könnte durchaus sein, den letzten State vom Tesla zu persistieren und nach einem Neustart von evcc wieder raus zu holen, bis der erste API call durchkommt.

Empfang ist kein Thema, unsere Garage hat einen Repeater der über Netzwerkkabel am Hausnetz hängt. Das Auto geht einfach in einen Schlafmodus und antwortet deshalb nicht. Das ist aber normal, ich hab da nichts gemacht.

andig, 2022-11-28:
Nochmal: wenn wir keinen SoC haben wird bei minSoC geladen. Mehr ist erstmal vorgesehen. Fahrbereitschaft des Fahrzeugs hat Vorrang!