evcc-io/evcc #5303: Handle L2 inactive

andig, 2022-11-28:
@RTTTC I realise that your case is special. You habe L1+3 active while UI shows L1+2. Need to rethink that, even if border case.

_Originally posted by @andig in https://github.com/evcc-io/evcc/discussions/5297#discussioncomment-4253470_

RTTTC, 2022-11-28:
When car is not charging or heating - current it is  0A 0A 0A

As the weather is getting colder - more often preheat is used, looks like it's not utilising all phases equally while preheating, but actually all phases active.

Should be same for all VAG MEB platform vehicles.

premultiply, 2022-11-28:
As L3 is active it should be displayed as 3p charging regardless of L2 < 1.0A

Imho should be like
1p: L1&!L2&!L3
2p: L1&L2&!L3
3p: else

/cc @naltatis

andig, 2022-11-28:
The general question is which phases the ui should display if n phases are active and are not continguous? Note this is display-only.

premultiply, 2022-11-28:
Imho as written before: 1p and 2p only in "special cases". In any other case 3p.

andig, 2022-11-30:
Vorschlag:
- im UI zeigen wir 1/2 Phasen gemäß `phasesActive` genau dann an, wenn das die beiden Phasen mit den höchsten Strömen sind
- in allen anderen Fällen wird 3p angezeigt

das ist identisch zu https://github.com/evcc-io/evcc/issues/5303#issuecomment-1329778959, aber vielleicht einfacher zu verstehen ;)

premultiply, 2022-11-30:
Klingt gut 👍

naltatis, 2022-11-30:
@andig @premultiply Ich glaub ich verstehe noch nicht ganz was die Idee für die Umsetzung ist.

Aktuell wird in der UI die Anzahl der Balken durch den Wert in `phasesActive` bestimmt. Die Currents werde nicht interpretiert sondern nur für die Balkenfüllung [1..3] verwendet. Das finde ich auch gut weil es die Logik aus der UI hält.

Heißt um den gewünschten Effekt zu bekommen würde ich erwarten, dass die API bei 10A, 0.5A, 10A ein `phasesActive: 3` liefert, richtig?

premultiply, 2022-12-01:
Ja, werden wir wohl nochmal überdenken müssen.

andig, 2022-12-01:
Müssen wir nicht ;). Der Wert ist für die Ladesteuerung relevant und passt dafür auch. Es wird auf 2p geladen. Hier passt einfach die Visualisierung nicht. War mit @naltatis auch so besprochen.

andig, 2022-12-01:
Die zu treffende Entscheidung ist ob wir den Sonderfall korrekt visualisieren wollen (3p) oder einfach Sonderfall sein lassen.

RTTTC, 2022-12-01:
What if three bars were always shown, but only filled as the corresponding phase gets utilised?

PR #5345 (2022-12-06): Better phase UI edge cases
Handling asymmetric phase usages, like Skoda's climating with [10A, 0A, 10A]. Previously `phasesActive` defined the number of bars. Now we check the individual currents (if available). The highest active phase defines the UI.

fixes #5303

All states
[image]

---

New Cases

[image]
[image]
[image]
closed 2022-12-06 by naltatis