evcc-io/evcc #5308: mincurrect / maxcurrent wird nicht berücksichtigt

Olf1337, 2022-11-28:
### Describe the bug

if there is no meter for the wallbox evcc will always display the maximum possible power for the configures phases:

e.g.:
phases = 1 = 3,7kwH
phases = 3 = 11 kwH

### Steps to reproduce

1. configure evcc
2. don't use a meter
3. configure mincurrent and maxcurrent for a vehicle
4.   phases: 1
  minCurrent: 6
  maxCurrent: 12.65
5. start charging and you'll see always 3,7kwH instead of configured in step 4 2,9kwH

### Configuration details

```yaml
vehicles:
- type: template
  template: audi
  name: ev1
  title: Name IT
  user: xxxxxxxxxxxxxxxxx
  password: xxxxxxxxxxxxxxx  
  vin: xxxxxxxxxxxxx
  capacity: xx
  phases: 1
  cache: 5
  mode: now
  minCurrent: 6
  maxCurrent: 12.65
```

### Log details

```text
evcc --log debug        
[main  ] INFO 2022/11/28 16:45:16 evcc 0.108.2
[main  ] INFO 2022/11/28 16:45:16 using config file: /etc/evcc.yaml
[main  ] INFO 2022/11/28 16:45:16 starting ui and api at :7070
[eebus ] WARN 2022/11/28 16:45:18 failed to send current limit data:  no eletrical paramaters, measurements, or limits available yet
[site  ] INFO 2022/11/28 16:45:18 site config:
[site  ] INFO 2022/11/28 16:45:18   meters:      grid â pv â battery â
[site  ] INFO 2022/11/28 16:45:18     grid:      power â energy â currents â
[site  ] INFO 2022/11/28 16:45:18     pv 1:      power â energy â currents â
[site  ] INFO 2022/11/28 16:45:18     battery 1: power â energy â currents â soc â
[site  ] INFO 2022/11/28 16:45:18   vehicles:
[site  ] INFO 2022/11/28 16:45:18     vehicle 1: range â finish â status â climate â wakeup â
[lp-1  ] INFO 2022/11/28 16:45:18 loadpoint 1:
[lp-1  ] INFO 2022/11/28 16:45:18   mode:        now
[lp-1  ] INFO 2022/11/28 16:45:18   charger:     power â energy â currents â phases â wakeup â
[lp-1  ] INFO 2022/11/28 16:45:18   meters:      charge â
[lp-2  ] INFO 2022/11/28 16:45:18 loadpoint 2:
[lp-2  ] INFO 2022/11/28 16:45:18   mode:        now
[lp-2  ] INFO 2022/11/28 16:45:18   charger:     power â energy â currents â phases â wakeup â
[lp-2  ] INFO 2022/11/28 16:45:18   meters:      charge â
[lp-1  ] DEBUG 2022/11/28 16:45:18 phase timer inactive
[lp-1  ] DEBUG 2022/11/28 16:45:18 pv timer inactive
[lp-2  ] DEBUG 2022/11/28 16:45:18 phase timer inactive
[lp-2  ] DEBUG 2022/11/28 16:45:18 pv timer inactive
[site  ] DEBUG 2022/11/28 16:45:18 ----
[lp-1  ] DEBUG 2022/11/28 16:45:18 charge power: 0W
[lp-2  ] DEBUG 2022/11/28 16:45:18 charge power: 0W
[site  ] DEBUG 2022/11/28 16:45:20 pv power: 0W
[site  ] DEBUG 2022/11/28 16:45:22 battery power: -19W
[site  ] DEBUG 2022/11/28 16:45:22 grid power: 1886W
[site  ] DEBUG 2022/11/28 16:45:25 grid currents: [1.25 1.66 8.89]A
[site  ] DEBUG 2022/11/28 16:45:27 battery soc 0: 2%
[site  ] DEBUG 2022/11/28 16:45:27 site power: 1867W
[lp-1  ] DEBUG 2022/11/28 16:45:27 charger status: B
[lp-1  ] INFO 2022/11/28 16:45:27 car connected
[lp-1  ] DEBUG 2022/11/28 16:45:27 vehicle api refresh
[lp-1  ] DEBUG 2022/11/28 16:45:27 pv timer elapse
[lp-1  ] DEBUG 2022/11/28 16:45:27 pv timer inactive
[lp-1  ] ERROR 2022/11/28 16:45:27 charger vehicle id: must retry
[main  ] DEBUG 2022/11/28 16:45:28 vehicle status: B (Name IT)
[lp-1  ] INFO 2022/11/28 16:45:28 vehicle updated: unknown -> Audi A3 TFSIe
[lp-1  ] DEBUG 2022/11/28 16:45:28 set charge mode: now
[lp-1  ] DEBUG 2022/11/28 16:45:28 set min current: 6
[lp-1  ] DEBUG 2022/11/28 16:45:28 set max current: 12.65
[lp-1  ] DEBUG 2022/11/28 16:45:28 set target soc: 100
[lp-1  ] DEBUG 2022/11/28 16:45:28 vehicle status: B
[lp-1  ] DEBUG 2022/11/28 16:45:28 vehicle soc: 100%
[lp-1  ] DEBUG 2022/11/28 16:45:28 vehicle range: 51km
[lp-1  ] ERROR 2022/11/28 16:45:28 max charge current 12.7A: no eletrical paramaters, measurements, or limits available yet
[site  ] DEBUG 2022/11/28 16:45:28 ----
[lp-1  ] DEBUG 2022/11/28 16:45:28 charge power: 0W
[lp-2  ] DEBUG 2022/11/28 16:45:28 charge power: 0W
[site  ] DEBUG 2022/11/28 16:45:29 pv power: 0W
[site  ] DEBUG 2022/11/28 16:45:30 battery power: -19W
[eebus ] WARN 2022/11/28 16:45:30 failed to send current limit data:  no ev connected
[site  ] DEBUG 2022/11/28 16:45:30 grid power: 3235W
[eebus ] WARN 2022/11/28 16:45:31 failed to send current limit data:  no eletrical paramaters, measurements, or limits available yet
[site  ] DEBUG 2022/11/28 16:45:32 grid currents: [1.25 1.66 7.82]A
[lp-1  ] DEBUG 2022/11/28 16:45:33 set max current: 16
[site  ] DEBUG 2022/11/28 16:45:34 battery soc 0: 2%
[site  ] DEBUG 2022/11/28 16:45:34 site power: 3216W
[lp-2  ] DEBUG 2022/11/28 16:45:34 charger status: A
[lp-2  ] INFO 2022/11/28 16:45:34 car disconnected
[lp-2  ] DEBUG 2022/11/28 16:45:34 set charge mode: now
[lp-2  ] DEBUG 2022/11/28 16:45:34 set min current: 6
[lp-2  ] DEBUG 2022/11/28 16:45:34 set max current: 16
[lp-2  ] DEBUG 2022/11/28 16:45:34 set min soc: 0
[lp-2  ] DEBUG 2022/11/28 16:45:34 set target soc: 100
[site  ] DEBUG 2022/11/28 16:45:38 ----
[lp-1  ] DEBUG 2022/11/28 16:45:38 charge power: 0W
[lp-2  ] DEBUG 2022/11/28 16:45:38 charge power: 0W
[site  ] DEBUG 2022/11/28 16:45:38 pv power: 0W
[site  ] DEBUG 2022/11/28 16:45:38 battery power: -19W
[site  ] DEBUG 2022/11/28 16:45:39 grid power: 2141W
[site  ] DEBUG 2022/11/28 16:45:41 grid currents: [1.23 1.66 8.05]A
[site  ] DEBUG 2022/11/28 16:45:43 battery soc 0: 2%
[site  ] DEBUG 2022/11/28 16:45:43 site power: 2122W
[lp-1  ] DEBUG 2022/11/28 16:45:44 vehicle odometer: 8236km
[lp-1  ] DEBUG 2022/11/28 16:45:44 charger status: B
[lp-1  ] DEBUG 2022/11/28 16:45:44 max charge current: 16A
[lp-1  ] DEBUG 2022/11/28 16:45:44 charger enable
[lp-1  ] DEBUG 2022/11/28 16:45:44 wake-up timer: start
[site  ] DEBUG 2022/11/28 16:45:48 ----
[lp-1  ] DEBUG 2022/11/28 16:45:48 charge power: 0W
[lp-2  ] DEBUG 2022/11/28 16:45:48 charge power: 0W
[site  ] DEBUG 2022/11/28 16:45:50 pv power: 0W
[site  ] DEBUG 2022/11/28 16:45:50 battery power: -18W
[site  ] DEBUG 2022/11/28 16:45:51 grid power: 2139W
[site  ] DEBUG 2022/11/28 16:45:53 grid currents: [1.25 1.67 10]A
[site  ] DEBUG 2022/11/28 16:45:55 battery soc 0: 2%
[site  ] DEBUG 2022/11/28 16:45:55 site power: 2121W
[lp-2  ] DEBUG 2022/11/28 16:45:55 charger status: A
[site  ] DEBUG 2022/11/28 16:45:58 ----
[lp-1  ] DEBUG 2022/11/28 16:45:58 charge power: 0W
[lp-2  ] DEBUG 2022/11/28 16:45:58 charge power: 0W
[site  ] DEBUG 2022/11/28 16:45:59 pv power: 0W
[site  ] DEBUG 2022/11/28 16:45:59 battery power: -19W
[site  ] DEBUG 2022/11/28 16:46:00 grid power: 3295W
[site  ] DEBUG 2022/11/28 16:46:03 grid currents: [1.25 1.66 13.8]A
[site  ] DEBUG 2022/11/28 16:46:05 battery soc 0: 2%
[site  ] DEBUG 2022/11/28 16:46:05 site power: 3276W
[lp-1  ] DEBUG 2022/11/28 16:46:05 charger status: C
[lp-1  ] INFO 2022/11/28 16:46:05 start charging ->
[lp-1  ] DEBUG 2022/11/28 16:46:05 soc estimated: 100.00% (vehicle: 100.00%)
[lp-1  ] DEBUG 2022/11/28 16:46:05 vehicle soc: 100%
[lp-1  ] DEBUG 2022/11/28 16:46:05 vehicle range: 51km
[site  ] DEBUG 2022/11/28 16:46:08 ----
[lp-1  ] DEBUG 2022/11/28 16:46:08 charge power: 3680W
[lp-2  ] DEBUG 2022/11/28 16:46:08 charge power: 0W
[site  ] DEBUG 2022/11/28 16:46:10 pv power: 0W
[site  ] DEBUG 2022/11/28 16:46:10 battery power: -19W
[site  ] DEBUG 2022/11/28 16:46:11 grid power: 3550W
[site  ] DEBUG 2022/11/28 16:46:13 grid currents: [1.24 1.66 14.3]A
[site  ] DEBUG 2022/11/28 16:46:17 battery soc 0: 2%
[site  ] DEBUG 2022/11/28 16:46:17 site power: 3531W
[lp-2  ] DEBUG 2022/11/28 16:46:17 charger status: A
[site  ] DEBUG 2022/11/28 16:46:18 ----
[lp-1  ] DEBUG 2022/11/28 16:46:18 charge power: 3680W
[lp-2  ] DEBUG 2022/11/28 16:46:18 charge power: 0W
[site  ] DEBUG 2022/11/28 16:46:19 pv power: 0W
[site  ] DEBUG 2022/11/28 16:46:20 battery power: -19W
[site  ] DEBUG 2022/11/28 16:46:20 grid power: 3599W
[site  ] DEBUG 2022/11/28 16:46:22 grid currents: [1.24 1.66 14.5]A
[site  ] DEBUG 2022/11/28 16:46:24 battery soc 0: 2%
[site  ] DEBUG 2022/11/28 16:46:24 site power: 3580W
[lp-1  ] DEBUG 2022/11/28 16:46:24 charger status: C
[lp-1  ] DEBUG 2022/11/28 16:46:24 soc estimated: 100.00% (vehicle: 100.00%)
[lp-1  ] DEBUG 2022/11/28 16:46:24 vehicle soc: 100%
[lp-1  ] DEBUG 2022/11/28 16:46:24 vehicle range: 51km
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.108.2

andig, 2022-11-30:
@Olf1337 kann ich nicht nachvollziehen. Wenn ich in der Demo einen Charger ohne Zähler nehme und am Auto 3p/10A max konfiguriere dann sehe ich 6,9kW was genau der Erwartungswert ist. 

> max charge current 12.7A: no eletrical paramaters, measurements, or limits available yet

Du betreibst keine freigegebene Version.

/cc @hoermto evtl. noch eine Thema für Deinen PR, kann ich nicht beurteilen.

hoermto, 2022-11-30:
@Olf1337 ich kann das Problem noch nicht nachvollziehen. Der max Ladestrom wird nicht benutzt, wenn im Charger definiert?
Benutzt Du die Mainrelease 0.108.2 oder die vom Lastmanagement? Falls Lastmanagement, tritt das Verhalten auch auf wenn Du die offizielle 0.108.2 verwendest? Im Log kann ich keine Circuits erkennen, daher scheint mir das nicht am Lastmanagement zu liegen @andig 
Allerdings sehe ich im Log mehrere Warnings / Error wie zB `[eebus ] WARN 2022/11/28 16:45:31 failed to send current limit data:  no eletrical paramaters, measurements, or limits available yet`

andig, 2022-11-30:
Ah, sorry. Dann hab ich das falsch zugeordnet. In der Demo kann ich auf jeden Fall nachvollziehen, dass die Leistung stimmt.

Olf1337, 2022-12-02:
Hä?

Im log wird nur die Grid Power ausgewiesen. Das ist aber kumuliert was das Haus verbraucht inkl. was der Charger abgibt. 

Dadurch das evcc die maxcurrent wohl nicht korrekt anzeigt, steht der Charger immer bei 3,7kwh obwohl nur Max 2,7 möglich sind. Egal wenn ich beim Auto 1p/12A angebe oder sonst was.

andig, 2022-12-03:
> Dadurch das evcc die maxcurrent wohl nicht korrekt anzeigt, steht der Charger immer bei 3,7kwh obwohl nur Max 2,7 möglich sind. 

Dieser Satz kein Sinn. Wenn Du weitere Hilfe brauchst gib bitte Deine vollständige Config an.

andig, 2022-12-03:
Wir haben die Ursache nach Diskussion in Slack identifiziert:

```
[eebus ] WARN 2022/11/28 16:45:31 failed to send current limit data:  no eletrical paramaters, measurements, or limits available yet
[site  ] DEBUG 2022/11/28 16:45:32 grid currents: [1.25 1.66 7.82]A
[lp-1  ] DEBUG 2022/11/28 16:45:33 set max current: 16
```

Das Auto übersteuert via EEBus die Einstellungen des Loadpoints.

DerAndereAndi, 2022-12-03:
Mit EEBUS werden die Limits neu gesetzt, da manche Wallboxen auch Laden per ISO15118-2 ermöglichen und damit minCurrent von 2,2A. Das bedeutet das Minimum ist abhängig von der Kombination: angeschlossenes EV an der Wallbox und eingestellte Kommunikation.

Um das sauber auch für nicht EEBUS Wallboxen mit ISO zu unterstützen, brächte es vielleicht eine generelle Logik wie in etwa:
- EV wird angesteckt
- Limits werden von der Wallbox mitgeteilt und gesetzt
- Wenn das EV erkannt ist und eigene Limits hat, werden diese genommen und die Wallbox darf keine Limits setzen

andig, 2022-12-03:
In diesem Fall hier wurde erst dann Fahrzeug erkannt (und dessen Limits gesetzt)- dann hat die Wallbox wieder übersteuert.

Olf1337, 2022-12-05:
@andig  anbei die config:
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  site: error
#  lp-2: trace

# unique installation id
plant: x

interval: 10s # control cycle interval

meters:
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.178.8  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.178.8  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.178.8  
  port: 502  
  usage: battery  
  modbus: tcpip  
  name: battery3

chargers:
- type: template
  template: elliconnect 
  ski: EB93 6EA4 1C62 2FD3 A99E 4CF6 DF9D 24EB 4647 BAE2  
  ip: 192.168.178.10  
  name: wallbox9
- type: template
  template: elliconnect 
  ski: 3D16 3CFD C326 9CD1 4D58 D3EC 2242 CDEA CA1B 26DC 
  ip: 192.168.178.9  
  name: wallbox8

vehicles:
- type: template
  template: audi
  name: ev10
  title: Audi A3 TFSIe 
  user: x  
  password: x  
  vin: x
  capacity: 13
  phases: 1
  cache: 5
  mode: now
  minCurrent: 6
  targetSoC: 100
  maxCurrent: 12.65

loadpoints:
- title: Links
  charger: wallbox9
  mode: now
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
- title: Rechts
  charger: wallbox8
  mode: now
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true

site:
  title: w00tcharge
  meters:
    grid: grid1
    pvs:
    - pv2
    batteries:
    - battery3

eebus:
  certificate:
      private: |
          -----BEGIN EC PRIVATE KEY-----
          x
          -----END EC PRIVATE KEY-----
      public: |
          -----BEGIN CERTIFICATE-----
     x
          -----END CERTIFICATE-----

falls Ihr sonst noch Daten braucht kann ich die gerne liefern.