evcc-io/evcc #5311: go-eCharger V3 with 054.7 connecting using OCPP

stumbaumr, 2022-11-28:
### Describe the bug

Since 054.0ff go-eCharger V3 and Gemini implement OCPP. After successfully playing around with cmd/ocpp and getting a config to work I tried using the go-e with OCPP and failed. There is not even output coming in the trace log.

### Steps to reproduce

1. Configure evcc to use OCPP
```chargers:
  - name: go-eCharger
    type: template
#    template: go-e-v3
#    host: 192.168.178.22
    template: ocpp
    meter: false
#    stationid: go-echarger-05XXXX
#    stationid: cp0001
    idtag: 1
```
2. Configure go-eCharger to use OCPP (-> Internet -> OCPP )
4. Save
5. Look at the output...

### Configuration details

```yaml
chargers:
  - name: go-eCharger
    type: template
#    template: go-e-v3
#    host: 192.168.178.22
    template: ocpp
    meter: false
#    stationid: go-echarger-05XXXX
#    stationid: cp0001
    idtag: 1
```

### Log details

```text
root@raspberrypi:~# ./evcc
[main  ] INFO 2022/11/28 22:22:02 evcc 0.108.2
[main  ] INFO 2022/11/28 22:22:02 using config file: /root/evcc.yaml
[main  ] INFO 2022/11/28 22:22:02 starting ui and api at :7070
[db    ] TRACE 2022/11/28 22:22:02 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2022/11/28 22:22:02 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2022/11/28 22:22:02 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2022/11/28 22:22:02 SELECT * FROM `settings` 6 <nil>
[ocpp  ] DEBUG 2022/11/28 22:22:03 waiting for chargepoint: 10m0s

That's it...

I do have a tcpdump...
```

### What type of operating system are you running?

Docker container

### Version

evcc version 0.108.2

stumbaumr, 2022-11-28:
[wireshark.zip](https://github.com/evcc-io/evcc/files/10107864/wireshark.zip)

Sillium007, 2022-11-29:
This might be completely wrong but:

> [ocpp  ] DEBUG 2022/11/28 22:22:03 waiting for chargepoint: 10m0s

Did you wait 10 minutes? I don't know how OCPP works but at least this sounds to me like the Charger is supposed to do something and maybe the charger only does this initialization (or whatever it does) every now and then

stumbaumr, 2022-11-29:
Thanks for looking into it, but... ;-)
- This was my first idea as well, so yes, I waited the first time for 10 minutes...
- The tcpdump shows that the charger connects to the evcc and then receives a 404 not found response.

I used evcc's own OCPP test client in ./cmd/ocpp to try out my evcc config and also took a tcpdump from that - there the evcc does not respond with a 404 - but the request from the charger to me looks the same as from the test client...