evcc-io/evcc #5314: Add div() function/wrapper to calc

panzerdev, 2022-11-29:
To be able to calculate a the `L1, L2, L3` meter `current` a division operator would be handy. In my setup (Sungrow grid meter) there are no current values in the modbus registers I could find. But there is the `power per phase` and the `voltage per phase`. With that you can take the `power per phase` and divide it by the current `voltage per phase` to get the `current per phase`.

```
meters
  - name: grid
    type: custom
    [...]
    currents:      
      - source: calc
        div:
          - source: modbus
            [...]
          - source: modbus
            [...]
      - source: calc
        div:
          - source: modbus
            [...]
          - source: modbus
            [...]
      - source: calc
        div:
          - source: modbus
            [...]
          - source: modbus
            [...]
```

The code change is small, tested in my usecase and I would make a PR if this is something that could get merged into to repository.

andig, 2022-11-29:
You would then divide the first param bei all following ones? Or should that be

    dividend:
    divisor:

/cc @premultiply

panzerdev, 2022-11-29:
I think there are 2 reasonable ways to go about it

1: only 2 item in the div array are allowed and the first one is dividend and the 2nd one is the divisor.
2: like with add and multiply the result of the previous division is divided by the next value.

I think both have their pro's and cons. The 2nd approach would make it more flexible if the 3rd division would make it a percentage or so.

But in general I would keep it an array and not make it keys in the yaml.

andig, 2022-11-29:
> But in general I would keep it an array and not make it keys in the yaml.

Happy to take a PR

panzerdev, 2022-11-29:
PR #5319 is here for you to have a look at.

closed 2022-11-30 by andig