evcc-io/evcc #5337: Change "cheap" (tariffs) parameter via API/mqtt

souhl, 2022-12-01:
**Is your feature request related to a problem? Please describe.**
When using Tibber or Awattar in the season with not much PV production it is all about loading the car, when the energy is cheap.
Unfortunately the energy costs are not very stable these days and "cheap" can be 30 cent today and 40 cent tomorrow.
Currently you would need to update the evcc.yaml and restart EVCC once a day to update the value.

**Describe the solution you'd like**
Having the option to update the value via API/mqtt would allow to implement a switch in iobroker/home assistant to change the value manual or even automated by code.

[image]

**Additional context**
Tibber tariffs over the last days:

[image]

andig, 2022-12-01:
That's pretty easy to add. The "real" solution would be https://github.com/evcc-io/evcc/pull/2211 though?

souhl, 2022-12-01:
Ich habe eben nochmal in GraphQL API von Tibber geschaut und dort wird sogar mitgeliefert, ob der aktuelle Preis "cheap" ist. (PriceInfo ->today-> level) Man könnte also auch diese API nutzen, um darüber zu schalten, wenn der Preis (PriceLevel) "cheap" ist und wieder aufhören, wenn nicht mehr "cheap". So habe ich das jetzt gerade mal in iobroker nachgebaut und muss mal sehen, ob das so funtkioniert:

https://developer.tibber.com/docs/reference

```
switch(getState("tibberconnect.0.Homes.XXXXXXX.CurrentPrice.level").val)
    {
        case 'EXPENSIVE':
            console.log('Strom ist teuer');
            if (getState("0_userdata.0.roday_charge_cheap").val)
            {
                setState("mqtt.0.evcc.loadpoints.1.mode.set", "minpv");
            }
            if (getState("0_userdata.0.speedy_charge_cheap").val)
            {
                setState("mqtt.0.evcc.loadpoints.2.mode.set", "minpv");
            }
            break;
        case 'VERY_EXPENSIVE':
            console.log('Strom ist sehr teuer');
            if (getState("0_userdata.0.roday_charge_cheap").val)
            {
                setState("mqtt.0.evcc.loadpoints.1.mode.set", "minpv");
            }
            if (getState("0_userdata.0.speedy_charge_cheap").val)
            {
                setState("mqtt.0.evcc.loadpoints.2.mode.set", "minpv");
            }
            break;

        case 'CHEAP':
            console.log('Strom ist günstig');
            if (getState("0_userdata.0.roday_charge_cheap").val)
            {
                setState("mqtt.0.evcc.loadpoints.1.mode.set", "now");
            }
            if (getState("0_userdata.0.speedy_charge_cheap").val)
            {
                setState("mqtt.0.evcc.loadpoints.2.mode.set", "now");
            }
            break;

        case 'VERY_CHEAP':
            console.log('Strom ist sehr günstig');
            if (getState("0_userdata.0.roday_charge_cheap").val)
            {
                setState("mqtt.0.evcc.loadpoints.1.mode.set", "now");
            }
            if (getState("0_userdata.0.speedy_charge_cheap").val)
            {
                setState("mqtt.0.evcc.loadpoints.2.mode.set", "now");
            }
            break;

        default:
            console.log('Strompreis unbekannt');
    }

});
```

andig, 2022-12-03:
Wie soll Tibber entscheiden, ob der Preis für Dich günstig ist? Meine bevorzugte Lösung (ohne Planner) wäre immer noch ein Api dafür. Spricht da was dagegen?

ThiloBaWue, 2022-12-04:
Möchte auch mein starkes Interesse an einer bequemen Möglichkeit zum kostengünstigen Laden mit Tibber bekunden. Tibber Pulse ist installiert und seit gestern das EAuto geliefert. PV-Überschuss aber nicht verfügbar.
Wäre es nicht, eine auch programmtechnisch einfache Möglichkeit, alle Stundenpreise bis zur Zeit des Ziel-SOc zu laden und zu vergleichen und dann zu den günstigsten Zeiten zu laden?
Klar ist das eigentliche ermitteln des günstigsten Zeitpunktes einfach, aber das es zumeist in einer Stunde nicht getan ist, macht es schwieriger. Die aufzuladende Energiemenge und die Ladegeschwindigkeit wäre dann auch zu berücksichtigen.

andig, 2022-12-04:
> Wäre es nicht, eine auch programmtechnisch einfache Möglichkeit, alle Stundenpreise bis zur Zeit des Ziel-SOc zu laden und zu vergleichen und dann zu den günstigsten Zeiten zu laden?

Ich wiederhole mich. Das ist https://github.com/evcc-io/evcc/pull/2211.

andig, 2022-12-11:
#2211 wird durch https://github.com/evcc-io/evcc/pull/5445 abgelöst. Damit gibt es keinen "günstigen" Tarif mehr, sondern der Planner wählt die jeweils günstigsten Ladeslots aus um das Ziel zu erreichen.

closed 2022-12-23 by andig