evcc-io/evcc #5338: allow minCurrent setting of less than 6A

stewartwatson321, 2022-12-01:
Is it possible to allow/overwrite minCurrent setting to be less than 6A?

I appreciate that IEC 61851 specifies a minimum of 6A, but some wallboxes and vehicles will allow charge at lower currents. 

This would be very useful for example, in winter time when my PV generation is only around 400 - 600W (and others will have similar issues too I imagine).  It would be good to be able to allow the vehicle to 'trickle charge'.

Even if this setting is 'not advisable' is there some mechanism to override the minimum charging in evcc to make it lower for 'testing'?

premultiply, 2022-12-02:
No, that is not possible.
evcc controls the technically maximum allowed phase current.
The vehicle may consume up to this or less current at any time.
Even at 6A the losses of all the charging related computers and electronics inside the car are already very high compared to the power that enters the battery.

closed 2022-12-02 by premultiply

andig, 2022-12-03:
@stewartwatson321 there should be nothing restraining you from setting a 5A min limit if your wallbox supports that.