evcc-io/evcc #5346: Etrel: Geladene Energie 0,0 kWh

premultiply, 2022-12-02:
### Discussed in https://github.com/evcc-io/evcc/discussions/5343

<div type='discussions-op-text'>

<sup>Originally posted by **obrenner** December  2, 2022</sup>
Hallo,

nachdem ich das release 0.108.3 installiert habe, funktionier mein EVCC wieder und mein Sonnencharger 
Funktioniert wieder, und wird auch gesteuert wie es mein evcc vorgibt.

Leider wird beim laden zwar die Ladeleistung und zeit im Ladepunktfenster angezeigt, aber die geladene Energie, zählt sich nicht hoch, was bisher immer funktioniert hatte.
In meiner Konfiguration hat sich nichts geändert, und ich finde es auch auch nicht. Hat sich beim update ein fenster eingeschlichen oder muss ich da in der konfig was anpassen?

das angeschlossene Auto ist ein Hyundai Kona und EVCC holt die Fahrzeugdaten per Bluelink
</div>

obrenner, 2022-12-02:
Ich habe den EVCC aktuell auf nem Homeassistant am laufen, muss ich evcc charger —log trace über die configdatei aktivieren? Oder in welchem verzeichnis in der CLI ist das auszuführen?

sorry bin aber leider in dem Thema Homeassistant und evcc log eher ein neuling :-(

andig, 2022-12-03:
Schau bitte in die Doku. Für HA bist Du auf Dich gestellt, das können wir nicht supporten.

andig, 2022-12-03:
Bzgl. Session Energy: da hat sich nichts geändert. Es wird Register 30 ausgelesen (Connector 1). Das ist laut Doku korrekt.

premultiply, 2022-12-04:
@andig Vorher wurde die geladene Energie nicht vom Charger gelesen.

andig, 2022-12-04:
I see. Dann bauen wir die einfach wieder aus?

premultiply, 2022-12-04:
Ja, wenn es sonst nicht geht schon.
Erst würde ich aber dennoch gerne verstehen warum es nicht geht und dies womöglich fixen.

obrenner, 2022-12-04:
Hallo, also bei mir wurde bis zum letzten update die geladene Energie als angezeigt im evcc. Um evcc korrekt das logging zu erstellen, werde ich evcc als eigene Instanz installieren und vom Homeassistant trennen. Hatte ich eh schon länger mal vor..
Gruß Oliver

andig, 2022-12-04:
Es würde ja völlig reichen mal

    docker run -v ....config.... evcc/evcc evcc charger --log trace

zu machen. Da muss gar nix installiert werden.

andig, 2022-12-05:
> @andig Vorher wurde die geladene Energie nicht vom Charger gelesen.

@premultiply laut Diff (https://github.com/evcc-io/evcc/commit/b50f0d4ac09e6b93853fe7e80b018e503b1be3d8) hatten wir das vorher schon. ich verstehe also nicht, was sich geändert haben soll?

@obrenner geht denn die alte Version noch? Schneller wären wir sicher, wenn Du uns (info@evcc.io) einfach mal Remotezugang geben könntest (=Portweiterleitung).

premultiply, 2022-12-05:
https://github.com/evcc-io/evcc/commit/3f8849a205c84ba7de4f7d4e564378905a8cf931#diff-4e419fdc34c360500fe6d2f269e04234a677f86de992249274afec74c326e17cR205

obrenner, 2022-12-05:
hallo,

habe evcc als Standalone auf ner maschine am Laufen und das Log erstellt dabei gibt es folgenden outpout:

administrator@evccsrv01:~$ sudo systemctl stop evcc
[sudo] password for administrator:
administrator@evccsrv01:~$ evcc charger --log trace
[main  ] INFO 2022/12/05 18:58:50 evcc 0.108.3 (42ca83d3)
[main  ] INFO 2022/12/05 18:58:50 using config file: /etc/evcc.yaml
[db    ] TRACE 2022/12/05 18:58:50 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2022/12/05 18:58:50 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2022/12/05 18:58:50 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2022/12/05 18:58:50 SELECT * FROM `settings` 6 <nil>
[etrel ] TRACE 2022/12/05 18:58:50 modbus: send 00 01 00 00 00 06 01 04 00 1a 00 02
[etrel ] TRACE 2022/12/05 18:58:50 modbus: recv 00 01 00 00 00 07 01 04 04 40 48 b4 39
[etrel ] TRACE 2022/12/05 18:58:50 modbus: send 00 02 00 00 00 06 01 04 00 0e 00 06
[etrel ] TRACE 2022/12/05 18:58:50 modbus: recv 00 02 00 00 00 0f 01 04 0c 40 ab 02 0c 40 a7 ef 9e 40 a8 72 b0
[etrel ] TRACE 2022/12/05 18:58:50 modbus: send 00 03 00 00 00 06 01 04 00 00 00 01
[etrel ] TRACE 2022/12/05 18:58:50 modbus: recv 00 03 00 00 00 05 01 04 02 00 04
[etrel ] TRACE 2022/12/05 18:58:50 modbus: send 00 04 00 00 00 06 01 04 00 04 00 02
[etrel ] TRACE 2022/12/05 18:58:50 modbus: recv 00 04 00 00 00 07 01 04 04 40 c0 00 00
[etrel ] TRACE 2022/12/05 18:58:50 modbus: send 00 05 00 00 00 06 01 04 00 1e 00 02
[etrel ] TRACE 2022/12/05 18:58:50 modbus: recv 00 05 00 00 00 07 01 04 04 00 00 00 00
[etrel ] TRACE 2022/12/05 18:58:50 modbus: send 00 06 00 00 00 06 01 04 00 20 00 04
[etrel ] TRACE 2022/12/05 18:58:50 modbus: recv 00 06 00 00 00 0b 01 04 08 00 00 00 00 00 00 00 00
Power:          3136W
Current L1..L3: 5.34A 5.25A 5.26A
Charge status:  C
Enabled:        true
Charged:        0.0kWh
Duration:       0s
administrator@evccsrv01:~$ sudo systemctl start evcc
administrator@evccsrv01:~$

hoff da sind nun alle wichtigen infos drin, ist auch die 0.108.3 und hat auch den gleichen Fehler mit der Anzeige der geladenen Energie wie die instanz die ich im Homassistant laufen hatte.

die Instanz unter homeassistant gibt es noch ist aber gestoppt.

hoffe es hilft euch weiter, aber wenn ich es richtig verfolge habt ihr da schon was entdeckt :-)

PR #5384 (2022-12-06): Etrel: remove ChargeTimer and ChargeRater
Charged energy and charges time registers return zero only

Fixes #5346
closed 2022-12-06 by andig