evcc-io/evcc #5369: DaheimLaden API, Phasenspannungen auslesen

VolkerK62, 2022-12-04:
Im Status-Abruf der DaheimLaden API sind auch die Phasenspannungen enthalten. Könnte man diese auch per MQTT zur Verfügung stellen?
Hintergrund: 
Je nach Schaltzustand des Lasttrennschalters könnte ich damit den richtigen `phases` Wert automatisch per API senden.

premultiply, 2022-12-04:
Mmmmmh.
Wenn man so drüber nachdenkt könnten wir dies theoretisch auch so in den Charger implementieren dass die Phasenspannungen mit abgefragt und für diesen Zweck (vergleichbar mit den Strömen) ausgewertet werden.
Wenn Spannung aussschließlich auf L1 dann loadpoint: phases: 1 sonst loadpoint: phases 3
Dann würde es mit externem Lasttrennschalter automatisch (vor Ladebeginn) passend umspringen können.

/cc @andig

andig, 2022-12-04:
Wofür brauchen wir das wenn wir doch die Ströme haben? Wenn die WB nur 1p angeschlossen ist sollte das in der Config stehen. Was ist der zusätzliche Nutzen?

VolkerK62, 2022-12-04:
Die Ströme hat man ja erst, wenn geladen wird. Dann funktioniert die Erkennung super.
Die Spannungen hast du auch vor Beginn der Ladung schon. Man spart sich dadurch das manuelle ändern der `phases` per API und hat so immer die richtige `phases` Einstellung, damit die Ladung auch korrekt startet.

Mir würden die Werte per MQTT reichen, das bau ich mir dann extern = der VW.
Die Idee von premultipy hat sehr viel Charme und erreicht mehrere Nutzer = der Mercedes  ;-)

andig, 2022-12-04:
Also mit meinen Worten:
- wenn man eine dumme WB hat
- die über einen 3p Zähler verfügt
- der Phasenspannungen gegen Neutral liefert
- und die WB manuell zwischen 1p/3p umschaltet
- und die evcc Config auf "auto" (`phases: 0`) stehen lässt
- ...kann man sich den Api-Aufruf sparen und die Warnung beim Programmstart entfällt-

Das ist sicher machbar, Kosten/Nutzen scheinen mir für die Spezialanwendung aber kein gutes Verhältnis zu haben- zumal man erstmal einen Anwender finden muss, der das überhaupt versteht.

premultiply, 2022-12-04:
Ja, mit dem Unterschied dass ich den Mehrwert als durchaus relevant betrachte.
Die meisten marktüblichen Standardwallboxen mit integrierter Messeinrichtung kann man so völlig umstandslos und sonstiges "Spezialwissen" mit einem Lasttrennschalter sehr einfach aufwerten.
"Low hanging fruits" und Alleinstellungsmerkmal sozusagen.

VolkerK62, 2022-12-04:
>zumal man erstmal einen Anwender finden muss, der das überhaupt versteht.

gefunden ;-)

Ich schreib dann auch die Doku dazu.

andig, 2022-12-04:
> Die meisten marktüblichen Standardwallboxen mit integrierter Messeinrichtung kann man so völlig umstandslos und sonstiges "Spezialwissen" mit einem Lasttrennschalter sehr einfach aufwerten.

Dann müssen wir das bei jedem Durchlauf zusätzlich einlesen? Weiterer Traffic für einen super seltenen Anwendungsfall.

premultiply, 2022-12-04:
Ja, zumindest vor dem Laden.
Den Impact halte ich für vertretbar. Sollten ja meist wie die Phasenströme en Block lesbar sein und wäre ja auch nur für die Chargemeter.

VolkerK62, 2022-12-04:
Reicht es nicht, das erst im Fall des connect zu machen?
So habe ich es aktuell in meiner externen Lösung geregelt.

andig, 2022-12-04:
Und woher soll evcc wissen, ob du nicht nach einem Connect Deine WB abgestöpselt hast?

VolkerK62, 2022-12-04:
du erkennst doch auch `car connected` und `car disconnected`. Das kommt doch von der WB, oder?
Eigentlich braucht man die Aktion doch nur einmal bei `car connected` ausführen.

premultiply, 2022-12-05:
Die Umschaltung könnte auch im Zustand connected erfolgen.
Ok, sagen wir die Spannungen werden auch einfach zyklisch wie alles andere auch eingelesen. 😉
Dürfte den Bock auch jetzt nicht so viel fetter machen...

closed 2022-12-29 by premultiply