evcc-io/evcc #5379: Regler schwingt bei Ladestart bei Verwendung von "decay"

markus80d, 2022-12-05:
Hallo zusammen,

ich habe immer noch Probleme mit Reglerschwingen beim Ladestart des Autos wenn z.B. 3000W Einspeisung gerade nachmittags da sind und eine Mittellang mittels "decay" verwendet wird. Intervall 15s, decay 0.1.

Problem ist dass dann der Mittelwert bei 3000W Einspeisung steht und dann erst sehr langsam runtergeht. Während dessen wird voll hochgeregelt inkl. Phasenumschaltung. Dann zieht der Mittelwert in Netzbezug und der Regler regelt voll runter inkl. Phasenumschaltung auf 1-phasig.

Und das geht dann immer so weiter.

Problem ist dass der Regler alle 15s updatet, aber der Mittelwert erst verzögert reagiert. Bei decay=1 schwingt nix, aber ich möchte decay nutzen und da habe ich beim Ladestart immer Probleme.

Gut wäre wenn es wie schon weiter oben Mal geschrieben die Möglichkeit gäbe ein separates Intervall für Reglerupdate einstellen zu können oder entsprechend dem eingestellten decay automatisch anzupassen (z.B. Intervall 15s, decay 0.1 würde ein Reglerupdate alle 150s evtl. gut sein).

[image]

Meiner Meinung nach wäre hier bei Verwendung von "decay" ein zusätzliches "Intervall Reglerupdate" eine Lösung.

Danke und Grüße,
Markus

_Originally posted by @markus80d in https://github.com/evcc-io/evcc/discussions/5004#discussioncomment-4314884_

andig, 2022-12-05:
Bitte keine Beschäftigungstherapie. Es reicht wenn wir das an einer Stelle diskutieren.

markus80d, 2022-12-05:
OK. Wusste nicht an welche Stelle es gehört (Discussion oder issued). Tue mir eh schon schwer genug hier an dieser Stelle gut gemeinte Verbesserungsvorschläge anzubringen... ich denke ich gebe es dann jetzt auch auf, denn mehrmals das gleiche nochmal anbringen zu müssen, um dann wieder gleich geschlossen zu werden, frustriert mich dann doch langsam.

Ich kann leider aktiv aufgrund meiner anderen Fachrichtung nicht anders beitragen als Sachen die mir auffallen anzubringen...

markus80d, 2022-12-05:
Möchtest du dann evtl. "decay" ganz rausnehmen? Weil so funktioniert der meiner Meinung nach nicht. Ich hab den Ladestart auch in Excel gestern einfach kurz nachgestellt und da kommt genau das reale Verhalten raus das ich heute nochmal nachgestellt habe.

andig, 2022-12-05:
An Decay ist nix falsch. Es passt einfach nicht für Deinen Anwendungsfall. Siehe Discussion. 

> OK. Wusste nicht an welche Stelle es gehört (Discussion oder issued).

Das ist egal. In jedem Fall nur an eine.