evcc-io/evcc #5388: VW ID /status call throwing 404

panzerdev, 2022-12-05:
### Describe the bug

Call `https://mobileapi.apps.emea.vwapps.io/vehicles/WVWZZZE1****/status` returns `404 {"message":"Not Found"}`

Tested solution would be to switch to 
`https://mobileapi.apps.emea.vwapps.io/vehicles/WVWZZZE1****/selectivestatus?jobs=all` which returns all data just in a different response JSON. A bit of object transformation does the trick.

I'll do a PR tomorrow.

Example:
```
{
    "automation": {
        "climatisationTimer": {
            "value": {
                "timers": [
                    {
                        "id": 1,
                        "enabled": true,
                        "recurringTimer": {
                            "startTime": "04:55",
                            "recurringOn": {
                                "mondays": false,
                                "tuesdays": true,
                                "wednesdays": true,
                                "thursdays": true,
                                "fridays": true,
                                "saturdays": false,
                                "sundays": false
                            }
                        }
                    },
                    {
                        "id": 2,
                        "enabled": false,
                        "singleTimer": {
                            "startDateTime": "2022-11-26T06:45:00Z"
                        }
                    }
                ],
                "carCapturedTimestamp": "2022-12-05T21:25:30.835Z",
                "timeInCar": "2022-12-05T22:25:31+01:00"
            }
        },
        "chargingProfiles": {
            "value": {
                "carCapturedTimestamp": "2022-12-05T21:25:32.234Z",
                "timeInCar": "2022-12-05T22:25:31+01:00",
                "profiles": []
            }
        }
    },
    "userCapabilities": {
        "capabilitiesStatus": {
            "value": [
                {
                    "id": "webApp",
                    "userDisablingAllowed": false
                },
                {
                    "id": "automation",
                    "expirationDate": "2025-02-12T23:59:59Z",
                    "userDisablingAllowed": true
                },
                {
                    "id": "charging",
                    "expirationDate": "2025-02-12T23:59:59Z",
                    "userDisablingAllowed": true
                },
                {
[... 226 lines cut ...]
            }
        }
    }
}
```

### Steps to reproduce

See bug description

### Configuration details

```yaml
VW ID car configured
```

### Log details

```text
404 in log
```

### What type of operating system are you running?

Linux

rivengh, 2022-12-06:
The 404 error also appears with legacy VW vehicle type (here: e-up).

markuspulch, 2022-12-06:
same here for Hybrid Tiguan

panzerdev, 2022-12-06:
I am sorry but I can only fix the ID version of the API integration due to only having access to an ID car for testing.

closed 2022-12-07 by andig