evcc-io/evcc #5389: Problem Status auslesen Schrack i-charge CION

thomas-hutterer-tik, 2022-12-06:
### Describe the bug

Ich versuche eine Schrack i-charge CION Wall Box zu steuern und habe dabei das Problem, dass der Status (A, B, C) nicht in EVCC ankommt, sondern immer angezeigt wird dass kein Auto verbunden ist unabhängig davon welchen Status die Wall Box liefert. Im Anhang das evcc.yaml und debug log. Ich vermute, dass das Problem, damit zusammenhängt dass der "charger status" im Log in einer eigenen neuen Zeile angezeigt wird.
Wenn ich das Status Register 139 mit einem Tool wie mbpoll abfrage, erhalte ich den Wert 65 (= A)

### Steps to reproduce

1. evcc-template.yaml als evcc.yaml einspielen
2. evcc neu starten
3. Auto an Schrack i-charge CION anstecken
4. evcc UI zeigt an, dass kein Auto verbunden ist
...

### Configuration details

```yaml
---
network:
  schema: "http"
  host: "0.0.0.0"
  port: 7070
interval: "2s"
log: "debug"
meters:
- name: "Sage Controller "
  type: "custom"
  power:
    source: "mqtt"
    topic: "wa-pi/grid"
- name: "Sage Batterie"
  type: "custom"
  power:
    source: "calc"
    mul:
    - source: "modbus"
      uri: "10.68.11.30:502"
      id: 1
      register:
        address: 26
        type: "holding"
        decode: "int16"
      scale: -0.1
    - source: "modbus"
      uri: "10.68.11.30:502"
      id: 1
      register:
        address: 27
        type: "holding"
        decode: "int16"
  soc:
    source: "modbus"
    uri: "10.68.11.30:502"
    id: 1
    register:
      address: 20
      type: "holding"
      decode: "uint16"
- name: "Sage Fronius"
  type: "template"
  template: "fronius-solarapi-v1"
  usage: "pv"
  host: "10.68.10.80"
chargers:
- name: "Schrack_i_Charge_TCP-1"
  type: "template"
  template: ichargecion
  modbus: tcpip
  host: 10.68.10.36
  port: 502
  id: 2
site:
  title: "SageStud Gumpersberg"
  meters:
    grid: "Sage Controller "
    pvs:
    - "Sage Fronius"
    batteries:
    - "Sage Batterie"
loadpoints:
- title: "Schrack_i_Charge_TCP-1"
  charger: "Schrack_i_Charge_TCP-1"
  phases: 1
mqtt:
  broker: "10.68.11.77"
  topic: "evcc"
  user: "********"
  password: "*********"
```

### Log details

```text
Dec 06 07:40:06 iwattcontroller evcc[1835]: [site  ] DEBUG 2022/12/06 07:40:06 ----
Dec 06 07:40:06 iwattcontroller evcc[1835]: [lp-1  ] DEBUG 2022/12/06 07:40:06 charge power: 0W
Dec 06 07:40:07 iwattcontroller evcc[1835]: [site  ] DEBUG 2022/12/06 07:40:07 pv power: 527W
Dec 06 07:40:07 iwattcontroller evcc[1835]: [site  ] DEBUG 2022/12/06 07:40:07 battery power: -336W
Dec 06 07:40:07 iwattcontroller evcc[1835]: [site  ] DEBUG 2022/12/06 07:40:07 grid power: 22296W
Dec 06 07:40:07 iwattcontroller evcc[1835]: [site  ] DEBUG 2022/12/06 07:40:07 battery soc 0: 24%
Dec 06 07:40:07 iwattcontroller evcc[1835]: [site  ] DEBUG 2022/12/06 07:40:07 site power: 21959W
Dec 06 07:40:07 iwattcontroller evcc[1835]: [lp-1  ] DEBUG 2022/12/06 07:40:07 charger status:
Dec 06 07:40:07 iwattcontroller evcc[1835]: A
Dec 06 07:40:08 iwattcontroller evcc[1835]: [site  ] DEBUG 2022/12/06 07:40:08 ----
Dec 06 07:40:08 iwattcontroller evcc[1835]: [lp-1  ] DEBUG 2022/12/06 07:40:08 charge power: 0W
Dec 06 07:40:09 iwattcontroller evcc[1835]: [site  ] DEBUG 2022/12/06 07:40:09 pv power: 549W
Dec 06 07:40:09 iwattcontroller evcc[1835]: [site  ] DEBUG 2022/12/06 07:40:09 battery power: -336W
Dec 06 07:40:09 iwattcontroller evcc[1835]: [site  ] DEBUG 2022/12/06 07:40:09 grid power: 22291W
Dec 06 07:40:09 iwattcontroller evcc[1835]: [site  ] DEBUG 2022/12/06 07:40:09 battery soc 0: 24%
Dec 06 07:40:09 iwattcontroller evcc[1835]: [site  ] DEBUG 2022/12/06 07:40:09 site power: 21954W
Dec 06 07:40:09 iwattcontroller evcc[1835]: [lp-1  ] DEBUG 2022/12/06 07:40:09 charger status:
Dec 06 07:40:09 iwattcontroller evcc[1835]: A
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.108.0

premultiply, 2022-12-06:
evcc-Dienst stoppen.
Wallbox ohne evcc Auto laden lassen.
`evcc charger --log trace` ausführen.

andig, 2022-12-06:
Woher kommen eigentlich die ganzen Anfürhungszeichen in Deiner yaml? Die können alle weg...

stephandiehl, 2022-12-07:
Ich hab die gleiche Wallbox und habe es mit dem Template nicht hinbekommen. Wenn ich mich recht erinnere waren es 2 Probleme:

- der Status wird nicht richtig übernommen, ich glaube die Variable innerhalb von evvc war 00 65 anstatt von 65, also eine Stelle zu viel. Damit bleibt dann der Status in loadpoint.go "hängen". 
- im template wird der Strom auf Adresse 127 geschrieben, die ist aber read-only laut Dokumentation. Richtig ist 101, siehe https://image.schrackcdn.com/bedienungsanleitungen/a_emcion_homecontroller_modbus_lookup_table_de.pdf

Nachdem ich nicht verstanden habe wie ich das Template für Punkt 1 ändern muss, hab ich eine eigen eigene Implementierung gemacht. (Punkt 2 ist leicht behebbar). Sie funktioniert so weit, allerdings muss ich sagen, dass die Wallbox etwas dumm ist und den aktuellen Strom nicht ausgibt. Damit stimmt die Ladestatistik nicht, da in meinem Fall z.b. 6A von evcc eingestellt werden, das Fahrzeug aber nur mit 3,5 kW lädt. Läuft bei mir mit der Einschränkung recht gut.

Ich bin Go-Programmierer - es funktioniert und ich kann einen PR machen. Hatte ich eh vor, da das selber bauen von evcc sicher nicht der beste weg ist.

premultiply, 2022-12-07:
Schaue ich mir an.
Kannst du mal bitte ein `evcc charger --log trace` bei ladendem Fahrzeug machen?

stephandiehl, 2022-12-07:
Gern! Du meinst mit meiner Implementierung oder mit dem Template?

thomas-hutterer-tik, 2022-12-07:
> evcc-Dienst stoppen. Wallbox ohne evcc Auto laden lassen. `evcc charger --log trace` ausführen.

bin am Montag wieder vor Ort, dann mach is das

thomas-hutterer-tik, 2022-12-07:
> Woher kommen eigentlich die ganzen Anführungszeichen in Deiner yaml? Die können alle weg...
Weiss ich, Ich generiere die yaml mit einer Java library, die macht die Anführungszeichen rein. Stören Gott sei Dank nicht

stephandiehl, 2022-12-07:
Hier das log bei meiner Implementierung
```
[main  ] INFO 2022/12/07 15:39:38 evcc 0.108.2 (1681b927)
[main  ] INFO 2022/12/07 15:39:38 using config file: /etc/evcc.yaml
[db    ] TRACE 2022/12/07 15:39:38 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2022/12/07 15:39:38 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2022/12/07 15:39:38 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2022/12/07 15:39:38 SELECT * FROM `settings` 0 <nil>
[cion  ] TRACE 2022/12/07 15:39:38 modbus: send 00 01 00 00 00 06 01 04 00 8b 00 01
[cion  ] TRACE 2022/12/07 15:39:38 modbus: recv 00 01 00 00 00 05 01 04 02 00 42
[cion  ] TRACE 2022/12/07 15:39:38 modbus: send 00 02 00 00 00 06 01 03 00 64 00 01
[cion  ] TRACE 2022/12/07 15:39:38 modbus: recv 00 02 00 00 00 05 01 03 02 00 01
```

stephandiehl, 2022-12-07:
Hier noch kurz was ich bei der Verwendung vom Template für den status zurückbekomme:

[image]

andig, 2022-12-07:
@stephandiehl das Problem sollte mit https://github.com/evcc-io/evcc/commit/f7acf467b28f8637d14d43f361d28435395ad7be behoben sein. Jetzt noch das falsche Register ändern und schon sollte das rennen?

stephandiehl, 2022-12-08:
@andig Vielen Dank! Komm leider erst am WE dazu zu testen.

andig, 2022-12-09:
Register ist in master ebenfalls gefixed, ab morgen im Nightly. Ich mach hier schonmal zu- wenn das Problem trotzdem noch auftritt bitte kurzes Update.

thomas-hutterer-tik, 2022-12-11:
Vielen Dank, ich werde das morgen testen, Ist 0.109.2 die richtige Version dafür?

thomas-hutterer-tik, 2022-12-11:
Das Log sieht mit der neuen Version gut aus:

Dec 11 19:31:21 iwattcontroller evcc[2816]: [site  ] DEBUG 2022/12/11 19:31:21 ----
Dec 11 19:31:21 iwattcontroller evcc[2816]: [lp-1  ] DEBUG 2022/12/11 19:31:21 charge power: 0W
Dec 11 19:31:21 iwattcontroller evcc[2816]: [site  ] DEBUG 2022/12/11 19:31:21 pv power: 0W
Dec 11 19:31:21 iwattcontroller evcc[2816]: [site  ] DEBUG 2022/12/11 19:31:21 battery power: -271W
Dec 11 19:31:21 iwattcontroller evcc[2816]: [site  ] DEBUG 2022/12/11 19:31:21 grid power: 37617W
Dec 11 19:31:22 iwattcontroller evcc[2816]: [site  ] DEBUG 2022/12/11 19:31:22 battery soc 0: 24%
Dec 11 19:31:22 iwattcontroller evcc[2816]: [site  ] DEBUG 2022/12/11 19:31:22 site power: 37346W
Dec 11 19:31:22 iwattcontroller evcc[2816]: [lp-1  ] DEBUG 2022/12/11 19:31:22 charger status: A
Dec 11 19:31:22 iwattcontroller evcc[2816]: [lp-1  ] DEBUG 2022/12/11 19:31:22 charger disable: contactor delay 4m33s
Dec 11 19:31:23 iwattcontroller evcc[2816]: [site  ] DEBUG 2022/12/11 19:31:23 ----

charger status: A wird jetzt in der gleichen Zeile angezeigt

stephandiehl, 2022-12-12:
@andig hatte leider keine Zeit zu testen. Ich hab gesehen, du hast das Register für maxcurrent auf 121 geändert. Ich glaub das passt nicht. Für mein Verständnis, maxcurrent ist die Ladestromvorgabe und das müsste laut Tabelle 101 sein, richtig?

thomas-hutterer-tik, 2022-12-12:
Test ist OK, Laden funktioniert!

thomas-hutterer-tik, 2022-12-12:
Ja in 101 ist maxcurrent

thomas-hutterer-tik, 2022-12-13:
Das sehe ich auch so.
BG Thomas

stephandiehl, 2022-12-17:
Ja, läuft bei mir jetzt mit dem template. Vielen Dank!