evcc-io/evcc #5418: HomeAssistant Addon Not starting after upgrade to 109.1

iStehle, 2022-12-07:
### Describe the bug

After latest upgrade add on is not starting anymore 

### Steps to reproduce

1. Update
2. Evcc Not Running 
3.after manual start “ Image evcc/evcc:0.109.1 does not exist for addon_49686a9f_evcc”
...
Not working after backup, maybe hassio problem 

### Configuration details

```yaml
Not appreciate
```

### Log details

```text
Not appreciate
```

### What type of operating system are you running?

Docker container

### Version

109.1

iStehle, 2022-12-07:
Works After Downgrade

goebelmeier, 2022-12-07:
Strange. Label has been pushed for all major archs: https://hub.docker.com/layers/evcc/evcc/0.109.1/images/sha256-84a3cb03ba2e9c356b9aa02c43b4b2ce0f36449b51c98d4dc7caa179db7ac089?context=explore

iStehle, 2022-12-07:
After HomeAssistant Core to 2022.12.0 and another upgrade to 0.109.1 Evcc is working

closed 2022-12-07 by iStehle