evcc-io/evcc #5423: Phasenvisualisierung falsch

andig, 2022-12-09:
Mir ist gerade aufgefallen, dass bei mir mit der 0.109.2 beim 1-Phasigem laden neuerdings drei Balken angezeigt werden. Wenn ich die Diskussion hier richtig verstanden habe war das so nicht gewollt oder irre ich mich?
[image]

[site  ] DEBUG 2022/12/09 11:47:44 ----
[lp-1  ] DEBUG 2022/12/09 11:47:44 charge power: 1611W
[site  ] DEBUG 2022/12/09 11:47:44 pv power: 1816W
[site  ] DEBUG 2022/12/09 11:47:44 battery power: 191W
[site  ] DEBUG 2022/12/09 11:47:44 grid power: -15W
[site  ] DEBUG 2022/12/09 11:47:44 battery soc 0: 38%
[site  ] DEBUG 2022/12/09 11:47:44 site power: 276W
[lp-1  ] DEBUG 2022/12/09 11:47:44 charge currents: [7.73 0.013 0.008]A
[lp-1  ] DEBUG 2022/12/09 11:47:44 detected phases: 1p
[lp-1  ] DEBUG 2022/12/09 11:47:44 charger status: C
[lp-1  ] DEBUG 2022/12/09 11:47:44 next soc poll remaining time: 59m39s
[lp-1  ] DEBUG 2022/12/09 11:47:44 vehicle soc: 66%
[lp-1  ] DEBUG 2022/12/09 11:47:44 vehicle range: 53km
[lp-1  ] DEBUG 2022/12/09 11:47:44 pv charge current: 5.8A = 7A + -1.2A (276W @ 1p)
[lp-1  ] DEBUG 2022/12/09 11:47:44 climater active: false, target temp: 21.0°C, outside temp: 0.0°C
[lp-1  ] DEBUG 2022/12/09 11:47:44 site power 276W >= 100W disable threshold
[lp-1  ] DEBUG 2022/12/09 11:47:44 pv disable in 2m40s
[lp-1  ] DEBUG 2022/12/09 11:47:44 pv disable timer remaining: 2m40s
[site  ] DEBUG 2022/12/09 11:47:44 telemetry: charge: Δ2495/2512Wh @ 1611W
[site  ] ERROR 2022/12/09 11:47:45 telemetry: upload failed: unexpected status: 403
[site  ] DEBUG 2022/12/09 11:48:04 ----

_Originally posted by @Robbe64 in https://github.com/evcc-io/evcc/issues/5345#issuecomment-1344149279_

naltatis, 2022-12-09:
Ha, vermutlich kleine Ströme auf L2 und L3. Dann werden wir den Minimum Check wohl doch auch in der UI brauchen.

RTTTC, 2022-12-09:
> Ha, vermutlich kleine Ströme auf L2 und L3

`[lp-1 ] DEBUG 2022/12/09 11:47:44 charge currents: [7.73 0.013 0.008]A `

RTTTC, 2022-12-11:
What if when hovered mouse or clicked on these bars it would show charge amps in text?
Maybe even released by evcc amps and taken by vehicle amps?

PR #5437 (2022-12-11): Ignore low currents in phase UI
fixes #5423

- Currents below 1A are ignored for detecting 1 bar or 2 bar visualisation.

[image]

- 2 bar visualisation is only used if L1 and L2 are active. One-phase charging on L2 will fall back to 3 bar visualisation.

[image]
closed 2022-12-11 by andig