evcc-io/evcc #5424: Reset der Phase, wenn resetOnDisconnect=true

manny4566, 2022-12-09:
Hi,

wäre es möglich beim konfigurierten Wert "resetOnDisconnect=true" auch die Phasen zurück zu setzen, nachdem das Kabel wieder entfernt wurde? 
Es geht um den Fall, dass das Laden temporär für einen Vorgang auf 1-phasig umgestellt wurde manuell und danach automatisch zurückgestellt werden soll auf den in der Konfig gesetzten Wert, damit man es nicht wieder manuell machen muss oder vergisst. 

```
loadpoints:
- title: Holzhütte
  charger: wallbox
  mode: minpv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
```
In der Gui bleibt der Wert auf 1-phasig bis zum Neustart oder manuellen zurücksetzen:
[image]

andig, 2022-12-09:
Wenn Du "absichtlich" auf 1p stellst kann das ja auch andere Gründe haben als nur für den Ladevorgang. "Kann" man machen, aber das erscheint mir nicht allgemeingültig zu sein?

manny4566, 2022-12-09:
> Wenn Du "absichtlich" auf 1p stellst kann das ja auch andere Gründe haben als nur für den Ladevorgang. "Kann" man machen, aber das erscheint mir nicht allgemeingültig zu sein?

Klar, aber möchte ich es dauerhaft haben oder hab einen tieferen Sinn darin, dass es auf 1p stehen soll, ändere ich es doch dauerhaft in der Konfig ab oder nicht ?

andig, 2022-12-10:
/cc @premultiply

premultiply, 2022-12-11:
Halte ich für wenig sinnvoll, da `phases` hier eine technische Eigenschaft des LPs ist und diese sich durch das Abstecken nicht ändert.
Das würde z.B. bei der Lasttrennschalterlösung sonst auch neue Probleme schaffen.

andig, 2022-12-11:
Hier gehts ja um den Fall, dass der Ladepunkt „auto“ kann. Das ist bei Trennschalter nicht der Fall?

premultiply, 2022-12-11:
Davon ist dort aber nichts zu lesen.

andig, 2022-12-26:
Scheint als hätten wir keinen Konsens das aktuelle Verhalten zu ändern und keine weiteren Argumente. Closing as wontfix.

closed 2022-12-26 by andig