evcc-io/evcc #5427: Add Tibber Smart Pulse as a meter provider

michis0806, 2022-12-09:
**Is your feature request related to a problem? Please describe.**
I just switched to tibber and also installed "tibber pulse", basically an IR-Sensor the delivers meter-states to tibber for their hourly accounting. I could already use tibber as a tariff provider but with the tibber pulse, the API should also provide current values of the production and consumption totals of my meter as well as the current consumption/production.

**Describe the solution you'd like**
I'd like to use tibber as a meter provider beside the current tariff implementation

**Describe alternatives you've considered**
Currently, I use the tibber integration in Home Assistant and use the HA API to call meter values to EVCC.

Thanks for your work!
Michael

premultiply, 2022-12-09:
Please provide more details regarding the interfaces (API description, REST interface, Modbus register description etc.).

michis0806, 2022-12-09:
@premultiply It's the Cloud API from tibber described here: https://developer.tibber.com/docs/reference

When I get it right, evcc already uses the API connection to get the current price when you set your tariff node to type: tibber... 
So it should be possible to connect to another endpoint that provides the meter data?

Thanks a lot!
Michael

Hofyyy, 2022-12-09:
It is supportet since EVCC 0.107 as a meter?! or do you mean something different?

andig, 2022-12-10:
Already implemented: https://docs.evcc.io/docs/devices/meters#tibber-pulse

closed 2022-12-10 by andig

michis0806, 2022-12-10:
Sorry, for that! I only looked here... https://docs.evcc.io/docs/reference/configuration/meters