evcc-io/evcc #5432: Since Upgrade from 108.x ti 109.2 the configured vehicle is no more shown, only as "unavailable"

mmrazovic, 2022-12-10:
### Describe the bug

Since Upgrade from 108.x ti 109.2 the configured vehicle is no more shown, only as "unavailable". I did not edit the config at all, it was just a software update to the latest version. Before 109.2 the vehicle name was shown ("Taycan" as configured in the config yaml).

### Steps to reproduce

1. Configure a Porsche vehicle
2.Upgrade from 108.x to 109.2
3.the vehicle name is no more shown, only as "unavailable"
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  eebus: info
  cache: error

# unique installation id
plant: xxx

interval: 30s ## control cycle interval

meters:
- type: template
  template: eastron 
  id: 1  
  host: modbusgw  
  port: 502  
  usage: grid  
  modbus: rs485tcpip  
  name: grid1
- type: template
  template: kostal-piko-legacy 
  usage: pv  
  host: xxx  
  user: xxx  
  password: xxx  
  name: pv2
- type: custom
  power:
    source: http
    uri: http://192.168.1.61/meter/0/power
    method: GET
    jq: .power
    timeout: 10s 
  name: ShellyPlug1

chargers:
- type: template
  template: ellipro 
  ski: xxx xxx xxx xxx xxx xxx B6AC 06E7 421B 104E  
  ip: 192.168.1.39  
  name: wallbox4

vehicles:
- type: template
  template: porsche 
  title: Taycan 4S CUV  
  user: xxx  
  password: xxx  
  vin: WP0ZZZxxx  
  capacity: 84  
  name: ev3

loadpoints:
- title: Elli Charger Pro Wallbox
  charger: wallbox4
  mode: now
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.36 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.11 # [currency]/kWh

site:
  title: Schönbuchweg 9
  meters:
    grid: grid1
    pvs:
    - pv2
```

### Log details

```text
Dec 09 16:43:24 pi3bplus evcc[1015]: [main  ] ERROR 2022/12/09 16:43:24 creating vehicle ev3 failed: cannot create vehicle 'template': cannot create vehicle 'po>
Dec 09 16:43:24 pi3bplus evcc[1015]: [site  ] INFO 2022/12/09 16:43:24 site config:
Dec 09 16:43:24 pi3bplus evcc[1015]: [site  ] INFO 2022/12/09 16:43:24   meters:      grid ✓ pv ✓ battery ✗
Dec 09 16:43:24 pi3bplus evcc[1015]: [site  ] INFO 2022/12/09 16:43:24     grid:      power ✓ energy ✗ currents ✓
Dec 09 16:43:24 pi3bplus evcc[1015]: [site  ] INFO 2022/12/09 16:43:24     pv 1:      power ✓ energy ✗ currents ✗
Dec 09 16:43:24 pi3bplus evcc[1015]: [site  ] INFO 2022/12/09 16:43:24   vehicles:
Dec 09 16:43:24 pi3bplus evcc[1015]: [site  ] INFO 2022/12/09 16:43:24     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
Dec 09 16:43:24 pi3bplus evcc[1015]: [lp-1  ] INFO 2022/12/09 16:43:24 loadpoint 1:
Dec 09 16:43:24 pi3bplus evcc[1015]: [lp-1  ] INFO 2022/12/09 16:43:24   mode:        now
Dec 09 16:43:24 pi3bplus evcc[1015]: [lp-1  ] INFO 2022/12/09 16:43:24   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
Dec 09 16:43:24 pi3bplus evcc[1015]: [lp-1  ] INFO 2022/12/09 16:43:24   meters:      charge ✓
Dec 09 16:43:24 pi3bplus evcc[1015]: [lp-1  ] INFO 2022/12/09 16:43:24     charge:    power ✓ energy ✗ currents ✓
Dec 09 16:43:24 pi3bplus evcc[1015]: [lp-1  ] INFO 2022/12/09 16:43:24 car disconnected
Dec 09 16:44:18 pi3bplus evcc[1015]: [lp-1  ] INFO 2022/12/09 16:44:18 vehicle updated: unknown -> unavailable
Dec 09 16:44:23 pi3bplus evcc[1015]: [lp-1  ] INFO 2022/12/09 16:44:23 vehicle updated: unavailable -> unknown
Dec 09 16:44:27 pi3bplus evcc[1015]: [lp-1  ] INFO 2022/12/09 16:44:27 vehicle updated: unknown -> unavailable
Dec 09 16:44:30 pi3bplus evcc[1015]: [lp-1  ] INFO 2022/12/09 16:44:30 vehicle updated: unavailable -> unknown
Dec 09 17:11:11 pi3bplus evcc[1015]: [lp-1  ] INFO 2022/12/09 17:11:11 vehicle updated: unknown -> unavailable
Dec 09 17:11:12 pi3bplus evcc[1015]: [lp-1  ] INFO 2022/12/09 17:11:12 vehicle updated: unavailable -> unknown
Dec 09 17:11:23 pi3bplus evcc[1015]: [lp-1  ] INFO 2022/12/09 17:11:23 vehicle updated: unknown -> unavailable
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.109.2

mmrazovic, 2022-12-10:
[image]

VolkerK62, 2022-12-10:
did you try `sudo systemctl restart evcc`?

mmrazovic, 2022-12-10:
I just did it, and it works! Problem solved! Thanks!

closed 2022-12-10 by mmrazovic